feelfreelinux/octo4a #343: Unknown serial driver Artillery Sidewinder X2

windowslucker1121, 2023-01-09:
Hey hey,
i wanted to setup octoprint on my Xiaomi Mi 9 but it seems im stuck on the following error:
[image]

I tried every serial driver and every baudrate which was available in octoprint but i just get the error messages "no serial port found"
[image]

I tried the latest apk, which is 1.1.2 and i builded the apk myself off the latest master branch which didnt worked either.

am i missing something?

Greetings and keep up the good work!

Filo301, 2023-01-10:
I had problem with stock mi9, give up pretty fast and flashed pixel experience. Then worked without any problem

windowslucker1121, 2023-01-10:
> I had problem with stock mi9, give up pretty fast and flashed pixel experience. Then worked without any problem. You should check which model of usb-ttl converter is inside your printer (if any)

Like you suggested, i installed the Pixel Rom on my Mi 9 and installed my selfbuild APK onto the device, sadly the same behaviour as described above.
I tried it with different USB hubs but sadly its not working.

dohamann, 2023-01-13:
I'm currently in a similar situation, trying to get my Sidewinder X2 to work with Octo4a and though I'm a lot further down the line than @windowslucker1121, I can only recommend to not use Octo4a. It's not nearly stable enough to use and causes more issues than it's worth. There are barely any resources to look up. I really like the concept but reality just doesn't hold up.

Abandon ship. Get a pi.

dohamann, 2023-01-13:
To actually respond to your issue: It's a bit weird that the app recognizes the device itself but doesn't offer any serial port. Have you tried having both the printer and your smartphone running and then remove all cables and put them back together?

May sound dumb, but I recall that I had an issue reconnecting after having rebooted the phone and for some reason it didn't work. There's a dialog in android  (at least the version of it that I'm using) that requests permission for the app to use the USB device, which I have to check every single time. It wouldn't work until I removed the cable from the phone. Re-plugging the printer-side of the cable didn't work.

[image]. 
[image].

CDC sounds about right. Auto-detection chooses a baud rate of 115200. But I do have issues with prints stuttering which leaves nasty printing artifacts, so I obviously can't vouch for these parameters actually working. But I guess being connected to the printer and being able to print is a big step in the right direction from where you are right now.

windowslucker1121, 2023-01-31:
to update on you guys:
@dohamann yes, i retried it multiple times and yes i granted the permission for accessing the usb device.
and lastly i followed your advice and bought an used rpi 3 b, which is working perfectly fine. 
fiddling around with that for 2 days was definitly not worth it :D

dohamann, 2023-02-01:
@windowslucker1121 In this case I really hate to be right. Anyway, have fun with your new toy!

closed 2023-03-21 by alufers