feelfreelinux/octo4a #355: Is a USB C to USB B cable appropriate?

salamanders, 2023-01-22:
I have a USB C phone.  The board in the printer has a USB B port.
All I need is a single OTG cable that runs from C (male) to B (male) and I'm good to go for charging the phone, controlling the printer, etc., correct?

[Like this one?](https://www.amazon.com/gp/product/B08MDDTY6B)

Or do I need to bounce through a USB hub for power?

abrugsch, 2023-02-06:
Your printer PROBABLY won't power the phone from the USB-A port. That isn't to say yours won't but every printer (not that many TBH) that I've used, will accept power FROM the host so for instance connecting the printer whilst powered off will allow the host to do comms with the microcontroller by powering from the host (PC). in this instance the host is your tablet so I'd say it's highly unlikely. I'm using a 4-port USB-C OTG splitter with a switch to enable one micro USB-B socket to be either a charge passthrough or a data port. [one of these, but USB-C](https://www.amazon.co.uk/Adapter-USB-C-Splitter-Keyboard-Mobile/dp/B0BNG2WK33) (found it on ebay but the product name was the same, it just had an option for USB-C in the listing)

alufers, 2023-03-21:
@abrugsch is right. The connection wil, work, but charging most ikely won't

closed 2023-03-21 by alufers