feelfreelinux/octo4a #357: Python traceback installing OctoPrint

tildebyte, 2023-01-24:
TL;DR (Firstly, I definitely understand that *this repo is not OctoPrint*, so none of this is directed at the maintainers here):

OctoPrint has adopted and is self-maintaining some Python abandonware package named "versioneer" (which I've never heard of, in spite of being a Python SWE by trade), which fails to find a valid version while `setup` is running. Lame reason to break an install (oh, also REALLY LOVING those `exec`s in there - very professional)

I tried hacking around versioneer's peculiar ideas about what a version is and where one comes from, but got nowhere.

I don't have any kind of theory as to why versioneer doesn't work on (some?) Android devices (not familiar enough with the platform's peculiarities)

```python
Processing /root/OctoPrint-OctoPrint-ab3ebf1
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-7sjc5l_2/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-7sjc5l_2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-9r1o78om
         cwd: /tmp/pip-req-build-7sjc5l_2/
    Complete output (31 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.9/site-packages/setuptools/__init__.py", line 16, in <module>
        import setuptools.version
      File "/usr/lib/python3.9/site-packages/setuptools/version.py", line 1, in <module>
        import pkg_resources
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3243, in <module>
        def _initialize_master_working_set():
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3226, in _call_aside
        f(*args, **kwargs)
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3255, in _initialize_master_working_set
        working_set = WorkingSet._build_master()
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 559, in _build_master
        ws = cls()
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 552, in __init__
        self.add_entry(entry)
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 608, in add_entry
        for dist in find_distributions(entry, True):
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2059, in find_on_path
        path_item_entries = _by_version_descending(filtered)
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2029, in _by_version_descending
        return sorted(names, key=_by_version, reverse=True)
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2027, in _by_version
        return [packaging.version.parse(part) for part in parts]
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2027, in <listcomp>
        return [packaging.version.parse(part) for part in parts]
      File "/usr/lib/python3.9/site-packages/packaging/version.py", line 52, in parse
        return Version(version)
      File "/usr/lib/python3.9/site-packages/packaging/version.py", line 197, in __init__
        raise InvalidVersion(f"Invalid version: '{version}'")
    packaging.version.InvalidVersion: Invalid version: '/usr/lib/python3.9/site'
    ----------------------------------------
WARNING: Discarding file:///root/OctoPrint-OctoPrint-ab3ebf1. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

tildebyte, 2023-01-25:
All of of this was done via ssh from my laptop.

Got 'octoprint'==1.8.6 from PyPI to install by installing 'wheel', *un*installing 'packaging' (it gets dep-installed later), and downgrading 'pip'==20.2.3 and 'setuptools'==52, all using `--force --user`. Also had to `apk add libffi-dev` to get 'argon2-cffi-bindings' to install.

Then, the octoprint scripts are installed in "/root/.local/bin" (not on $PATH), so `octoprint` must be launched manually e.g. `octoprint serve`

...except that permissions are jacked... So I moved everything from '~/.local' to the corresponding dirs in '/usr/local'.

My current issue is that '_psutil_linux.cpython-39-arm-linux-musleabihf.so' "was built for (5.8.0 instead of 5.9.4)", so I've jacked some dependency up during the process... more tomorrow, maybe.

tildebyte, 2023-01-25:
Ignore-all-that-⬆️-do-this:

### AS ROOT:

- `apk add bash bash-completion bash-doc binutils binutils-doc build-base cmake coreutils findutils gcc gcc-doc grep iproute2 libffi-dev linux-headers musl pciutils python3-dev sudo usbutils util-linux`  # install build tools and deps
- `pip3 uninstall packaging`  # don't know why
- `rm -fr /usr/lib/python3.9/site-packages/packaging`  # above doesn't completely remove it
- `pip3 install --force --user wheel pip==20.2.3  setuptools==52 octoprint==1.8.6`  # necessary downgrades; *latest* octoprint 🎉 
- `cp -rv .argon-fix .octoCredentials ioctlHook.so .octoprint extensions .local /home/octoprint/`  # secret sauce

### AS OCTOPRINT:

- `chmod u+x * /home/octoprint/.local/bin`   # octoprint user must own all the things
- `octoprint serve`  # alternately, `octoprint daemon start`

### Theory of operation

- 'octoprint' user doesn't have permissions to: install into a venv, install using `pip --user`; do much of anything EXCEPT actually run the server, which...
- ...'root' user is NOT allowed to start. But 'root' *can* build/install all the things
- do the installing/building as 'root'
- copy the results out to 'octoprint's home
- run `octoprint` as the 'octoprint' user

Of course, all of this is merely a workaround for whatever bizarre permissions issues are at play in alpine

jccraig26, 2023-01-27:
@tildebyte - Can I ask for some clarification?  I am a noob to this.  

I could not access my phone via SSH because Octoprint did not seem to make it far enough along in its startup (that is my theory since I get a connection error).  I can connect via simpleSSHD running on the phone but I am in the directory for that app.  Can you provide some more details on the directories I should be in when running those commands you gave?  

Thanks!

tildebyte, 2023-01-27:
It's all relative to whichever directory the *app's* sshd uses.

Unfortunately, I'm not able to gain any insight into that; I tried poking around in the sources in the repo here (for this and other reasons), but I don't hardly know anything about Android app development. so... :shrug:

alufers, 2023-03-23:
I think this should be fixed in 1.2.2 

Please ping me if it is not.

closed 2023-03-23 by alufers

jasta, 2023-12-21:
I got this issue after updating to OctoPrint 1.9.3 and even after upgrading to octo4a 1.2.5.  Specifically:

 https://gist.github.com/jasta/73580e36881db707407ec7f8cc562087