feelfreelinux/octo4a #360: Webcam question

RickzFPV, 2023-01-27:
Guys been running octo4a for a bit now and I do like it, even with the issues of getting installed.  Just had to figure some things out on it which I have done now, and it’s running really well now.  My question is can you only use the phone or tablet camera as the webcam or can I add a usb hub and a webcam.

plano-harry, 2023-02-09:
@RickzFPV did you ever get the Blu View 2 B131DL working?  I can't get in on SSH. "connection refused".   SSH works fine into my Fire HD 8, so I know that part is right, but I can't get the B131DL working.

RickzFPV, 2023-02-09:
I was never able to ssh into mine.  I think the problem is with android 11 not Octo4a

plano-harry, 2023-02-10:
Thanks, that must be it, although @fresh3dprints on YouTube supposedly got it running on the B131DL. That is the reason I bought it. Looks like the inovato Quadra may be the best way to go.

RickzFPV, 2023-02-10:
@plano-harry yeah I have been running the Quadra now for about a week using octoprint deploy, and can control and monitor 4 printers at the same time with it.  Best 35.00 I have spent in a while for 3D printing lol

alufers, 2023-03-21:
Duplicate of #95