feelfreelinux/octo4a #362: I forgot my password to OctoPrint what do i do ?

tondalic, 2023-01-31:
I tryed to SSH to the Octo4a and managed to connect to the IP after searching a lot ( port is 8022 !).
Have no idea what to put in the 

login as:
password:

as the defaul one to proceed further.

Is there any other way ? I just need to bea able to run the wizard again... Removing the app from the phone does not do that.

I am a noob in this field, so dumbed down instructions will be appreaciated.

THANKS

lmichoux, 2023-02-18:
I guess ou should do this:
1) Click the Octoprint icon
2) Click on the setting wheel icon at the bottom
3) Click on change the SSH password for the 'root' user
4) make sure "Enable SSH server" is selected.

Then ssh to the phone, with root user and your password.
done.

tondalic, 2023-02-23:
I managed to ssh to the phone ( thank you) , but what next ? 
I even deleted the users.yaml, but  my config.yaml does not have the "firstRun: false" line.
 I am completly lost by now.

alufers, 2023-03-21:
Please connect via SSH and see: https://community.octoprint.org/t/i-forgot-my-octoprint-password-how-can-i-reset-it/215