feelfreelinux/octo4a #363: The video stream freezes after a few seconds.

jhoniregis, 2023-02-02:
The camera stream only works for about 1 minute and then freezes. The only way I found to not freeze is by keeping the screen on with the preview open in the camera server, in the octoprint application. Any solution so I don't have to keep the screen always on with the preview open?
Device: Samsung Galaxy S9.
Octoprint Version 1.8.6

jhoniregis, 2023-02-02:
I think i found a solution.

I found this application at the link below that serves to keep the smartphone screen on. But even minimizing the Octoprint and locking the screen to turn it off, the video stream continues to work.

https://apkpure.com/br/it-s-on-keep-the-screen-on-q/br.not.sitedoicaro.itson

alufers, 2023-02-06:
Have you tried disabling battery optimization for the app?

alufers, 2023-03-24:
Looks like a problem with this specific device. I can't really help you :(

closed 2023-03-24 by alufers