feelfreelinux/octo4a #364: Octo4a consumes all available space over time

pricedev, 2023-02-04:
octo4a has been running on an S8+ for over a year.  with 53GB configured on the phone, only 200MB are available for use.  

The remaining space has been consumed but cannot determine what is consuming the space.  

O4a installation is base installation.  Camera is not configured.  

What and where the space is being taken up cannot be seen via the Camera's Android interface, which only shows 50+GB consumed by "Other".

pricedev, 2023-02-04:
Hmm, using the ncdu utility doesn't show any space hogs, and total consumed space is 3.5G, so something else is amiss.  Closing for now.

univers-ally, 2023-02-07:
Might be related to #368 since I have an S8. I've been seeing some generally strange behavior.

joejtrsmoe, 2023-02-11:
I had similar problem on s7. Found folder show hidden .octoprint\timelapse\tmp had 5gb of jpgs. Is it supposed to delete those after each render?
Thanks