feelfreelinux/octo4a #365: install problem with moto g

Wyatt-76, 2023-02-06:
I have tried to install octoprint 4a 1.2.0 on a moto g, but it was saying that the install was corrupted. I tried to do what issue #325 says to do. but when I went to open the ssh to run the command, but it was not wanting to connect. when I looked at the log on the phone it had password not found ran two times then a killed sshd. is there anything that I can do to fix this.

joejtrsmoe, 2023-02-06:
No longer able to install on galaxy s5. I used to be able to use solution in #325 but stays corrupted now.

Wyatt-76, 2023-02-07:
the main problem that I am having is that I can not get connected to the ssh

dzidza, 2023-02-07:
I am also still having the same problems on Samsung S8 Active

RickzFPV, 2023-02-09:
Sometimes I think it is the devices themselves, I have had Zero issues getting it installed on 2 Samsung J3 Phones running Android 9, also installed and ran fine, on Lenovo Tablets running Android 10, and on My other 2 Samsung tablets, running android 12 and 13, But have had no luck with 2 Blue phones running android 11.  I uninstalled and updated to the 1.2.0 version on all of them and it was sucessful on all of them except for the 2 Blu phones still no luck with them.

RickzFPV, 2023-02-09:
@Wyatt-76 that is the same problem I have with these 2 BLU phones no matter what I try or what I do they will not connect to ssh and that has to happen to get it working.  I decided to go with an Invato Quadra for 35.00 that is running linux and it solved all my issues.  https://www.inovato.com/

AAMBIENTT, 2023-04-14:
Ongoing with long attached. Looks like the person maintaining the repo is slacking. Someone else may need to pick up

`No bootstrap detected, proceeding with installation Arch: arm Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-armv7a.zip Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/2cd65898-00b6-4b5a-bd0f-7a036eff0c4f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230414%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230414T233252Z&X-Amz-Expires=300&X-Amz-Signature=45d593081bd7e133dbbe44038ce44455a8c06f0340a0e19e0e01cf2a41c09af7&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-armv7a.zip&response-content-type=application%2Foctet-stream returned status code 200 Bootstrap extracted, setting it up... >ls add-user.sh fake_proc_stat install-bootstrap.sh ioctlHook.so root root-pre5 rootfs.tar.xz run-bootstrap.sh >chmod -R 700 . >sh install-bootstrap.sh install-bootstrap.sh[8]: ../root/bin/minitar: can't execute: Permission denied install-bootstrap.sh[13]: can't create bootstrap/etc/resolv.conf: No such file or directory bootstrap ready, run with run-bootstrap.sh >sh add-user.sh octoprint add-user.sh[5]: can't create bootstrap/etc/passwd: No such file or directory An exception has occurred at:java.lang.RuntimeException: Process exited with error code 1. [add-user.sh[5]: can't create bootstrap/etc/passwd	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:38)	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:12)	at com.octo4a.repository.BootstrapRepositoryImpl$setupBootstrap$2.invokeSuspend(BootstrapRepository.kt:152)	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
Exception:java.lang.RuntimeException: Process exited with error code 1. [add-user.sh[5]: can't create bootstrap/etc/passwd`