feelfreelinux/octo4a #366: Still broken on Android 11

9cento, 2023-02-06:
Running Octo4a 1.2.0, same as before:
• Installation broken
• SSH broken
• TTYD extension broken
Literally nothing changed, no way to manually fix it at all without a working console, which there isn't.

RickzFPV, 2023-02-09:
@9cento i don’t think it is the octo4a in this circumstance, I think it is android 11 that is the problem I have been able to install it with no issues except for a device on android 11

9cento, 2023-02-09:
@RickzFPV yeah that is just watching the issue from the opposite perspective but we're back to square one since the app needs to be fixed and not the OS, don't you think?

RickzFPV, 2023-02-09:
@9cento yeah I agree I am not super fluent with android but you would think the developers could do some patches for the specific version of android but I am not too sure how much work that entails.  I personally ran into some other issues with it that got me to look at other options and move that direction

9cento, 2023-02-09:
I think a fix is definitely feasible, but we have to respect the fact that the devs are doing their work free of charge and they got to deal with everyday life. However in the meantime, do you care to share what other options you found? Because for the moment I'm stuck

sv0o, 2023-03-27:
I got one of the newer BLU phones that have A11 preinstalled.  Octo4a wouldn't install until I tried the latest version.
But I'm running my CR-6 SE right now off the BLU phone.  I am still trying to get Octolapse installed, but that's a separate issue.

sv0o, 2023-03-27:
Yes, for me B131DL with Android 11 works great.  With USB C and a brand new battery, it might be the ideal octoprint setup once I can get Octolapse running.  It's camera seems to make a bright and clear webcam as well.

RickzFPV, 2023-03-27:
@alufers I have it running on one of my B131DL’s also gonna do some testing with it now and see how it does.  If everything goes well this will be a great in expensive option for a lot of people for running octoprint.  I hope to be able to do some more work with you in the future for improvements moving forward.  I will now be adding a specific octo4a to my discord channel.

Rick

sv0o, 2023-03-27:
well this is pretty weird. I was 86% done with a 3hour print when octo4a on the BLU phone apparently crapped out.  The printer stopped, octoprint was no longer accessible, and when I woke up the phone, octo4a wasn't running anymore.  I started it again from the phone screen and it eventually came back up but there was no way to resume the print.  Octoprint was oblivious.  

was there a place to find octo4a logs?  I'm in ssh poking around.

RickzFPV, 2023-03-27:
@sv0o you need to go into developer tools and turn on the stay awake option.  Also not sure how you have yours setup but I am running a usb hub that allows me to connect to printer and charge phone at the same time.  Android as far as I can remember never gave you the option to keep the phone on all the time.  The keep awake In developers tools allows you to do so as long as it is plugged in.  I got mine setup earlier this morning with the new version but have only ran some smaller prints, I will run some longer prints on it to test and see how it does

RickzFPV, 2023-03-27:
@9cento sorry for the late reply, I haven’t been on here for a while one of the other inexpensive options I found was the inovato Quadra.  I have been using it for a while now and has given my the convenience of being able to control and monitor 4 printers from the one device.  They are only 35.00 US.  The only thing is the company will only ship to the US and Canada so if you are outside there they are not available.  I am also running 3 more printers with a RPI 3B.  With Octoprint Deploy it is now really easy to set up multiple instances of octoprint on 1 device

alufers, 2023-03-28:
> well this is pretty weird. I was 86% done with a 3hour print when octo4a on the BLU phone apparently crapped out. The printer stopped, octoprint was no longer accessible, and when I woke up the [...]

Sounds like some battery optimization to me, but I will try to add more logging

closed 2023-03-28 by alufers

sv0o, 2023-03-30:
OK, I was doing pretty well for a couple of days of small prints and then again today, it stopped dead in the middle of a bigger print.  When octo4a came back up it said camera server was disabled.  I had to disable it and re enable it to get video back.  So maybe its a camera server crash?

RickzFPV, 2023-03-30:
I just watched a video on this last night what is probably happening is whatever device you are running octo4a on is not processing the gcode fast enough for the printer and is causing octoprint to crash

sv0o, 2023-03-30:
OK.  maybe you'll have better luck with your B131DL.
Right now, I want to get two large prints done before Saturday, so I'm going back to the Remix MINI and I'll be able to use Octolapse pulling snapshots off the B131DL.  (I do like how bright and clear the camera is) 

Another possibility I'm considering is that running too long without a reboot of the android device could be a factor.

protocold, 2023-06-03:
sorry to resurrect this... @sv0o do you still have a working B131DL? Mine will install but i have encountered a bunch of permission related issue.

do you know the specific version you are running by any chance? Many thanks in advance.

sv0o, 2023-06-03:
> sorry to resurrect this... @sv0o do you still have a working B131DL? Mine will install but i have encountered a bunch of permission related issue.
> 
> do you know the specific version you are running by any chance? Many thanks in advance.

I do.  The O4A 1.2.4 installed on it without a problem when that version came out if I remember correctly.  If I reboot it or power cycle the phone every couple of weeks, it is quite stable for me. 

I recently updated Octoprint to 1.9.0 from Octoprint's web interface.  Nothing changed with O4A when that happened.  I hesitate to draw any conclusions about that.  I would like to note that I still cannot install Octolapse on it, but a recent discussion on that github page suggests that Octolapse doesn't play well with the x86 Alpine Linux environment or that it expects Arm64.  I still have an old LG G4 that runs O4A and Octolapse but I haven't used in in months because it's hard to keep it powered up.

9cento, 2023-06-03:
@sv0o I'm the guy from the Octolapse issue, it's not a cpu/architecture problem, it's just Android 11 specifically that doesn't play well in this scenario. The developer of Octolapse is investigating on that as we speak so hopefully we'll get a fix from him, otherwise it's up to @feelfreelinux to enter the game.

protocold, 2023-06-03:
Do you mind telling me the exact version of Android/update you are on? (there were updates since phone was purchased)? Sent from my iPhone

sv0o, 2023-06-03:
@protocold Yes, an Android security update was installed.  

Custom build version BLU B131DL V11.0.05.05.05.05 FSec

Build number RP1A.200720.011

protocold, 2023-06-04:
@sv0o thanks i checked the version it is same as mine..

but when i click on the log windows I saw a lot of permission related error with respect to /dev 

do you see them?

I went onto check this via ssh into the phone and i couldnt do even 'ls' from within /dev

sv0o, 2023-06-04:
I see the errors, but I don't think they affect Octoprint.

for these, I don't know if Alpine supports them or if the root user has all the same capabilities expected in a classic linux command line
1) same.  Permission denied but I don't think it affects octoprint
2)same.  Function not implemented

/dev/bus/usb/001/003
VID 1a86 / PID 7523
CH341 serial driver

protocold, 2023-06-04:
many thanks for your help... 

you have an Ender 3? Did you try using Klipper?

sv0o, 2023-06-04:
> many thanks for your help...
> 
> you have an Ender 3? Did you try using Klipper?

No, I have a CR-6 SE with Creality Firmware November 2022.
I haven't tried Klipper.  I'm still a noob and there is a lot to learn before I try to go fast.
I thought I was getting good, but then I tried printing PETG...

protocold, 2023-06-04:
dont know if it makes a difference.. i am on ender 3 with klipper firmware installed, it worked with rasberry pi.