feelfreelinux/octo4a #371: Error while scanning for serial ports: "Permission denied: '/dev'"

Kaassouffleetje, 2023-02-07:
I have setup Octo4a on an old Samsung, running LineageOS 18.1 (A11), rooted, without GApps. Everything went well, the app recognizes the printer and it asks for serial permissions. Even though I gave these permissions, when I go to the web interface it says that no serial ports have been found. I found the following error in the logs:

```🐙 [31m2023-02-06 21:55:06,467 - octoprint.util.comm - ERROR - Could not scan /dev for serial ports on the system [0m
🐙 Traceback (most recent call last):
🐙   File "/usr/lib/python3.10/site-packages/octoprint/util/comm.py", line 202, in serialList
🐙     with os.scandir("/dev") as it:
🐙 PermissionError: [Errno 13] Permission denied: '/dev'``

Mr-Savag3, 2023-03-04:
I'm having the same issue.

fritzbaratto, 2023-04-09:
I've installed Octo4a on my android smartphone (Xiaomi MI A3) and run it as OctoPi. I can access it from my laptop and install Plugins etc., but it cannot connect the printer (Flsun Q5) via my phone.
I get always 'no serial port found' so I suspect there is no, or bad, connection to the actual port from the printer.

I use the original cable that comes with the printer. It seems to work as I connected the cable directly to the laptop and the printer can be managed through the laptop via Repetier.

Whwre is the issue?
[image]

[image]

fritzbaratto, 2023-04-09:
[image]

fritzbaratto, 2023-04-10:
[image]

fritzbaratto, 2023-04-10:
Screenshots from  the app and the laptop

protocold, 2023-06-04:
i am also getting the same error on my B131DL phone... but some other user got it to work.. not sure how..

Drzet, 2023-06-04:
Same here on android nougat NVidia shield tablet. The app does not have permission to open devices.
On a separate note, I was only able to install octo4a after rooting the device and troubleshooting a million things about the install. It is very raw.

SvDvorak, 2023-07-13:
My old phone was not rooted and @Drzet's command did not work for me. But connecting works and when connected it stops trying to scan dev so it stops spamming the log.

n808, 2023-07-27:
Also happens on both my Android phones, Motorola 5GS and Huawai P30 Pro. 
But at least on Motorola, octoprint server works fine, except it drains so much battery that charging via OTG can't keep up. This may not be caused by this permission denied issue though.

WMSDevelop, 2023-10-01:
The same here...

mattmon, 2023-12-08:
Working for me after configuring SELinux to permissive mode.

See [my comment](https://github.com/feelfreelinux/octo4a/issues/439#issuecomment-1847793577) for [issue 439](https://github.com/feelfreelinux/octo4a/issues/439)

manudevelopia, 2024-01-04:
same here, is possible to solve it and run on a non-rooted device?

kardashov, 2024-04-19:
Having the same problem on not rooted Sony Xperia Compact X.

bhellema, 2024-04-28:
Same issue on Motorola One 5G ace, running version 1.2.6

feelfreelinux, 2024-05-17:
Hi, this message is mostly harmless - octo4a's serial driver will still be functional, as its hardcoded to use `/dev/ttyOcto4a` as a serial port. The log is displayed when OctoPrint tries to scan `/dev/` serial ports periodically, which is not functional - nor necessary. Octo4a 2.0.0 suppresses this error by default, by disabling "autorefresh" in OctoPrint's serial settings.

closed 2024-05-17 by feelfreelinux