feelfreelinux/octo4a #373: Octolapse Install not working eventhough Plugins are installed

JZiZu, 2023-02-08:
Hi! I am very thankfull this project was set up because of the prices of the RaspberryPis.
I hav now installed the new Version of Octo4a (1.2.0) on my old Samsung Galaxy A5 (2017) with Android 8.0.0 installed.
The server itself is working right now and I installed the plugin extras but when I try to install Octolapse I am getting many errors and it fails installing.
`error: command '/usr/bin/gcc' failed with exit code 1 [end of output] note: This error originates from a subprocess, and is likely not a problem with pip. error: legacy-install-failure × Encountered error while trying to install package. ╰─> Octolapse note: This is an issue with the package mentioned above, not pip. hint: See above for output from the failure. Fehler! Could not parse output from pip, see plugin_pluginmanager_console.log for generated output `
This is what I get at the end in the logs.
Have anyone else faced this problem yet? Or am I missing something?
PS: I just noticed that I was suppoesd to write an issue and not a discussion.

trytofly94, 2023-02-10:
I have the same problem today.

trytofly94, 2023-02-10:
A bit more of the error: 
There are a bunch of packages with the following:
```
############################
# Package would be ignored #
############################
Python recognizes 'octoprint_octolapse.data' as an importable package,
but it is not listed in the `packages` configuration of setuptools.

'octoprint_octolapse.data' has been automatically added to the distribution only
because it may contain data files, but this behavior is likely to change
in future versions of setuptools (and therefore is considered deprecated).

Please make sure that 'octoprint_octolapse.data' is included as a package by using
the `packages` configuration field or the proper discovery methods
(for example by using `find_namespace_packages(...)`/`find_namespace:`
instead of `find_packages(...)`/`find:`).

You can read more about "package discovery" and "data files" on setuptools
documentation page.

!!

check.warn(importable)
/usr/lib/python3.10/site-packages/setuptools/command/build_py.py:202: SetuptoolsDeprecationWarning:     Installing 'octoprint_octolapse.data.Images' as data is deprecated, please list it in `packages`.
!!
```
Her is the output after all of the "copying xxx"
```
UPDATING build/lib.linux-aarch64-cpython-310/octoprint_octolapse/_version.py
set build/lib.linux-aarch64-cpython-310/octoprint_octolapse/_version.py to 'v0.4.1'
running build_ext
Compiling Octolapse Parser Extension with <distutils.unixccompiler.UnixCCompiler object at 0x72a08bbeb0>.
building 'GcodePositionProcessor' extension
creating build/temp.linux-aarch64-cpython-310
creating build/temp.linux-aarch64-cpython-310/octoprint_octolapse
creating build/temp.linux-aarch64-cpython-310/octoprint_octolapse/data
creating build/temp.linux-aarch64-cpython-310/octoprint_octolapse/data/lib
creating build/temp.linux-aarch64-cpython-310/octoprint_octolapse/data/lib/c
gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -g -O2 -Os -g -O2 -Os -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/include/python3.10 -c octoprint_octolapse/data/lib/c/extruder.cpp -o build/temp.linux-aarch64-cpython-310/octoprint_octolapse/data/lib/c/extruder.o -O3 -std=c++11
gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -g -O2 -Os -g -O2 -Os -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/include/python3.10 -c octoprint_octolapse/data/lib/c/gcode_comment_processor.cpp -o build/temp.linux-aarch64-cpython-310/octoprint_octolapse/data/lib/c/gcode_comment_processor.o -O3 -std=c++11
octoprint_octolapse/data/lib/c/gcode_comment_processor.cpp: In member function 'void gcode_comment_processor::update_feature_from_section(position&) const':
octoprint_octolapse/data/lib/c/gcode_comment_processor.cpp:67:16: warning: enumeration value 'section_type_no_section' not handled in switch [-Wswitch]
67 |         switch (current_section_)
|                ^
gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -g -O2 -Os -g -O2 -Os -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/include/python3.10 -c octoprint_octolapse/data/lib/c/gcode_parser.cpp -o build/temp.linux-aarch64-cpython-310/octoprint_octolapse/data/lib/c/gcode_parser.o -O3 -std=c++11
octoprint_octolapse/data/lib/c/gcode_parser.cpp: In member function 'bool gcode_parser::try_parse_gcode(const char*, parsed_command&)':
octoprint_octolapse/data/lib/c/gcode_parser.cpp:151:59: warning: suggest parentheses around '&&' within '||' [-Wparentheses]
151 |                 else if (cur_char > 32 || cur_char == ' ' && has_seen_character)
|                                           ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -g -O2 -Os -g -O2 -Os -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/include/python3.10 -c octoprint_octolapse/data/lib/c/gcode_position.cpp -o build/temp.linux-aarch64-cpython-310/octoprint_octolapse/data/lib/c/gcode_position.o -O3 -std=c++11
octoprint_octolapse/data/lib/c/gcode_position.cpp: In member function 'void gcode_position::add_position(position&)':
octoprint_octolapse/data/lib/c/gcode_position.cpp:395:18: warning: operation on '((gcode_position*)this)->gcode_position::cur_pos_' may be undefined [-Wsequence-point]
395 |         cur_pos_ = (++cur_pos_) % NUM_POSITIONS;
octoprint_octolapse/data/lib/c/gcode_position.cpp: In member function 'void gcode_position::add_position(parsed_command&)':
octoprint_octolapse/data/lib/c/gcode_position.cpp:402:18: warning: operation on '((gcode_position*)this)->gcode_position::cur_pos_' may be undefined [-Wsequence-point]
402 |         cur_pos_ = (++cur_pos_) % NUM_POSITIONS;
octoprint_octolapse/data/lib/c/gcode_position.cpp: In member function 'void gcode_position::update(parsed_command&, long int, long int, long int)':
octoprint_octolapse/data/lib/c/gcode_position.cpp:633:79: warning: comparison of integer expressions of different signedness: 'const unsigned int' and 'int' [-Wsign-compare]
633 |                                                                 if (increment > p_current_pos->height_increment && increment > 1)
|                                                                     ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
octoprint_octolapse/data/lib/c/gcode_position.cpp: In member function 'void gcode_position::process_g10(position*, parsed_command&)':
octoprint_octolapse/data/lib/c/gcode_position.cpp:927:16: warning: variable 's' set but not used [-Wunused-but-set-variable]
927 |         double s = 0;
|                ^
[... cut ...]
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> Octolapse

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
Fehler!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```
I'm sorry if this is not helpful. I don't really know what one needs to get a solution...

Gorpokki, 2023-02-12:
Having similar errors attempting to install Octolapse. Also tried installing the python-dev package for python 3 as specified in Octolapse's installation instructions.

JZiZu, 2023-02-12:
I have now installed a wheel thanks to [lmichoux ](https://github.com/feelfreelinux/octo4a/discussions/370#discussioncomment-4945463) using this command:
pip3 install --upgrade setuptools wheel tornado --no-cache-dir
This seemed to change something but there is still something missing because I now get following message when installing Octolapse:

```
Installiere Plugin "Octolapse" von https://github.com/FormerLurker/Octolapse/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp307fdmix/Octolapse-master.zip --no-cache-dir
Processing /tmp/tmp307fdmix/Octolapse-master.zip
Preparing metadata (setup.py): started
error: subprocess-exited-with-error
× python setup.py egg_info did not run successfully.
│ exit code: 1
╰─> [10 lines of output]
Argon2 passlib backend is not available, not using it for password hashing
/tmp/pip-req-build-zvgish8u/setup.py:58: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
if LooseVersion(octoprint.server.VERSION) < LooseVersion("1.4"):
error in Octolapse setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after version specifier)
pillow >=6.2.0<7.0.0
~~~~~~~^
likely tags: v0.4.1
picking v0.4.1
got version from expanded keyword {'version': 'v0.4.1', 'full-revisionid': 'cec243c65a979e8bbe90f09fc5aec6d0844d2d3b', 'dirty': False, 'error': None, 'date': '2020-12-04T18:35:29-0600'}
Found packages: {'octoprint_octolapse_setuptools', 'octoprint_octolapse.test', 'octoprint_octolapse'}
[end of output]
note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed
× Encountered error while generating package metadata.
╰─> See above for output.
note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Preparing metadata (setup.py): finished with status 'error'
Fehler!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```
I do not know what to do now...

FatNat63, 2023-02-17:
Also having same issue

WombatMayhem, 2023-02-21:
I am also encountering this exact issue! Looking forward to hearing if anyone finds a solution for this one!

guywholikestobreakthings, 2023-02-23:
Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp1s3v4tg4/Octolapse-master.zip --no-cache-dir --user
Processing /tmp/tmp1s3v4tg4/Octolapse-master.zip
ERROR: Command errored out with exit status 1:
command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-ifqwedqx/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-ifqwedqx/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-lvje7d2s
cwd: /tmp/pip-req-build-ifqwedqx/
Complete output (10 lines):
Argon2 passlib backend is not available, not using it for password hashing
/tmp/pip-req-build-ifqwedqx/setup.py:58: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
if LooseVersion(octoprint.server.VERSION) < LooseVersion("1.4"):
error in Octolapse setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after version specifier)
pillow >=6.2.0<7.0.0
~~~~~~~^
likely tags: v0.4.1
picking v0.4.1
got version from expanded keyword {'version': 'v0.4.1', 'full-revisionid': 'cec243c65a979e8bbe90f09fc5aec6d0844d2d3b', 'dirty': False, 'error': None, 'date': '2020-12-04T18:35:29-0600'}
Found packages: {'octoprint_octolapse', 'octoprint_octolapse.test', 'octoprint_octolapse_setuptools'}
----------------------------------------
WARNING: Discarding file:///tmp/tmp1s3v4tg4/Octolapse-master.zip. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

a-c-m, 2023-02-25:
Another hitting the same issue. As this was first reported 2 weeks ago and 1.2 came out around then - i wonder if this is a bug introduced by that release?

a-c-m, 2023-02-26:
@guywholikestobreakthings / @JZiZu and others. I made a quick fork which added that missing semi comma - you can install directly from the UI using the following link: https://github.com/a-c-m/Octolapse/archive/refs/heads/master.zip

It doesn't fully work for me, but does manage to install the pillow dependency correctly. For me, it then gets to 

`Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): finished with status 'error'`
...
`42 | static const char* send_parsed_command_first = "first";|                    ^~~~~~~~~~~~~~~~~~~~~~~~~octoprint_octolapse/data/lib/c/stabilization.h:41:20: warning: 'snapshot_action' defined but not used [-Wunused-variable]41 | static const char* snapshot_action = "snapshot";|                    ^~~~~~~~~~~~~~~octoprint_octolapse/data/lib/c/stabilization.h:40:20: warning: 'travel_action' defined but not used [-Wunused-variable]40 | static const char* travel_action = "travel";|                    ^~~~~~~~~~~~~error: command '/usr/bin/gcc' failed with exit code 1[end of output]note: This error originates from a subprocess, and is likely not a problem with pip.ERROR: Failed building wheel for Octolapse`

I think i'm missing something else now. But i'm also using a very old phone.

a-c-m, 2023-02-26:
@guywholikestobreakthings / @JZiZu and others. I made a quick fork which added that missing semi comma - you can install directly from the UI using the following link: https://github.com/a-c-m/Octolapse/archive/refs/heads/master.zip

It doesn't fully work for me, but does manage to install the pillow dependency correctly. For me, it then gets to 

`Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): finished with status 'error'`
...
`42 | static const char* send_parsed_command_first = "first";|                    ^~~~~~~~~~~~~~~~~~~~~~~~~octoprint_octolapse/data/lib/c/stabilization.h:41:20: warning: 'snapshot_action' defined but not used [-Wunused-variable]41 | static const char* snapshot_action = "snapshot";|                    ^~~~~~~~~~~~~~~octoprint_octolapse/data/lib/c/stabilization.h:40:20: warning: 'travel_action' defined but not used [-Wunused-variable]40 | static const char* travel_action = "travel";|                    ^~~~~~~~~~~~~error: command '/usr/bin/gcc' failed with exit code 1[end of output]note: This error originates from a subprocess, and is likely not a problem with pip.ERROR: Failed building wheel for Octolapse`

I installed cmake and zlib as i thought they may be missing using apk

`apk add cmake`
`apk add zlib`

But it still failed with the building wheel error.

JZiZu, 2023-02-26:
Here you can find the temporary solution lmichoux came up with. 
It works for my installation. [https://github.com/feelfreelinux/octo4a/discussions/370](https://github.com/feelfreelinux/octo4a/discussions/370 )

guywholikestobreakthings, 2023-02-26:
> Here you can find the temporary solution lmichoux came up with. It works for my installation. #370

I dunno - Neither of them seems to work for me.

a-c-m, 2023-02-26:
I updated my fork with the code from #370 - which kind of worked, octolapse would build by pillow would fail, so i had to uninstall it and re-install it.

`pip remove pillow` `pip install --force-reinstall -vI "pillow==6.2.2"`
but it needed zlib and jpeg headers, so :

`apk add zlib-dev jpeg-dev`

then it worked without exploding! But... 

Even after installing 
`Successfully installed Octolapse-0.4.1+u.af13f5b
Done!`

and restarting.... i don't get it octolapse anywhere in the UI - or any errors. So now i'm confused.

guywholikestobreakthings, 2023-02-26:
Yes, this.  I was able to "successfully" install the older version, but it showed as "Unknown" installed, then doesn't appear anywhere in the UI under any pseudonym.

a-c-m, 2023-02-26:
Ok, I got it working. Once you've installed, don't restart right away! Save the changes and check its showing in the Plugin Manager list. 

That worked for me.

sv0o, 2023-03-03:
> Ok, I got it working. Once you've installed, don't restart right away! Save the changes and check its showing in the Plugin Manager list.
> 
> That worked for me.

I'm at this point.  your fork installed with errors (pillow).  Octoprint.log enumerates Octolapse along with the other plugins but it doesn't show up in plugin manager yet.  are you suggesting reinstalling again before restarting?

sv0o, 2023-03-27:
following up on this issue: Octolapse still works great for the Remix MINI with android 5
Since the new octo4A 1.2.3 and 1.2.4 added Android 11 support, I tried with the BLU B131 and now some new compiler issue appears to occur when Octolapse is installing .  I have tried the official Octolapse and the forked version. 

[plugin_pluginmanager_console.log](https://github.com/feelfreelinux/octo4a/files/11079786/plugin_pluginmanager_console.log)
 I'm just trying to figure what the error is at this point.

masademan, 2023-03-31:
I can't install octolapse either

TommyStrobbe, 2023-06-07:
> 

i have the same problem, did u got it to work ?

TommyStrobbe, 2023-06-07:
> Ok, I got it working. Once you've installed, don't restart right away! Save the changes and check its showing in the Plugin Manager list.
> 
> That worked for me.

didn't work for me :(

TommyStrobbe, 2023-06-07:
> 
i have the same problem, did u got it to work?

TommyStrobbe, 2023-06-07:
> are you referring to the problem installing into Android 11 or the earlier problem with deprecated dependencies? If you haven't tried the a-c-m branch, it's worth a shot https://github.com/a-c-m/Octolapse/archive/refs/heads/master.zip
> 
> but the latest official version has other updates in it. (see releases page https://github.com/feelfreelinux/octo4a/releases)

I try to install octolapse on macOS, I have python 3.9.2, no problems with other plugins, just this one won't work, tried the a-c-m and it did install as 'unknown' but I can't find it back in plugin manager, tried to save before restarting octoprint without success.

BaTaTaAdb, 2023-08-15:
Same issue here! On a Huawei P40 lite with Android 12 and installed Plugins Extra.

gleneccles, 2023-08-27:
@alufers Any chance we can get the formed version from @a-c-m merged so we can get octolapse working with octo4a? It is the only thing I have found so far that doesnt work in octo4a.

BaTaTaAdb, 2023-08-29:
@gleneccles I found that an older vesion of octolapse works flawlessly. I have been using for a month now and it's been great.
[Link](https://github.com/a-c-m/Octolapse/archive/refs/heads/master.zip)

Install it in the plugin browser through the link or the zip folder. 
[image]

gleneccles, 2023-09-01:
Ooh this is exciting. I will try it out tomorrow. Thank you for the idea, @BaTaTaAdb.

guywholikestobreakthings, 2023-09-01:
No dice for me...

Running setup.py clean for pillowerror: subprocess-exited-with-error×
python setup.py bdist_wheel did not run successfully.│ exit code: 1╰─>
[182 lines of output]running bdist_wheelrunning buildrunning
build_pycreating buildcreating
build/lib.linux-aarch64-cpython-310creating
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/MpoImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageMode.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/PngImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/XbmImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/PcxImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/SunImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageFile.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/SpiderImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/TarIO.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/FitsStubImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/MpegImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/BdfFontFile.py
-> build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/GribStubImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageStat.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/PixarImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/GimpPaletteFile.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageColor.py
-> build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/ContainerIO.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/MspImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/MicImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/_version.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/ImtImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/GifImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/PalmImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageQt.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageMath.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/PaletteFile.py
-> build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/FontFile.py
-> build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/PdfParser.py
-> build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ExifTags.py
-> build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageCms.py
-> build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/FpxImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageChops.py
-> build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/BufrStubImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/PSDraw.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/PcdImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImageFilter.py
-> build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/ImageDraw2.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/ImagePath.py ->
build/lib.linux-aarch64-cpython-310/PILcopying
src/PIL/DcxImagePlugin.py ->
build/lib.linux-aarch64-cpython-310/PILcopying src/PIL/__init__.py ->
[... 364 lines cut ...]
line 988, in run_commandcmd_obj.run()File
"/usr/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py",
line 346, in runself.build_extensions()File
"/tmp/pip-install-1z3384w4/pillow_21bd40cb0e204f509710eacaa70a074c/setup.py",
line 687, in build_extensionsraise
RequiredDependencyException(f)__main__.RequiredDependencyException:
zlibDuring handling of the above exception, another exception
occurred:Traceback (most recent call last):File "<string>", line 2, in
<module>File "<pip-setuptools-caller>", line 34, in <module>File
"/tmp/pip-install-1z3384w4/pillow_21bd40cb0e204f509710eacaa70a074c/setup.py",
line 903, in <module>raise
RequiredDependencyException(msg)__main__.RequiredDependencyException:The
headers or library files could not be found for zlib,a required
dependency when compiling Pillow from source.Please see the install
instructions at:https://pillow.readthedocs.io/en/latest/installation.html[end
of output]note: This error originates from a subprocess, and is likely
not a problem with pip.error: legacy-install-failure× Encountered
error while trying to install package.╰─> pillownote: This is an issue
with the package mentioned above, not pip.hint: See above for output
from the failure.Running setup.py install for pillow: finished with
status 'error'Looks like the plugin was already installed. Forcing a
reinstall./usr/bin/python3 -m pip --disable-pip-version-check install
file:///tmp/tmp4_bu4f5m.zip --no-cache-dir --ignore-installed
--force-reinstall --no-deps --userProcessing
/tmp/tmp4_bu4f5m.zipPreparing metadata (setup.py): started

BaTaTaAdb, 2023-09-01:
@gleneccles Are you sure you installed plugin extras? 
[image]

That is a requirement for the octolapse to work and It was really hard for me to find it when I was having issues.

guywholikestobreakthings, 2023-09-01:
Yup, Mine is on.

gleneccles, 2023-09-03:
Well, it worked @BaTaTaAdb =)

Don't try to upgrade it though lol. It will fail the same way.

BaTaTaAdb, 2023-09-04:
Thanks for heads up @gleneccles ! And I'm glad I could help :)