feelfreelinux/octo4a #374: Is it possible to install TailScale?

Androxilogin, 2023-02-09:
Been using this on my Pi Zero the past few nights and was hoping this was an option but all I get is the error:

+ rc-update add tailscale
sh: rc-update: not found

lmichoux, 2023-02-18:
If you need a specific tool you can install it using the Alpine package manager.
Octo4a is running on alpine linux distribution (https://docs.alpinelinux.org)
Here is the command to install tailscale:
#  apk  add tailscale

alufers, 2023-03-21:
Instaling tailscae through the built in shell is not recommended (I don't think it will even work, as it needs to create a TUN device).

You should use this on Android:  https://play.google.com/store/apps/details?id=com.tailscale.ipn&hl=pl&gl=IT

closed 2023-03-21 by alufers

Androxilogin, 2023-07-31:
> Instaling tailscae through the built in shell is not recommended (I don't think it will even work, as it needs to create a TUN device).
> 
> You should use this on Android: https://play.google.com/store/apps/details?id=com.tailscale.ipn&hl=pl&gl=IT

That's only compatible with certain versions of Android.