feelfreelinux/octo4a #375: The app works fine after initial hiccups regarding installation, but what about the charging while using it.

sanjiv204, 2023-02-10:
I have been able to install this app and connect my printer ender 3 v2 neo to the app.
However the phone which i use has micro usb charging port. While the phone is connected to the printer there is no way to charge the phone, which is so necessary because the battery drains fast.
What are the available solutions for the charging while connecting the phone to the printer?

sv0o, 2023-02-11:
what kind of phone are you using?  Some phones will power off USB without a battery.  Some can be tricked to do so under the right conditions.  Some will work if you solder plus and minus leads directly to the battery terminals or the tabs that normally connect to a wireless charging coil.  Wireless charging might be an option if your phone doesn't disable the USB port when it detects a wireless charger is connected.  There are a gazillion micro-usb OTG devices and Y cables and this one sort of works for me 75% of the time: https://www.ebay.com/itm/233154413526

sanjiv204, 2023-02-12:
The phone i am using is a discard old phone with micro USB port. I tried one of the y cables but either it provides charging or OTG depending upon the port phone is connected to.
I will try one of the cables suggested and add information later.

sv0o, 2023-02-12:
I understand how frustrating it can be. the LG phone I've been using didn't cost me anything but it fails, hangs or looses power without warning. Good Luck.

Sausages69, 2023-03-06:
Hi guys,
To charge while you use the phone hooked up to a printer you will need an OTC cable. Or easier to find, a dock that does pass-through charging as well. I use a Satechi branded one like this one. https://www.jbhifi.com.au/products/satechi-usb-c-slim-multiport-adapter-v1-black Pass through charing is the key thing here.

alufers, 2023-03-21:
Hi, Thanks for trying octo4a!
Unfortunately some phones just don't support charging and using OTG at the same time :/
This is a known problem, and unfortunately it is near-impossible to solve on many phones.

We maintain a crowd-sourced compatibility list of phones which work with octo4a:
The list: https://docs.google.com/spreadsheets/d/1pbKAELzvTrMVm6NpDpK8TfQR33w3QHQEgqkcGdbPFI0/edit?usp=sharing
Feel free to submit your phone here: https://forms.gle/UFzJebj67hCtLCBz7 / /

Sometimes it is possible to bypass this problem by trying these solutions:

- Using wireless charging. If your phone supports it, it might be possible to use a wireless charger and use the OTG feature at the same time. Some users have resorted to gluing a wireless charger to a phone case
  [image]
  
- Hard-wiring the battery. 
  **WARNING: CONNECTING A RANDOM POWER-SUPPLY STRAIGHT TO THE BATTERY MIGHT RESULT IN A FIRE OR EXPLOSION, EXERCISE EXTREME CAUTION** 
   Some people modify their phone to charge their battery externally, bypassing the phone's battery charging circuitry altogether. Some event removed the battery (the safest option)
   [image]
- Compiling a non-standard kernel. On some models the charging-and-OTG functionality is blocked in software, and this can be bypassed by using a non-stock kernel. Please be advised that compiling an Android kernel requires some skill and a lot of time.

If you need some guidance please feel free to ask on our telegram group: (https://t.me/octo4achat)[https://t.me/octo4achat]. The community is very likely to help you.

closed 2023-03-21 by alufers