feelfreelinux/octo4a #379: Corrupt Installation - Android 11

jccraig26, 2023-02-14:
I think this is a known issue, but I wanted to log it to be sure.  I am trying to install on a BLU View 2 (2022) 4G LTE.  It comes with Android 11 on it.  Even using 1.2 I am still getting the corrupt installation issue.  I am also unable to do the SSH workaround since Octoprint never seems to finish starting so SSH is not available.  

I will continue trying workarounds until a fix for Android 11 is available.

9cento, 2023-02-25:
> I think this is a known issue, but I wanted to log it to be sure. I am trying to install on a BLU View 2 (2022) 4G LTE. It comes with Android 11 on it. Even using 1.2 I am still getting the corrupt [...]

Did you find any workarounds?

real-artswan, 2023-02-28:
I have Samsung A10s running android 11 kernel 4.19. Here is what I've found
```
1|a10s:/data/user/0/com.octo4a/files/bootstrap $ ls -la bootstrap                                                                                                                                                                          
total 152
drwx------ 19 u0_a227 u0_a227 4096 2023-02-28 16:51 .
drwxrwxrwx  6 u0_a227 u0_a227 4096 2023-02-28 16:35 ..
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 bin
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 dev
drwx------ 17 u0_a227 u0_a227 4096 2023-02-28 16:24 etc
drwx------  2 u0_a227 u0_a227 4096 2023-02-28 16:51 home
drwx------  7 u0_a227 u0_a227 4096 2023-01-09 14:47 lib
drwx------  5 u0_a227 u0_a227 4096 2023-01-09 14:47 media
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 mnt
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 opt
dr-x------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 proc
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 root
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 run
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 sbin
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 srv
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 sys
drwx------  2 u0_a227 u0_a227 4096 2023-01-09 14:47 tmp
drwx------  7 u0_a227 u0_a227 4096 2023-01-09 14:47 usr
drwx------ 12 u0_a227 u0_a227 4096 2023-01-09 14:47 var
a10s:/data/user/0/com.octo4a/files/bootstrap $ PATH='/sbin:/usr/sbin:/bin:/usr/bin' ./root/bin/proot -r ./bootstrap -0 -w / --link2symlink -b /data/data/com.octo4a/files/bootstrap/bootstrap/sbin:/sbin -L /bin/sh -c "ls -la /"          
ls: /home: No such file or directory
ls: /tmp: No such file or directory
ls: /usr: No such file or directory
ls: /opt: No such file or directory
ls: /lib: No such file or directory
ls: /srv: No such file or directory
ls: /run: No such file or directory
ls: /root: No such file or directory
ls: /var: No such file or directory
ls: /media: No such file or directory
ls: /sbin: No such file or directory
total 8
drwxr-xr-x   25 root     root          4096 Dec 31  2008 .
drwxr-xr-x   25 root     root          4096 Dec 31  2008 ..
ls: /bin: cannot read link: not a symlink
lrw-r--r--    1 root     root            11 Dec 31  2008 bin
drwxr-xr-x   25 root     root          3900 Feb 27 20:30 dev
ls: /etc: cannot read link: not a symlink
lrw-r--r--    1 root     root            11 Dec 31  2008 etc
drwxr-xr-x   17 root     1000           360 Feb 27 20:30 mnt
dr-xr-xr-x  533 root     root             0 Jan  1  1970 proc
dr-xr-xr-x   14 root     root             0 Feb 27 20:30 sys
1|a10s:/data/user/0/com.octo4a/files/bootstrap $
```
As you can see proot doesn't mount many directories (btw, apk is inside /sbin), have no idea why it happens and what is the difference between e.g. mounted /bin and not mounted /sbin

real-artswan, 2023-03-02:
May some of devs tell why it uses exact that proot fork? Does vanilla proot has any issues? I'd like to take a try to use it, but will I just waste my time?

alufers, 2023-03-25:
Hi
Could you try it now? I have updated proot in the bootstrap and it should work on Android 11.

9cento, 2023-03-25:
> Hi Could you try it now? I have updated proot in the bootstrap and it should work on Android 11.

Currently my Android 11 device is soft-bricked but I'll give it a try nonetheless on some emulators, I respect your hard work and thank you!

9cento, 2023-03-25:
@alufers I can confirm that it went smooth on Android Studio's emulator (where the previous version used to crash) and I'm expecting that it will work flawlessly also on an actual physical device, so congrats on this! It will open the door to a lot more devices so don't forget to spread the word! 💚🐙💚

alufers, 2023-03-25:
I've republished that version, because of course I forgot to bump the version of the APK...

But anyway, please do let me know if it works. I've gotten a report from a user with a Tracfone BLU VIew 2 that it doesn't so I'm curious if it works on any others.

alufers, 2023-03-28:
Should be fixed in 1.2.4.

closed 2023-03-28 by alufers

jccraig26, 2023-03-28:
THANK YOU!!!!  It worked!

Sorry, I was out of town for a bit.  I just installed it and it works.

Thank again.

9cento, 2023-03-28:
> I've republished that version, because of course I forgot to bump the version of the APK...
> 
> But anyway, please do let me know if it works. I've gotten a report from a user with a Tracfone BLU VIew 2 that it doesn't so I'm curious if it works on any others.

I can confirm it also works on a Tanix X4 (TV Box) with Android 11 (AOSP, non-ATV). Good Job! 💪🏻