feelfreelinux/octo4a #380: v1.20 build info shows v1.1.2

cummingsrob, 2023-02-19:
Clicking the info circle button in app v1.2.0 shows that it is build 1.1.2.

Changelog for 1.2.0 includes "Fixed bug where the installation gets corrupted on first install". Could 1.2.0 still be including 1.1.2 code in the apk, leading to the corruption everyone is experiencing?

IzzySoft, 2023-02-20:
It not only shows 1.1.2, the APK is identical to the one attached to v1.1.2 – just with a newer timestamp. Due to that, my updater seeing a newer tag downloads the APK over and again each day just to throw it away as it's already here. So I'll have to disable automated updates for know.

@feelfreelinux could you please fix that, replacing the APK by the correct one and give us a ping here? Thanks in advance!

martinstoehr, 2023-03-02:
same here, seems to be the old version.

MeAgainAndAgain, 2023-03-13:
If you compare ths two apk's they are Not the same. They might be the same size but the contents are different.

AbbieDoobie, 2023-03-13:
Same issue here. It isn't causing any critical errors, but it is an issue none the less.

IzzySoft, 2023-03-13:
@martinstoehr my updater complained they are identical:

    2023-02-19 19:28:39 local WARNING  repo/com.octo4a_1001002.apk already exists and is identical to the one we downloaded; ignoring it

which means they are binary-identical. Otherwise the warning message would have been different, like in this other case:

    2023-02-27 19:29:37 local WARNING  repo/com.kaiwolfram.nozzle_7.apk already exists, quarantining before replacing it

So it's not just the same size. To be clear, `diff` stated they were identical – so yes, they have been compared. It's very unlikely that two binary identical APK files differ in content – so unlikely that it can be practically ruled out.

MeAgainAndAgain, 2023-03-13:
The files look different to me, There are quite a few different files that mostly appear to be config, for example, the strings relating to the camera server which seem evident by the way it behaves between the two apk's

[image]
[image]

IzzySoft, 2023-03-13:
Ah, wait, yes: As the versionName does not match the tag name, my updater downloaded the same file over and again (as the tag indicated a version newer than available locally here). So the second time it compared it to the very same file downloaded the day before – which of course is identical. Silly me. This went on for too long (logfile rotated since so I didn't see the initial culprit). A pity @feelfreelinux did not yet find time to chime in; so probably it was just the versions not increased before creating the release. So much confusion for just a few simple numbers…

alufers, 2023-03-21:
FIxed in 1.2.1.

Sorry for that, first time doing a release on this project :((((

closed 2023-03-21 by alufers

IzzySoft, 2023-03-21:
@alufers no prob, was my fault as well (see above). We're all constantly learning :wink:

> Whoever ceases to be a student has never been a student.

(George Iles)