feelfreelinux/octo4a #381: installation corrupted-Followed #325(comment)  but got error at step 8

QT7901, 2023-02-20:
My phone is samsung SM-G531H. I tried installing all versions but only successfully installed 0.0.5. 
I followed [https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359](url) but the installation in step 7-8 failed again
[image]

[image]

I hope to receive your help, my English is not good and I am using google to translate.

RedmodSK, 2023-02-26:
I am having the exact same error on two phones. Huawei P9 lite and Huawei Mate 8 - both running Anroid 7
Both are after factory reset. 
I have tried all i can but i am unable to resolve this.  
Also the web ssh seems to be running but i am unable to authenticate. It works with putty on port 8022 however.
I would welcome any suggestion.
[octo4a.log](https://github.com/feelfreelinux/octo4a/files/10834361/octo4a.log)

joejtrsmoe, 2023-02-27:
I love this software to recycle old phone. Wish I could get to work reliably on my galaxy s5 but wont event work after newest octoprint update. Was able to use issue #325 fix before but not now on galaxy s5. Was hoping permissions got fixed for unrooted s5 but seems to have gotten worse.
[image]

tidus9000, 2023-03-03:
I'm also having this exact same error on my Galaxy S7 Edge SM-G935F. I found a potential solution of changing some of the code and building but...I'm not used to the android dev environment and have absolutley no idea how to build it

PrintsLucas, 2023-03-16:
Same issue on Galaxy S6 Active SMG890A, Android 7.0

millerp6ATWIT, 2023-06-08:
I'm having this exact same issue on android 7.0 on an LG V20 and have tried all these same steps numerous times. Any way to fix it?

adampudliner, 2023-07-17:
Getting the same issue on moto razer running android 8. Like @tidus9000, I have tried many suggested solutions.

millerp6ATWIT, 2023-07-20:
> Getting the same issue on moto razer running android 8. Like @tidus9000, I have tried many suggested solutions.

If it helps, I solved my problem by updating from android 7 to 8. Sounds simple but I had to reflash it with a canadian firmware and OS so that it would update to android 8 since it wasn't released for US phones. After this it installed like normal. So maybe see if you can get a newer version of android on the phone, or downgrade to an older one.