feelfreelinux/octo4a #382: Camera on status not loading on phone startup

sharpie7, 2023-02-22:
Thanks for a great project - very useful and seems to run much better on a 2nd hand phone than on a Pi3.

One problem I have with version 1.2.0 is that the "camera on" status doesn't seem to load correctly when the phone starts up.

I have Octo4a set to load on power-on. The server starts fine, but shows "Camera Server Disabled" even though it was previously on.

Opening settings shows "Enable camera server on". If I flick this off and then on again then the camera server starts.

Phone is Sony XZ F8331 running Android 8.0.0

joejtrsmoe, 2023-02-27:
Same problem here. With galaxy s7.

brotherchris, 2023-03-09:
Same problem here, did anyone find a fix?

MeAgainAndAgain, 2023-03-13:
Solved (I think), I had the same problem with a  Sony Xperia XZ1 Premium and Samsung XCover 4s, After loading 1.2.0 did not start on boot and the Camera server not started until toggled settings. After a bit of playing, I did this- 
DO NOT Uninstall 1.2.0 but while it is running, run the apk for 1.1.2, It will NOT reinstall Octoprint just Octo4a but somehow fixes the problem (at least it did on my phones). When you open Octo4a it should still be there. Difficult to say what this has done as both 1.2.0 and 1.1.2 report as 1.1.2. I did this before installing plugins from a backup so might need to reload plugins after. Make sure you have an Octoprint Backup to save you time with this.  Hopefully, Plugins will be updated soon to work with OP 1.8.7.

closed 2023-03-21 by alufers

sharpie7, 2023-03-21:
Thanks . I am waiting for a new phone to use for Octo4a. I will check the new version out once I have that.

commit cdf7e20a (2023-03-21): FIx camera server not starting automatically (fixes #382)