feelfreelinux/octo4a #386: Phone drops connection to printer when charger plugged in

jahnj0584, 2023-03-05:
I got this hub which many people are using properly

https://www.amazon.ca/Adapter-Female-Charging-Splitter-Compatible/dp/B08KPD5S82/?_encoding=UTF8&pd_rd_w=ZB7Dh&content-id=amzn1.sym.72b03f0f-4e11-4b99-b598-5b2869715532&pf_rd_p=72b03f0f-4e11-4b99-b598-5b2869715532&pf_rd_r=8DWM3W2A55H9CF79SA9D&pd_rd_wg=mYcZU&pd_rd_r=24469e18-5ae4-4bfc-8b4c-27c8afdb7054&ref_=pd_gw_ci_mcx_mr_hp_atf_m

When the phone is only connected to the printer, it is connected and I have run and sent many prints with it. Once I plug in a USB C to charge it (I've tried 4 cables and 3 wall adapters) it automatically disconnects. The instant I unplug the C, it resumes the connection to the printer.

Sausages69, 2023-03-06:
The hub probably doesn't do pass thru charging.

jahnj0584, 2023-03-06:
Someone says this is an Huwei issue with setting it to Reverse Charge at all times.

sharpie7, 2023-03-18:
I don't think this is an issue with octo4a. It's more a practical problem with how (some?) Android phones use the USB port. I have a similar issue with my Sony F8331 Xperia XZ. It will charge though the USB hub on its own, or it will connect to an external USB device though the hub, but not both. Seems to be a limitation in the phone as far as I can tell.

sharpie7, 2023-03-18:
Further study on my Sony F8331 Xperia XZ:

If you connect a device to the USB socket it doesn't send any power to the external device until you go to settings->Device connection->USB Connectivity->Detect USB Device.

After that the external device will get power and start communicating with the phone.

So, it seems that the phone changes the power state of its USB port depending on whether you've asked it to connect to an external device or not. Therefore, I guess it's a fundamental limitation of the phone that it can only use the USB port as a charging source, or as a USB output but not both.

Maybe we could try and build a list of which phones do/don't support using the USB as both a charge source and a USB output.

sharpie7, 2023-03-18:
Actually, this information could be usefully added to the wiki, e.g. on [this page](https://github.com/feelfreelinux/octo4a/wiki/Connecting-to-printer-via-OTG,-simultaneous-charging). It seems that there isn't a way for the public to directly propose modifications to the Wiki though.

alufers, 2023-03-23:
//
Hi, Thanks for trying octo4a!
Unfortunately some phones just don't support charging and using OTG at the same time :/
This is a known problem, and unfortunately it is near-impossible to solve on many phones.

We maintain a crowd-sourced compatibility list of phones which work with octo4a:
The list: https://docs.google.com/spreadsheets/d/1pbKAELzvTrMVm6NpDpK8TfQR33w3QHQEgqkcGdbPFI0/edit?usp=sharing
Feel free to submit your phone here: https://forms.gle/UFzJebj67hCtLCBz7

Sometimes it is possible to bypass this problem by trying these solutions:

- Using wireless charging. If your phone supports it, it might be possible to use a wireless charger and use the OTG feature at the same time. Some users have resorted to gluing a wireless charger to a phone case
  [image]
  
- Hard-wiring the battery. 
  **WARNING: CONNECTING A RANDOM POWER-SUPPLY STRAIGHT TO THE BATTERY MIGHT RESULT IN A FIRE OR EXPLOSION, EXERCISE EXTREME CAUTION** 
   Some people modify their phone to charge their battery externally, bypassing the phone's battery charging circuitry altogether. Some event removed the battery (the safest option)
   [image]
- Compiling a non-standard kernel. On some models the charging-and-OTG functionality is blocked in software, and this can be bypassed by using a non-stock kernel. Please be advised that compiling an Android kernel requires some skill and a lot of time.

If you need some guidance please feel free to ask on our telegram group: (https://t.me/octo4achat)[https://t.me/octo4achat]. The community is very likely to help you.