feelfreelinux/octo4a #393: 1.2.3 broke installation

BarniRamirez, 2023-03-25:
yesterday the 1.2.2 was working on my device, today i decided to unistall and reinstall the app and now it says "installation corrupted". 
Is there a way to reinstall the old version?

BarniRamirez, 2023-03-25:
I reinstalled the old .apk (octo4a-1.2.2.apk) "installing bootstrap (arm)" is ok, but when it gets to 
"downloading octoprint (newest)..." the app crashes and the "installation corrupted" message appears

alufers, 2023-03-25:
What phone are you using?

BarniRamirez, 2023-03-25:
I'm using a samsung galaxy tab 3 (SM-T310) with LineageOS-14.1, Android 7.1.2

alufers, 2023-03-25:
Okay, I think it's an issue with 32-bit processors (armhf). I am currently testing a fix for this. A new release should be ready in a moment.

BarniRamirez, 2023-03-25:
Okay ty! I'll try with the new version.

BarniRamirez, 2023-03-25:
[image]
By the way this is the last thing it shows in the log right before it crashes ...

alufers, 2023-03-25:
Okay, I've releases 1.2.4. Could you check it?

BarniRamirez, 2023-03-25:
Okay now with the 1.2.4 it works again but (as with the 1.2.2) when i try to start the camera server the app crashes and tablet restarts.
However I solved the problem by keeping the camera permission in octoprint off and installing "ip webcam" (https://play.google.com/store/apps/details?id=com.pas.webcam&hl=it&gl=US) and running it simultaneously, then all I had to do was pasting the stream URL of the webcam app in the octoprint UI in settings > webcam & timelapse.

thank you for the support!

alufers, 2023-03-25:
Okay, glad it works for you. I will try to improve the camera server next.

closed 2023-03-25 by alufers

shijian321, 2023-11-25:
@alufers Can you integrate all the installation files into APK