feelfreelinux/octo4a #394: Failed to build PyYAML tornado

inv4dor, 2023-03-25:
🐸 Building wheels for collected packages: PyYAML, tornado
🐸   Building wheel for PyYAML (pyproject.toml): started
🐸   Building wheel for PyYAML (pyproject.toml): finished with status 'done'
🐸   WARNING: Building wheel for PyYAML failed: [Errno 1] Operation not permitted
🐸   Building wheel for tornado (pyproject.toml): started
🐸   Building wheel for tornado (pyproject.toml): finished with status 'done'
🐸   WARNING: Building wheel for tornado failed: [Errno 1] Operation not permitted
🐸 Failed to build PyYAML tornado
🐸 ERROR: Could not build wheels for PyYAML, tornado, which is required to install pyproject.toml-based projects
💻 An exception has occurred at:
java.lang.RuntimeException: Process exited with error code 1. [Processing /root/OctoPrint-OctoPrint-8e4c1f5  Pre
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:38)
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:12)
	at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:164)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
	at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [Processing /root/OctoPrint-OctoPrint-8e4c1f5  Pre

[PyYAML - No bootstrap.txt](https://github.com/feelfreelinux/octo4a/files/11070358/PyYAML.-.No.bootstrap.txt)

alufers, 2023-03-26:
I see. I will try to create a fix for that.

Mitchelb22, 2023-04-06:
I have the same problem, hopefully there is a fix for that

Mitchelb22, 2023-04-06:
> 

Do u have solved this problem this moment?

Hyyybris, 2023-04-13:
I have the exact same problem, except for the word Tornado. Trying to install on a Huawei Honor 9 running Nougat. Seems to blow up in the same way, but the aftermath of the failure to build wheels seems somewhat different. Hoping a fix for this will solve my problem as well.

RPBOHSH, 2023-12-06:
Also having the same issue with this.

SpacetravelDEV, 2023-12-14:
I also have this issue

MATZE-MAN, 2023-12-23:
Same on my old Android 7 Utopia Tablett