feelfreelinux/octo4a #395: 1.2.4 exception when installing octoprint

sharpie7, 2023-03-26:
Installing octoprint dependencies fails with exception with v1.2.4 on galaxy s5  . See log extract below . Note failure to install wheel which might be related 

━━━━━━━━━━━━━━━━━━━━━━━━━ 237.5/237.5 kB 1.7 MB/s eta 0:00:00
🐸 Collecting typing-extensions>=4
🐸   Downloading typing_extensions-4.5.0-py3-none-any.whl (27 kB)
🐸 Collecting urllib3<1.27,>=1.21.1
🐸   Downloading urllib3-1.26.15-py2.py3-none-any.whl (140 kB)
🐸      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 140.9/140.9 kB 1.4 MB/s eta 0:00:00
🐸 Collecting certifi>=2017.4.17
🐸   Downloading certifi-2022.12.7-py3-none-any.whl (155 kB)
🐸      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 155.3/155.3 kB 1.6 MB/s eta 0:00:00
🐸 Collecting idna<4,>=2.5
🐸   Downloading idna-3.4-py3-none-any.whl (61 kB)
🐸      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.5/61.5 kB 102.4 kB/s eta 0:00:00
🐸 Collecting charset-normalizer<4,>=2
🐸   Downloading charset_normalizer-3.1.0-py3-none-any.whl (46 kB)
🐸      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.2/46.2 kB 664.7 kB/s eta 0:00:00
🐸 Collecting ifaddr>=0.1.7
🐸   Downloading ifaddr-0.2.0-py3-none-any.whl (12 kB)
🐸 Collecting MarkupSafe>=2.0
🐸   Downloading MarkupSafe-2.1.2.tar.gz (19 kB)
🐸   Preparing metadata (setup.py): started
🐸   Preparing metadata (setup.py): finished with status 'done'
🐸 Collecting deprecated>=1.2
🐸   Downloading Deprecated-1.2.13-py2.py3-none-any.whl (9.6 kB)
🐸 Requirement already satisfied: setuptools in /usr/lib/python3.10/site-packages (from limits>=2.8->Flask-Limiter<3,>=2.6->OctoPrint==1.8.7) (65.6.0)
🐸 Requirement already satisfied: packaging<24,>=21 in /usr/lib/python3.10/site-packages (from limits>=2.8->Flask-Limiter<3,>=2.6->OctoPrint==1.8.7) (23.0)
🐸 Collecting importlib-resources>=1.3
🐸   Downloading importlib_resources-5.12.0-py3-none-any.whl (36 kB)
🐸 Collecting commonmark<0.10.0,>=0.9.0
🐸   Downloading commonmark-0.9.1-py2.py3-none-any.whl (51 kB)
🐸      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 51.1/51.1 kB 944.6 kB/s eta 0:00:00
🐸 Collecting pygments<3.0.0,>=2.6.0
🐸   Downloading Pygments-2.14.0-py3-none-any.whl (1.1 MB)
🐸      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 2.6 MB/s eta 0:00:00
🐸 Building wheels for collected packages: PyYAML, tornado
🐸   Building wheel for PyYAML (pyproject.toml): started
🐸   Building wheel for PyYAML (pyproject.toml): finished with status 'done'
🐸   WARNING: Building wheel for PyYAML failed: [Errno 1] Operation not permitted
🐸   Building wheel for tornado (pyproject.toml): started
🐸   Building wheel for tornado (pyproject.toml): finished with status 'done'
🐸   WARNING: Building wheel for tornado failed: [Errno 1] Operation not permitted
🐸 Failed to build PyYAML tornado
🐸 ERROR: Could not build wheels for PyYAML, tornado, which is required to install pyproject.toml-based projects
💻 An exception has occurred at:
java.lang.RuntimeException: Process exited with error code 1. [Processing /root/OctoPrint-OctoPrint-8e4c1f5  Pre
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:38)
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:12)
	at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:164)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
	at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [Processing /root/OctoPrint-OctoPrint-8e4c1f5  Pre

dynalogix, 2023-03-29:
Install stops for me at the same point installing dependencies on Galaxy Note 3 (on Android 4.3)

🐸 Building wheels for collected packages: PyYAML, tornado
🐸 Building wheel for PyYAML (pyproject.toml): started
🐸 Building wheel for PyYAML (pyproject.toml): finished with status 'done'
🐸 WARNING: Building wheel for PyYAML failed: [Errno 1] Operation not permitted
🐸 Building wheel for tornado (pyproject.toml): started
🐸 Building wheel for tornado (pyproject.toml): finished with status 'done'
🐸 WARNING: Building wheel for tornado failed: [Errno 1] Operation not permitted
🐸 Failed to build PyYAML tornado
🐸 ERROR: Could not build wheels for PyYAML, tornado, which is required to install pyproject.toml-based projects

cache4pat, 2023-07-13:
It stopped me to; attempting install of 1.2.5 .... See this Log
Octo4a install Log .....
=================

💻 No bootstrap detected, proceeding with installation
💻 Arch: arm
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-armv7a.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/2cd65898-00b6-4b5a-bd0f-7a036eff0c4f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230713%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230713T123222Z&X-Amz-Expires=300&X-Amz-Signature=ffca598396d64dfd0ad3275759f0ff4275a4bc1257ec06c91e1b636aab934495&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-armv7a.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap extracted, setting it up...
💻 >ls
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 root-pre5
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
💻 >chmod -R 700 .
💻 >sh install-bootstrap.sh
🐸 chmod: chmod 'serialpipe' to 120777: Permission denied
🐸 chmod: chmod 'serialpipe' to 120777: Permission denied
🐸 bootstrap ready, run with run-bootstrap.sh
💻 >sh add-user.sh octoprint
💻 >cat /etc/motd
🐸 Welcome to Alpine!
🐸
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <https://wiki.alpinelinux.org/>.
🐸
🐸 You can setup the system with the command: setup-alpine
🐸
🐸 You may change this message by editing /etc/motd.
🐸
💻 >env
🐸 EXTRA_BIND=-b /storage/emulated/0/Android/data/com.octo4a/files:/root -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a -b /data/data/com.octo4a/files/bootstrap/ioctlHook.so:/home/octoprint/ioctlHook.so
🐸 MC_AUTH_TOKEN_PATH=/efs
🐸 USER=root
🐸 KNOX_STORAGE=/data/knox/ext_sdcard
🐸 EXTERNAL_STORAGE=/sdcard
🐸 PROOT_L2S_DIR=/data/data/com.octo4a/files/bootstrap/bootstrap/.proot.meta
🐸 SHLVL=1
🐸 HOME=/root
🐸 ANDROID_ASSETS=/system/app
🐸 BOOTCLASSPATH=/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/sec_edm.jar:/system/framework/sagearpolicymanager.jar:/system/framework/secocsp.jar:/system/framework/timakeystore.jar:/system/framework/commonimsinterface.jar:/system/framework/imsmanager.jar:/system/framework/sprengine.jar:/system/framework/smartbondingservice.jar:/system/framework/knoxvpnuidtag.jar:/system/framework/sec_sdp_sdk.jar:/system/framework/sec_sdp_hidden_sdk.jar:/system/framework/simageis.jar:/system/framework/tcmiface.jar:/system/framework/qcmediaplayer.jar:/system/framework/qcom.fmradio.jar:/system/framework/oem-services.jar:/system/framework/com.qti.dpmframework.jar:/system/framework/dpmapi.jar:/system/framework/com.qti.location.sdk.jar
🐸 _=./root/bin/proot
🐸 ANDROID_DATA=/data
🐸 PATH=/sbin:/usr/sbin:/bin:/usr/bin
🐸 ANDROID_ROOT=/system
🐸 SECONDARY_STORAGE=/storage/extSdCard
🐸 LANG='en_US.UTF-8'
🐸 ENC_EMULATED_STORAGE_TARGET=/storage/enc_emulated
🐸 ANDROID_SOCKET_zygote=10
🐸 ASEC_MOUNTPOINT=/mnt/asec
🐸 PWD=/root
🐸 ANDROID_PROPERTY_WORKSPACE=9,0
🐸 SYSTEMSERVERCLASSPATH=/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar
🐸 ANDROID_STORAGE=/storage
🐸 PROOT_TMP_DIR=/data/data/com.octo4a/files/bootstrap/tmp
🐸 ANDROID_BOOTLOGO=1
💻 >ls /
🐸 bin
🐸 dev
🐸 etc
🐸 home
🐸 lib
🐸 media
🐸 mnt
🐸 opt
[... 1059 lines cut ...]
at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:43)
at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:16)
at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:167)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [Processing /root/OctoPrint-OctoPrint-1d519a2  Ins