feelfreelinux/octo4a #396: Displaying over other apps bug

JakubKermes, 2023-03-27:
On Android 6.0.1 and Sony Xperia Z3, the permissions window is seen as an overlay, which makes installation impossible. You have to go to settings and manually set permissions before opening the app. Then, the OctoPrint installation is successful.

sharpie7, 2023-03-28:
Yes, similar problem on a galaxy s5. The OctoPrint installer first asks for permission to access photos (IIRC) and then puts up an overlay permissions dialog even if the overlay permissions are already granted.

It seems to be a look at this point, but if you retry several times it eventually goes past the overlay permissions step though I am not sure what is different to allow it to progress.

alufers, 2023-03-28:
Could you post a screenshot of how it looks?

sharpie7, 2023-03-28:
Low-tech (because "reasons") photos below (in sequence). 

At least on my device it seems if you respond "slowly" to the "Allow to access photos...." dialog then you don't get the "overlay" one, but if you respond quickly then you do get this dialog.

As you can see, the "overlay" dialog appears even if the permission is already granted. There doesn't seem to be a way to go back from the settings screen to continue the installation. It goes back to the start again.

[image]
[image]
[image]
[image]
[image]

JakubKermes, 2023-03-28:
Exactly the same, the problem seems to be not exactly granting permission, but window redirecting to settings displaying over