feelfreelinux/octo4a #400: Fatal error when starting

patrickmac110, 2023-03-31:
🔌 Got extension Remote web terminal (ttyd)
🐙 2023-03-31 23:16:45,363 - octoprint.startup - INFO - ******************************************************************************[0m
🐙 2023-03-31 23:16:45,365 - octoprint.startup - INFO - Starting OctoPrint 1.8.2[0m
🐙 2023-03-31 23:16:45,373 - octoprint.startup - INFO - ******************************************************************************[0m
🐙 [1;31m2023-03-31 23:16:45,724 - octoprint.startup - CRITICAL - Could not initialize event manager: Invalid version: '/usr/lib/python3.9/site'[0m
🐙 Could not initialize event manager: Invalid version: '/usr/lib/python3.9/site'
🐙 [1;31m2023-03-31 23:16:45,725 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.[0m
🐙 There was a fatal error starting up OctoPrint.

jasta, 2023-12-21:
I'm getting the same, were you able to find a solution?

eredwan, 2024-01-27:
I am having this same problem after an update from 1.1.1 to the latest that is out now. I knew i shouldnt have updated! now im worried about my Octoprint configs and settings. How do I make sure my octoprint data does not go away if I reinstall?

eredwan, 2024-01-27:
I fixed it! In my case, I updated Octo4A from 1.1.1 to 1.2.5. After the update was finished, I attempted to run Octo4A again and it gave me this error in the logs.

Even though I had this error and OctoPrint wont start. I still could connect to the bash feature (which i guess is from ttyd) and that was running fine.
Connect to your tablet/phone using the bash info as seen in the Octo4A config settings. 
In my case, it was the OctoPrintIP:5002. If you dont know what that means, go to your web browser and put in the ip and port in this example format: 192.168.0.0:5002

then simply run: pip3 install --upgrade setuptools wheel tornado --no-cache-dir

and thats it! go back to your tablet/phone and force close/stop octo4a and run it again and everything should work fine.