feelfreelinux/octo4a #401: Galaxy s6 a7.0 Exception: java.lang.RuntimeException: Process exited with error code 1. [proot warning: can't sanitize binding "/data/data

GeizerG, 2023-04-01:
💻 No bootstrap detected, proceeding with installation
💻 Arch: aarch64
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-aarch64.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/ea14dee1-fef5-465a-9f22-4ff9e7631821?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230331%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230331T234742Z&X-Amz-Expires=300&X-Amz-Signature=c334ffae1c01eb0c1b665c1dc25c00c6235b24b5c2859b2efbcd1781c20a319c&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-aarch64.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap extracted, setting it up...
💻 >ls
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
💻 >chmod -R 700 .
💻 >sh install-bootstrap.sh
🐸 bootstrap ready, run with run-bootstrap.sh
💻 >sh add-user.sh octoprint
💻 >cat /etc/motd
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Welcome to Alpine!
🐸 
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <https://wiki.alpinelinux.org/>.
🐸 
🐸 You can setup the system with the command: setup-alpine
🐸 
🐸 You may change this message by editing /etc/motd.
🐸 
💻 >env
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 EXTRA_BIND=-b /storage/emulated/0/Android/data/com.octo4a/files:/root -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a -b /data/data/com.octo4a/files/bootstrap/ioctlHook.so:/home/octoprint/ioctlHook.so
🐸 MC_AUTH_TOKEN_PATH=/efs
🐸 USER=root
🐸 EXTERNAL_STORAGE=/sdcard
🐸 KNOX_STORAGE=/data/knox/ext_sdcard
🐸 PROOT_L2S_DIR=/data/data/com.octo4a/files/bootstrap/bootstrap/.proot.meta
🐸 SHLVL=1
🐸 HOME=/root
🐸 ANDROID_ASSETS=/system/app
🐸 BOOTCLASSPATH=/system/framework/core-oj.jar:/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/sprengine.jar:/system/framework/knoxsdk.jar:/system/framework/sec_edm.jar:/system/framework/sagearpolicymanager.jar:/system/framework/smartbondingservice.jar:/system/framework/timakeystore.jar:/system/framework/fipstimakeystore.jar:/system/framework/knoxvpnuidtag.jar:/system/framework/sec_sdp_sdk.jar:/system/framework/sec_sdp_hidden_sdk.jar:/system/framework/esecomm.jar:/system/framework/SemAudioThumbnail.jar:/system/framework/smatlib.jar:/system/framework/samsung-framework.jar
🐸 ANDROID_SOCKET_zygote_agent64=8
🐸 _=./root/bin/proot
🐸 ANDROID_DATA=/data
🐸 PATH=/sbin:/usr/sbin:/bin:/usr/bin
🐸 ANDROID_ROOT=/system
🐸 LANG='en_US.UTF-8'
🐸 ENC_EMULATED_STORAGE_TARGET=/storage/enc_emulated
🐸 ASEC_MOUNTPOINT=/mnt/asec
🐸 PWD=/root
🐸 SYSTEMSERVERCLASSPATH=/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar:/system/framework/samsung-services.jar:/system/framework/ssrm.jar:/system/framework/hqm.jar
🐸 ANDROID_STORAGE=/storage
🐸 PROOT_TMP_DIR=/data/data/com.octo4a/files/bootstrap/tmp
🐸 ANDROID_BOOTLOGO=1
💻 >ls /
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 bin
🐸 dev
🐸 etc
🐸 home
🐸 lib
🐸 media
🐸 mnt
🐸 opt
🐸 proc
🐸 root
🐸 run
🐸 sbin
🐸 srv
🐸 storage
[... 399 lines cut ...]
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:38)
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:12)
	at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:164)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
	at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [proot warning: can't sanitize binding "/data/data

dynalogix, 2023-04-01:
It's the same "ERROR: Could not build wheels for PyYAML, which is required to install pyproject.toml-based projects" others are having...

CannonFather, 2023-04-04:
Same here with data/data error
💻 No bootstrap detected, proceeding with installation
💻 Arch: arm
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-armv7a.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/2cd65898-00b6-4b5a-bd0f-7a036eff0c4f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230404%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230404T015300Z&X-Amz-Expires=300&X-Amz-Signature=2fe7cdadb9191e3d13e01d208143e73e4e01045a99655e49400145b7d24f7a29&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-armv7a.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap extracted, setting it up...
💻 >ls
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 root-pre5
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
💻 >chmod -R 700 .
💻 >sh install-bootstrap.sh
🐸 bootstrap ready, run with run-bootstrap.sh
💻 >sh add-user.sh octoprint
💻 >cat /etc/motd
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Welcome to Alpine!
🐸 
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <https://wiki.alpinelinux.org/>.
🐸 
🐸 You can setup the system with the command: setup-alpine
🐸 
🐸 You may change this message by editing /etc/motd.
🐸 
💻 >env
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 EXTRA_BIND=-b /storage/emulated/0/Android/data/com.octo4a/files:/root -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a -b /data/data/com.octo4a/files/bootstrap/ioctlHook.so:/home/octoprint/ioctlHook.so
🐸 MC_AUTH_TOKEN_PATH=/efs
🐸 USER=root
🐸 EXTERNAL_STORAGE=/sdcard
🐸 KNOX_STORAGE=/data/knox/ext_sdcard
🐸 PROOT_L2S_DIR=/data/data/com.octo4a/files/bootstrap/bootstrap/.proot.meta
🐸 SHLVL=1
🐸 HOME=/root
🐸 ANDROID_ASSETS=/system/app
🐸 BOOTCLASSPATH=/system/framework/core-oj.jar:/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/sprengine.jar:/system/framework/knoxsdk.jar:/system/framework/sec_edm.jar:/system/framework/sagearpolicymanager.jar:/system/framework/smartbondingservice.jar:/system/framework/timakeystore.jar:/system/framework/fipstimakeystore.jar:/system/framework/knoxvpnuidtag.jar:/system/framework/sec_sdp_sdk.jar:/system/framework/sec_sdp_hidden_sdk.jar:/system/framework/esecomm.jar:/system/framework/SemAudioThumbnail.jar:/system/framework/smatlib.jar:/system/framework/samsung-framework.jar
🐸 _=./root/bin/proot
🐸 ANDROID_DATA=/data
🐸 PATH=/sbin:/usr/sbin:/bin:/usr/bin
🐸 ANDROID_ROOT=/system
🐸 LANG='en_US.UTF-8'
🐸 ENC_EMULATED_STORAGE_TARGET=/storage/enc_emulated
🐸 ANDROID_SOCKET_zygote=8
🐸 ASEC_MOUNTPOINT=/mnt/asec
🐸 PWD=/root
🐸 SYSTEMSERVERCLASSPATH=/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar:/system/framework/samsung-services.jar:/system/framework/ssrm.jar:/system/framework/hqm.jar
🐸 ANDROID_STORAGE=/storage
🐸 PROOT_TMP_DIR=/data/data/com.octo4a/files/bootstrap/tmp
🐸 ANDROID_BOOTLOGO=1
💻 >ls /
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 bin
🐸 dev
🐸 etc
🐸 home
🐸 lib
🐸 media
🐸 mnt
🐸 opt
🐸 proc
🐸 root
🐸 run
🐸 sbin
[... 428 lines cut ...]
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:38)
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:12)
	at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:164)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
	at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [proot warning: can't sanitize binding "/data/data

CannonFather, 2023-04-04:
So I noticed that when I went to run the program when prompted after installing the apk, it would flash "do not have storage permission" notification or something to that effect. But when you install the apk, it says it needs no permissions. So I cleared data, uninstalled it and wiped what was left (even deleting the com.octo4a folder) ran the apk again, but didn't run the octo4a app. I changed the permissions of storage and camera. Then ran the app. Got to the same point and it died there again. To note the app didn't ask for storage permission, just camera.

dynalogix, 2023-04-04:
Exact same ERROR in all recently added issues. The same happens on my Galaxy Note 3 on API level 17, which is still free from the permission nightmare Android has since become. So I don't think this issue is permission related.

CannonFather, 2023-04-04:
Yeah I see that, It's kinda to bad we can't install an older version till this gets fixed, but all the apk's install the same files. so they all stop at the same point. Oh well. Que sara sara.

royaltongue, 2023-04-09:
Also seeing this error on a Galaxy Tab S2 (SM-T818V) on Android 7.0 (update version T818VVRS4BTJK)

jdkorzen, 2023-04-12:
Same error on Galaxy s6

troth530, 2023-04-14:
Same issue on Galaxy Tab A (SM-T350) on Android 7.1.1 
Build number NMF26X.T350XXU1CQJ5

madelponte, 2023-04-16:
Same error on pixel 4a on android 13

Bluep, 2023-04-17:
Same error here on a Samsung Galaxy J5 (2016).

IsYou77, 2023-04-23:
Same here, Huawei P9 lite, A7.0

Ethan-G, 2023-04-27:
Same here on a Galaxy S7

Whats-name, 2023-05-01:
I have the same on Sony Z1 and Huawei MediaPad Youth 7 when I try to install Octo4a :/

[image]

Cemen-nn, 2023-05-05:
Same error here on a Huawei MediaPadT3 10.

bleroy, 2023-05-12:
Same error on a Galaxy S5 and on a LG K20 Plus

calza86, 2023-05-17:
I tried install the version 1.2.4 in a Samsung S6 (Android 7.0, kernel 3.10.61-13830439) and I get the same error.
Later I tried install the same version on a Samsung S10 (Android 12, kernel 4.14.113-25257816) and on a Samsung A8 2018 (Android 9, kernel 4.4.111-22928167) and in both devices the error didn't appears, the installation ended correct and application start.

What is the minimum version of Android enough to run the installation and program?
Is it possible that the latest version is not compatible with the previous versions of Android 9?

Thanks
Marco

myAlterX, 2023-05-18:
You're likely being blocked from downloading some files.

Use cellular plan (turn off wifi) or use a VPN like ProtonVPN (free) and try again.

See here for an explanation of the issue https://github.com/feelfreelinux/octo4a/issues/68

bleroy, 2023-05-18:
Installed a VPN, tried again, same result.

royaltongue, 2023-05-18:
I tried with NordVPN and it didn't work either

calza86, 2023-05-19:
I tried install by a different connection but process ended in the same error.
Just to clarify I tried with 3 different devices connected to the same network and the process ends in success only for the devices with Android 9+ and kernel 4+, the process ends always in error for the device with Android 7 and kernel 3.X

rushl, 2023-05-31:
I too am getting this error. Using a Samsung S6, Android 7. I'm super disappointed! After watching review after review about using an old phone to run Octoprint, I bought a cable and a charger to use my old phone. Sadly, it seems not to be. Does anyone have a workaround?

lebeno, 2023-06-08:
> Does anyone have a workaround?

@rushl the workaround I found was installing release https://github.com/feelfreelinux/octo4a/releases/tag/0.0.5

kaanngunerr, 2023-06-23:
Anyone find any solution?

TheHellSite, 2023-06-28:
Same here on an old HTC One M7 running latest LineageOS 18.1 (release 20230627) / Android 11 with kernel 3.x.
As others have mentioned already it might have something to do with the kernel 3.x just beeing too old.

My OnePlus 7 runs the app no problem with Android 11 and kernel 4.x.
However IMO it is not suitable to use a $ 300+ device for this task. For me the only reason running OctoPrint as an app on an old phone rather than on something like a Raspberry is to give a 2nd life to otherwise wasted resources.

TheHellSite, 2023-06-28:
> > Does anyone have a workaround?
> 
> @rushl the workaround I found was installing release https://github.com/feelfreelinux/octo4a/releases/tag/0.0.5

How did you manage to install this on Android manually?

lopazopy, 2023-06-29:
Same issue with Samsung On5 (sm-g550t1) on Android 6.0.1.

SrGatoo, 2023-07-01:
Same issue on a Redmi 6A with android 11

andrasig-bme, 2023-07-02:
Same here with Sony Xperia Z5 Compact on Android 11 (Lineage OS 18.1).

ZuluPilgrim, 2023-07-11:
Same on Huawei P30 Pro

SpecialK85, 2023-07-18:
Same error 

java.lang.RuntimeException: Process exited with error code 1. [proot warning: can't sanitize binding "/data/data

S8+

fernplant, 2023-08-28:
Same on my S6 . Shame this has no fix

LucasPaivaSilva, 2023-10-20:
Same on my S6 Edge. Is there anything we can do to help?

rTarlini, 2023-11-04:
Same issue on Galaxy S6. 0.0.5 version workaround worked.

djansen1987, 2023-11-10:
Same with a Samsung S6

my errorlog:
```
💻 No bootstrap detected, proceeding with installation
💻 Arch: aarch64
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-aarch64.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/ea14dee1-fef5-465a-9f22-4ff9e7631821?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231110%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231110T182613Z&X-Amz-Expires=300&X-Amz-Signature=80039361702a7e3584e0c11dd5d19d993b81e1f2ff5c1533f35199c38ac5ab58&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-aarch64.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap extracted, setting it up...
💻 >ls
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
💻 >chmod -R 700 .
💻 >sh install-bootstrap.sh
🐸 bootstrap ready, run with run-bootstrap.sh
💻 >sh add-user.sh octoprint
💻 >cat /etc/motd
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Welcome to Alpine!
🐸
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <https://wiki.alpinelinux.org/>.
🐸
🐸 You can setup the system with the command: setup-alpine
🐸
🐸 You may change this message by editing /etc/motd.
🐸
💻 >env
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 EXTRA_BIND=-b /storage/emulated/0/Android/data/com.octo4a/files:/root -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a -b /data/data/com.octo4a/files/bootstrap/ioctlHook.so:/home/octoprint/ioctlHook.so
🐸 MC_AUTH_TOKEN_PATH=/efs
🐸 USER=root
🐸 EXTERNAL_STORAGE=/sdcard
🐸 KNOX_STORAGE=/data/knox/ext_sdcard
🐸 PROOT_L2S_DIR=/data/data/com.octo4a/files/bootstrap/bootstrap/.proot.meta
🐸 SHLVL=1
🐸 HOME=/root
🐸 ANDROID_ASSETS=/system/app
🐸 BOOTCLASSPATH=/system/framework/core-oj.jar:/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/sprengine.jar:/system/framework/knoxsdk.jar:/system/framework/sec_edm.jar:/system/framework/sagearpolicymanager.jar:/system/framework/smartbondingservice.jar:/system/framework/timakeystore.jar:/system/framework/fipstimakeystore.jar:/system/framework/knoxvpnuidtag.jar:/system/framework/sec_sdp_sdk.jar:/system/framework/sec_sdp_hidden_sdk.jar:/system/framework/esecomm.jar:/system/framework/SemAudioThumbnail.jar:/system/framework/smatlib.jar:/system/framework/samsung-framework.jar
🐸 ANDROID_SOCKET_zygote_agent64=8
🐸 _=./root/bin/proot
🐸 ANDROID_DATA=/data
🐸 PATH=/sbin:/usr/sbin:/bin:/usr/bin
🐸 ANDROID_ROOT=/system
🐸 LANG='en_US.UTF-8'
🐸 ENC_EMULATED_STORAGE_TARGET=/storage/enc_emulated
🐸 ASEC_MOUNTPOINT=/mnt/asec
🐸 PWD=/root
🐸 SYSTEMSERVERCLASSPATH=/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar:/system/framework/samsung-services.jar:/system/framework/ssrm.jar:/system/framework/hqm.jar
🐸 ANDROID_STORAGE=/storage
🐸 PROOT_TMP_DIR=/data/data/com.octo4a/files/bootstrap/tmp
🐸 ANDROID_BOOTLOGO=1
💻 >ls /
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 bin
🐸 dev
🐸 etc
🐸 home
🐸 lib
🐸 media
🐸 mnt
🐸 opt
🐸 proc
🐸 root
[... 990 lines cut ...]
at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:16)
at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:167)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [proot warning: can't sanitize binding "/data/data
```

Ethan-G, 2024-01-16:
I have hit this error on four different phones now on 3-4 versions of android. It's feeling like Octo4a doesn't actually work at all at this point.

kokospalme, 2024-02-06:
Same issue here (Honor 8).
Did someone try to root the phone? Maybe it's something wrong with the permission.
My fairphone 4 (Android 13) works fine as it is (no custom ROM, not rooted).