feelfreelinux/octo4a #402: Octo4a - the app shows an entirely wrong IP address

frMaciek, 2023-04-02:
I had a problem with this app: I ran it on an old TV box: **Minix Neo X7 mini** (android 4.4.1) - but when it started it showed an entirely wrong IP address (despite the that I'm connected to my home wifi network).

My home network's address pool is 192.168.88.1/24 but Octo4a shows 192.168.254.103...

And it doesn't matter if I'm connected to any network or not, whether I have a hotspot from the device or not ... The worst thing is that it can't be changed in any way and even trying to reinstall the application - nothing helps here.

Has anyone had a similar problem and know if anything can be done?
Thanks in advance!

sv0o, 2023-04-02:
yes.  this has happened to me a few times even after my DHCP server was set up to always give the device the same static IP.  If I turn wifi off and back on again, it usually gets the proper IP address.  Are you able to configure your DHCP to give the X7 a static IP?  When I use a phone with a screen, cycling wifi off and on is simple.  Since I don't want to connect a TV to my TV box, I also put it on a smart plug so I can power cycle it with google home.

 octo4a is configured to auto start when the device boots up, I can fix many issues by simply "turning it off and turning it back on again"

frMaciek, 2023-04-02:
Hmm, I didn't think to assign a fixed IP to this device - thanks for the hint! And I have another question - would there be an option to manually correct the IP address in the application?

frMaciek, 2023-04-02:
Ok, so I set a static IP for the device - but still, the problem remained and octo4a still shows the same IP address.
However, I noticed another interesting thing: when I restart the device - by selecting the appropriate option from the menu or the remote control, after restarting I get the information that "The installation is damaged, please reinstall Octo4a" - this is probably not normal behaviour, right?
For example, can the fact, that this device is not rooted, affect the operation of this application?

sv0o, 2023-04-03:
ok, that's a separate problem.  
Look at issue #325 you may have to ssh into the device to fix a corrupt install

frMaciek, 2024-01-06:
I tried this, but with no success

tsu257, 2024-05-01:
Did anyone find a solution for this issue? I had it working normally on an LG K52, but after an update the installation corrupted and I never got it to work again. I did the #325 procedure and the IP keeps showing 192.168.254.103

frMaciek, 2024-05-01:
Hello, being honest, I wasn't aware of any workaround procedure at all - I have to check it on my device and will let you know.