feelfreelinux/octo4a #409: Mass permission denied error using Moto devices.

AAMBIENTT, 2023-04-14:
When attempting to install to Moto G Pure running android 12, I recieve this as a log. I cannot access the OSSH Server as it spits out error code 126, and tapping the extentions tab crashes the application. My log is attached below. Hope it helps. I;m open to suggestions.
```
No bootstrap detected, proceeding with installation Arch: arm Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-armv7a.zip Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/2cd65898-00b6-4b5a-bd0f-7a036eff0c4f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230414%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230414T233252Z&X-Amz-Expires=300&X-Amz-Signature=45d593081bd7e133dbbe44038ce44455a8c06f0340a0e19e0e01cf2a41c09af7&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-armv7a.zip&response-content-type=application%2Foctet-stream returned status code 200 Bootstrap extracted, setting it up... >ls add-user.sh fake_proc_stat install-bootstrap.sh ioctlHook.so root root-pre5 rootfs.tar.xz run-bootstrap.sh >chmod -R 700 . >sh install-bootstrap.sh install-bootstrap.sh[8]: ../root/bin/minitar: can't execute: Permission denied install-bootstrap.sh[13]: can't create bootstrap/etc/resolv.conf: No such file or directory bootstrap ready, run with run-bootstrap.sh >sh add-user.sh octoprint add-user.sh[5]: can't create bootstrap/etc/passwd: No such file or directory An exception has occurred at:java.lang.RuntimeException: Process exited with error code 1. [add-user.sh[5]: can't create bootstrap/etc/passwd	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:38)	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:12)	at com.octo4a.repository.BootstrapRepositoryImpl$setupBootstrap$2.invokeSuspend(BootstrapRepository.kt:152)	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
Exception:java.lang.RuntimeException: Process exited with error code 1. [add-user.sh[5]: can't create bootstrap/etc/passwd
```

chriso463, 2023-04-15:
this happens to me too

codecompliant, 2023-08-20:
Same issues here. It did install on Android 11 but won't install on 12.

AAMBIENTT, 2023-09-05:
dead app support, will be forking and making one that properly works soon

closed 2023-09-05 by AAMBIENTT

Drasnore, 2025-10-05:
Hello, also using a Moto Edge 60 Fusion, and same issue.

If you need any help for testing I'm here.