feelfreelinux/octo4a #418: Install failed with proot error

hubdows, 2023-05-19:
Attempted install on Samsung Galaxy S6 Edge with Android 7.0 error log below.

💻 No bootstrap detected, proceeding with installation
💻 Arch: aarch64
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-aarch64.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/ea14dee1-fef5-465a-9f22-4ff9e7631821?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230519%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230519T202442Z&X-Amz-Expires=300&X-Amz-Signature=d84d0c906f78390f19022cf6652734cca09640c415b6c1e7a91ff4bb008d25a6&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-aarch64.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap extracted, setting it up...
💻 >ls
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
💻 >chmod -R 700 .
💻 >sh install-bootstrap.sh
🐸 bootstrap ready, run with run-bootstrap.sh
💻 >sh add-user.sh octoprint
💻 >cat /etc/motd
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Welcome to Alpine!
🐸 
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <https://wiki.alpinelinux.org/>.
🐸 
🐸 You can setup the system with the command: setup-alpine
🐸 
🐸 You may change this message by editing /etc/motd.
🐸 
💻 >env
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 EXTRA_BIND=-b /storage/emulated/0/Android/data/com.octo4a/files:/root -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a -b /data/data/com.octo4a/files/bootstrap/ioctlHook.so:/home/octoprint/ioctlHook.so
🐸 MC_AUTH_TOKEN_PATH=/efs
🐸 USER=root
🐸 EXTERNAL_STORAGE=/sdcard
🐸 KNOX_STORAGE=/data/knox/ext_sdcard
🐸 PROOT_L2S_DIR=/data/data/com.octo4a/files/bootstrap/bootstrap/.proot.meta
🐸 SHLVL=1
🐸 HOME=/root
🐸 ANDROID_ASSETS=/system/app
🐸 BOOTCLASSPATH=/system/framework/core-oj.jar:/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/sprengine.jar:/system/framework/knoxsdk.jar:/system/framework/sec_edm.jar:/system/framework/sagearpolicymanager.jar:/system/framework/smartbondingservice.jar:/system/framework/timakeystore.jar:/system/framework/fipstimakeystore.jar:/system/framework/knoxvpnuidtag.jar:/system/framework/sec_sdp_sdk.jar:/system/framework/sec_sdp_hidden_sdk.jar:/system/framework/esecomm.jar:/system/framework/SemAudioThumbnail.jar:/system/framework/smatlib.jar:/system/framework/samsung-framework.jar
🐸 ANDROID_SOCKET_zygote_agent64=8
🐸 _=./root/bin/proot
🐸 ANDROID_DATA=/data
🐸 PATH=/sbin:/usr/sbin:/bin:/usr/bin
🐸 ANDROID_ROOT=/system
🐸 LANG='en_US.UTF-8'
🐸 ENC_EMULATED_STORAGE_TARGET=/storage/enc_emulated
🐸 ASEC_MOUNTPOINT=/mnt/asec
🐸 PWD=/root
🐸 SYSTEMSERVERCLASSPATH=/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar:/system/framework/samsung-services.jar:/system/framework/ssrm.jar:/system/framework/hqm.jar
🐸 ANDROID_STORAGE=/storage
🐸 PROOT_TMP_DIR=/data/data/com.octo4a/files/bootstrap/tmp
🐸 ANDROID_BOOTLOGO=1
💻 >ls /
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 bin
🐸 dev
🐸 etc
🐸 home
🐸 lib
🐸 media
🐸 mnt
🐸 opt
🐸 proc
🐸 root
🐸 run
🐸 sbin
[... 402 lines cut ...]
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:38)
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:12)
	at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:164)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
	at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [proot warning: can't sanitize binding "/data/data

vztekloun605, 2023-05-21:
hi. i have the same problem. samsung j510fn, official android 7.1.1 :( 
here is more
https://github.com/feelfreelinux/octo4a/issues/401

fedetony, 2023-06-02:
Same #422

Flash2469, 2023-09-11:
Yep seriously broken
Same error on 4 different phones
Samsung J1 android 4.4
Samsung J1 mini android 5.1
Alcatel 4047a android 7
ZTE Blade a602 android 7

Fix is needed

SonOfSardaar, 2023-12-10:
It is not the phone that is the problem, it is the OS. After trying 5 different ROMs I got with worked with Lineage OS and CM ROMs but only Android 6 or 7.