feelfreelinux/octo4a #419: Octoprint 1.8.7 > 1.9 Failed Upgrade Octo4a

Madcamper, 2023-05-23:
Not sure if this is a octo4a or an octoprint issue. I tried updating to the latest release of octoprint 1.9 from 1.8.7 and got an upgrade error. The python version says 3.9.5 in my octoprint UI.

This is the tail end of my octoprint.log:
``` 2023-05-23 12:46:13,705 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.9.0...
2023-05-23 12:50:26,920 - octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 2187, in _perform_update
    update_result = updater.perform_update(
  File "/usr/lib/python3.9/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 126, in perform_update
    raise exceptions.UpdateError(
octoprint.plugins.softwareupdate.exceptions.UpdateError: Error while executing pip install
2023-05-23 12:50:26,976 - octoprint.plugins.softwareupdate - INFO - Saved update log to disk ``` 

This is the tail end of my plugin_softwareupdate_consol.log from octoprint:

``` 2023-05-23 12:49:24,898 > Collecting urllib3<3,>=1.21.1
2023-05-23 12:49:24,899 > Downloading urllib3-2.0.2-py3-none-any.whl (123 kB)
2023-05-23 12:49:26,082 > Using legacy 'setup.py install' for emoji, since package 'wheel' is not installed.
2023-05-23 12:49:26,082 > Using legacy 'setup.py install' for future, since package 'wheel' is not installed.
2023-05-23 12:49:26,084 > Using legacy 'setup.py install' for watchdog, since package 'wheel' is not installed.
2023-05-23 12:49:26,085 > Building wheels for collected packages: OctoPrint, frozendict, psutil
2023-05-23 12:49:26,093 > Building wheel for OctoPrint (PEP 517): started
2023-05-23 12:49:59,906 > Building wheel for OctoPrint (PEP 517): finished with status 'done'
2023-05-23 12:49:59,911 > Created wheel for OctoPrint: filename=OctoPrint-1.9.0-py2.py3-none-any.whl size=3288046 sha256=1d1cb1a383da59f83b908472ecb4cb69fc1559fbd6e2f4028b526ba0dcf31e02
2023-05-23 12:49:59,916 > Stored in directory: /tmp/pip-ephem-wheel-cache-rmv0tchq/wheels/56/b0/8b/6ef713f0941f640a75be3b17228ff0f3130bb682ea211ed72f
2023-05-23 12:49:59,917 > Building wheel for frozendict (PEP 517): started
2023-05-23 12:50:14,025 > Building wheel for frozendict (PEP 517): finished with status 'done'
2023-05-23 12:50:14,027 > Created wheel for frozendict: filename=frozendict-2.3.8-cp39-cp39-linux_armv7l.whl size=108343 sha256=f3bbd5462662af784b9b16e21fd4afade880635378a05f604786e7f804eb5b2a
2023-05-23 12:50:14,028 > Stored in directory: /tmp/pip-ephem-wheel-cache-rmv0tchq/wheels/b1/a6/53/75efbfe809926858688ffa6fc5aaee950fcee3d7548d40cf95
2023-05-23 12:50:14,030 > Building wheel for psutil (PEP 517): started
2023-05-23 12:50:19,698 ! ERROR: Command errored out with exit status 1:
2023-05-23 12:50:19,700 ! command: /usr/bin/python3 /usr/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmpit8l6m_3
2023-05-23 12:50:19,701 ! cwd: /tmp/pip-install-7my2z8xt/psutil_574c750365b24185b2de423314077cf5
2023-05-23 12:50:19,707 ! Complete output (52 lines):
2023-05-23 12:50:19,710 ! /tmp/tmpxwn2ats8.c:1:10: fatal error: linux/ethtool.h: No such file or directory
2023-05-23 12:50:19,711 ! 1 | #include <linux/ethtool.h>
2023-05-23 12:50:19,712 ! |          ^~~~~~~~~~~~~~~~~
2023-05-23 12:50:19,714 ! compilation terminated.
2023-05-23 12:50:19,714 ! running bdist_wheel
2023-05-23 12:50:19,718 ! running build
2023-05-23 12:50:19,720 ! running build_py
2023-05-23 12:50:19,721 ! creating build
2023-05-23 12:50:19,731 ! creating build/lib.linux-armv7l-cpython-39
2023-05-23 12:50:19,732 ! creating build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,734 ! copying psutil/_pswindows.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,735 ! copying psutil/_psposix.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,736 ! copying psutil/_psaix.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,739 ! copying psutil/_psbsd.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,740 ! copying psutil/__init__.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,741 ! copying psutil/_common.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,743 ! copying psutil/_pslinux.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,744 ! copying psutil/_psosx.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,745 ! copying psutil/_compat.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,746 ! copying psutil/_pssunos.py -> build/lib.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,748 ! creating build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,749 ! copying psutil/tests/test_connections.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,750 ! copying psutil/tests/test_contracts.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,751 ! copying psutil/tests/test_aix.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,752 ! copying psutil/tests/test_misc.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,753 ! copying psutil/tests/test_bsd.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,754 ! copying psutil/tests/__main__.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,756 ! copying psutil/tests/test_linux.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,759 ! copying psutil/tests/test_posix.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,761 ! copying psutil/tests/test_testutils.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,763 ! copying psutil/tests/runner.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,764 ! copying psutil/tests/test_memleaks.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,765 ! copying psutil/tests/test_windows.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,766 ! copying psutil/tests/test_process.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,767 ! copying psutil/tests/__init__.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,768 ! copying psutil/tests/test_system.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,768 ! copying psutil/tests/test_osx.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,769 ! copying psutil/tests/test_unicode.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,770 ! copying psutil/tests/test_sunos.py -> build/lib.linux-armv7l-cpython-39/psutil/tests
2023-05-23 12:50:19,771 ! running build_ext
2023-05-23 12:50:19,772 ! building 'psutil._psutil_linux' extension
2023-05-23 12:50:19,773 ! creating build/temp.linux-armv7l-cpython-39
2023-05-23 12:50:19,775 ! creating build/temp.linux-armv7l-cpython-39/psutil
2023-05-23 12:50:19,776 ! gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fomit-frame-pointer -g -fno-semantic-interposition -fomit-frame-pointer -g -fno-semantic-interposition -fomit-frame-pointer -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -DPSUTIL_POSIX=1 -DPSUTIL_SIZEOF_PID_T=4 -DPSUTIL_VERSION=595 -DPy_LIMITED_API=0x03060000 -DPSUTIL_ETHTOOL_MISSING_TYPES=1 -DPSUTIL_LINUX=1 -I/usr/include/python3.9 -c psutil/_psutil_common.c -o build/temp.linux-armv7l-cpython-39/psutil/_psutil_common.o
2023-05-23 12:50:19,777 ! gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fomit-frame-pointer -g -fno-semantic-interposition -fomit-frame-pointer -g -fno-semantic-interposition -fomit-frame-pointer -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -DPSUTIL_POSIX=1 -DPSUTIL_SIZEOF_PID_T=4 -DPSUTIL_VERSION=595 -DPy_LIMITED_API=0x03060000 -DPSUTIL_ETHTOOL_MISSING_TYPES=1 -DPSUTIL_LINUX=1 -I/usr/include/python3.9 -c psutil/_psutil_linux.c -o build/temp.linux-armv7l-cpython-39/psutil/_psutil_linux.o
2023-05-23 12:50:19,778 ! psutil/_psutil_linux.c:19:10: fatal error: linux/version.h: No such file or directory
2023-05-23 12:50:19,779 ! 19 | #include <linux/version.h>
2023-05-23 12:50:19,781 ! |          ^~~~~~~~~~~~~~~~~
2023-05-23 12:50:19,781 ! compilation terminated.
[... cut ...]
2023-05-23 12:50:19,784 ! error: command '/usr/bin/gcc' failed with exit code 1
2023-05-23 12:50:19,785 ! ----------------------------------------
2023-05-23 12:50:19,786 ! ERROR: Failed building wheel for psutil
2023-05-23 12:50:19,787 ! ERROR: Could not build wheels for psutil which use PEP 517 and cannot be installed directly
2023-05-23 12:50:20,299 > Building wheel for psutil (PEP 517): finished with status 'error'
2023-05-23 12:50:20,302 > Successfully built OctoPrint frozendict
2023-05-23 12:50:20,303 > Failed to build psutil ```

degroof, 2023-05-26:
I'm running into the same issue. What steps are required to get around that?

Madcamper, 2023-05-26:
The steps I took are as follows:

1. Make a backup of your octoprint config through the GUI and save it to your computer
2. On your phone, clear the data and cache of the app
3. Uninstall the octo4a app
4. Re-install the latest version
5. During octoprint setup, choose to restore from a backup and select the backed up .zip file you performed in step 1.

This should get it updated to 1.9.0 and update python without losing any if your octoprint configs.

degroof, 2023-05-26:
Thanks. That seems to have done the trick.

klausmaus, 2023-07-18:
Same issue, the solution works fine, thank you!