feelfreelinux/octo4a #421: Cant sanitize binding. Running on an Asus tablet with android version 7.0

Anoahnator1, 2023-06-01:
💻 No bootstrap detected, proceeding with installation
💻 Arch: aarch64
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-aarch64.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/ea14dee1-fef5-465a-9f22-4ff9e7631821?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230601%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230601T134055Z&X-Amz-Expires=300&X-Amz-Signature=2a5b550e4a0dd279d702687551e0c0797b19c328e19b99d891f645e0b652a0a1&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-aarch64.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap extracted, setting it up...
💻 >ls
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
💻 >chmod -R 700 .
💻 >sh install-bootstrap.sh
🐸 bootstrap ready, run with run-bootstrap.sh
💻 >sh add-user.sh octoprint
💻 >cat /etc/motd
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Welcome to Alpine!
🐸 
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <https://wiki.alpinelinux.org/>.
🐸 
🐸 You can setup the system with the command: setup-alpine
🐸 
🐸 You may change this message by editing /etc/motd.
🐸 
💻 >env
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 EXTRA_BIND=-b /storage/emulated/0/Android/data/com.octo4a/files:/root -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a -b /data/data/com.octo4a/files/bootstrap/ioctlHook.so:/home/octoprint/ioctlHook.so
🐸 USER=root
🐸 EXTERNAL_STORAGE=/sdcard
🐸 PROOT_L2S_DIR=/data/data/com.octo4a/files/bootstrap/bootstrap/.proot.meta
🐸 SHLVL=1
🐸 HOME=/root
🐸 ANDROID_ASSETS=/system/app
🐸 BOOTCLASSPATH=/system/framework/core-oj.jar:/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/mediatek-framework.jar
🐸 _=./root/bin/proot
🐸 ANDROID_DATA=/data
🐸 PATH=/sbin:/usr/sbin:/bin:/usr/bin
🐸 ANDROID_ROOT=/system
🐸 LANG='en_US.UTF-8'
🐸 ANDROID_SOCKET_zygote=8
🐸 ASEC_MOUNTPOINT=/mnt/asec
🐸 PWD=/root
🐸 SYSTEMSERVERCLASSPATH=/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar
🐸 ANDROID_STORAGE=/storage
🐸 PROOT_TMP_DIR=/data/data/com.octo4a/files/bootstrap/tmp
🐸 ANDROID_BOOTLOGO=1
💻 >ls /
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 bin
🐸 dev
🐸 etc
🐸 home
🐸 lib
🐸 media
🐸 mnt
🐸 opt
🐸 proc
🐸 root
🐸 run
🐸 sbin
🐸 srv
🐸 storage
🐸 sys
🐸 system
🐸 tmp
[... 958 lines cut ...]
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:38)
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:12)
	at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:164)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
	at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [proot warning: can't sanitize binding "/data/data

freakmaxi, 2023-06-07:
I have the same problem. Samsung Galaxy S4 - Android 5.0.1

carlos4276, 2023-06-15:
tengo el mismo problema en una galaxy note 10.1 n 8000 la actualice a linaje porque antes tenia android 4.1 y no dejaba descargar el problema y ahora me da error

sburns90, 2023-11-04:
Same problem here.
Lg V20 (lg h918)
Android 7
Any one have any ideas

PrintaholicID, 2023-11-14:
Similar situation, same error message.  "Can't sanitize binding." Android 6.0.1 Samsung Galaxy phone.

im36degrees, 2024-01-28:
same issue