feelfreelinux/octo4a #422: Fatal Error Motorola Razr- XT890 installation

fedetony, 2023-06-02:
Motorola Razr- XT890
linage Os ROM
Android version 9
![Screenshot_20230602-191
[image]
905_OctoPrint_For_Android](https://github.com/feelfreelinux/octo4a/assets/45215920/fc66fcee-758e-4f27-b75f-7683aa2b526d)
Just crashes at the dependencies installation.
[image]
[image]

Puff :(... wanted to use my old phone which I still love...
Thanks in advance :)

fernplant, 2023-08-28:
Got the same issue on Samsung S6

Jester2154, 2024-04-01:
Same problem on my Motorola