feelfreelinux/octo4a #436: webcam stream on firefox freezes tab

rmayergfx, 2023-08-04:
Firefox 116.0.1 is not working together with octo4a. Installet on a FireHD 8, working fine but no webcam stream is working with firefox. The tab with octo4a will freeeze and needs to be closed. Chrome works like a charme.

jtlarson, 2023-08-22:
I had a similar issue that was improved by turning off "Use hardware acceleration when available" in Firefox settings. You can also try turning down the resolution and frame rate of the webcam server.

rmayergfx, 2023-08-23:
Not a solution, just a workaround. Why should i change FireFox settings, it works fine on all other sites also with the original octoprint. Resolution change does not work, still freezing.

Rolleroo, 2024-05-31:
I'm having the same issue with Firefox (Ver 126.0.1). It appears to be working fine on Edge. Tried changing the video codec from the default libx264 to mpeg2video with no luck.

Appears to be a Firefox issue as the server still responds find to an Edge window running at the same time.

Rather not use a Chromium browser and would like the high res so a fix or work around would be good.

Rolleroo, 2024-09-12:
Something has changed in Firefox or the app. Getting Firefox responding appropriately at the moment.