feelfreelinux/octo4a #439: /dev Permission error

Madcamper, 2023-08-22:
I've been getting this spam error in my octo4a logs for a while, it stops after I've made a successful connection to the printer though. Today, I am having some weird issues with octo4a where it crashes when trying to connect or detection serial connection times out after about 30 seconds, camera server stops running randomly. This all happened out of no where, it worked fine last night, I woke up this morning and tried to connect to octoprint and it doesn't work. I am using octoprint fine on another printer with the same phone.

[image]

Madcamper, 2023-08-31:
Unfortunately I'm on a prusa and don't use custom firmware with it but hopefully this information will be helpful for anyone coming along and/or the dev to maybe pinpoint somewhere in the octo4a code to remediate it.

Pyenb, 2023-10-17:
Anyone got something new? I am stuck on the same problem on my old Oneplus 6. I already tried every chmod method, adding groups, trying out sudo, editing the octoprint user, etc...

Tought about either hardcoding the devices in the `comm-fix.py` or trying to ask for root access. I will try to build the app and test it out locally

mattmon, 2023-12-08:
I got it working by setting SELinux to permissive:

"setenforce 0" in android shell

to make it permanent, repack your boot.img with "androidboot.selinux=permissive" in kernel cmdline

Pyenb, 2023-12-08:
> "setenforce 0" in android shell

Thanks, will definitly try that out later. I tried rebuilding the app with it asking for root access, by editing the said `comm-fix.py`, but couldn't get it to compile.

Just to make sure, did you set it it inside octo4a or on the host phone?

mattmon, 2023-12-08:
> > "setenforce 0" in android shell
> 
> Thanks @mattmon, will definitly try that out later. I tried rebuilding the app with it asking for root access, by editing the said `comm-fix.py`, but couldn't get it to compile.
> 
> Just to make sure, did you set it it inside octo4a or on the host phone?

You need to set SELinux to permissive for android's kernel.

"adb shell su setenforce 0" is ephemeral

repacking the boot.img is permanent

Pyenb, 2023-12-08:
> "adb shell su setenforce 0" is ephemeral

Ok, thank you very much