feelfreelinux/octo4a #445: Samsung S4 1.2.5 Failed Install

ASCONTRACT1, 2023-09-15:
💻 No bootstrap detected, proceeding with installation
💻 Arch: arm
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-armv7a.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/2cd65898-00b6-4b5a-bd0f-7a036eff0c4f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230915%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230915T195836Z&X-Amz-Expires=300&X-Amz-Signature=0bd7e367e5abce8e2a97ecb18d2a28f80a095bdde594eb07f7d22b03be6fea5a&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-armv7a.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap extracted, setting it up...
💻 >ls
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 root-pre5
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
💻 >chmod -R 700 .
💻 >sh install-bootstrap.sh
🐸 Unable to chmod ..//serialpipe: Permission denied
🐸 Bad mode
🐸 bootstrap ready, run with run-bootstrap.sh
💻 >sh add-user.sh octoprint
💻 >cat /etc/motd
🐸 Welcome to Alpine!
🐸 
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <https://wiki.alpinelinux.org/>.
🐸 
🐸 You can setup the system with the command: setup-alpine
🐸 
🐸 You may change this message by editing /etc/motd.
🐸 
💻 >env
🐸 EXTRA_BIND=-b /storage/emulated/0/Android/data/com.octo4a/files:/root -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a -b /data/data/com.octo4a/files/bootstrap/ioctlHook.so:/home/octoprint/ioctlHook.so
🐸 MC_AUTH_TOKEN_PATH=/efs
🐸 USER=root
🐸 KNOX_STORAGE=/data/knox/ext_sdcard
🐸 EXTERNAL_STORAGE=/storage/emulated/legacy
🐸 LOOP_MOUNTPOINT=/mnt/obb
🐸 PROOT_L2S_DIR=/data/data/com.octo4a/files/bootstrap/bootstrap/.proot.meta
🐸 SHLVL=1
🐸 RANDOM=24211
🐸 VIBE_PIPE_PATH=/dev/pipes
🐸 HOME=/root
🐸 ANDROID_ASSETS=/system/app
🐸 BOOTCLASSPATH=/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/mms-common.jar:/system/framework/android.policy.jar:/system/framework/apache-xml.jar:/system/framework/sec_edm.jar:/system/framework/seccamera.jar:/system/framework/scrollpause.jar:/system/framework/stayrotation.jar:/system/framework/smartfaceservice.jar:/system/framework/smartbondingservice.jar:/system/framework/commonimsinterface.jar:/system/framework/imsmanager.jar:/system/framework/secocsp.jar:/system/framework/secEmailBC.jar:/system/framework/knoxvpnuidtag.jar
🐸 _=./root/bin/proot
🐸 ANDROID_DATA=/data
🐸 PATH=/sbin:/usr/sbin:/bin:/usr/bin
🐸 ANDROID_ROOT=/system
🐸 SECONDARY_STORAGE=/storage/extSdCard
🐸 LANG='en_US.UTF-8'
🐸 LD_PRELOAD=libsigchain.so:libNimsWrap.so
🐸 ANDROID_SOCKET_zygote=9
🐸 EMULATED_STORAGE_TARGET=/storage/emulated
🐸 ASEC_MOUNTPOINT=/mnt/asec
🐸 ANDROID_PROPERTY_WORKSPACE=8,0
🐸 PWD=/root
🐸 SYSTEMSERVERCLASSPATH=/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar
🐸 ANDROID_STORAGE=/storage
🐸 PROOT_TMP_DIR=/data/data/com.octo4a/files/bootstrap/tmp
🐸 ANDROID_BOOTLOGO=1
🐸 EMULATED_STORAGE_SOURCE=/mnt/shell/emulated
💻 >ls /
🐸 bin
🐸 dev
🐸 etc
🐸 home
🐸 lib
🐸 media
🐸 mnt
[... 473 lines cut ...]
at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:43)
at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:16)
at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:167)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [Processing /root/OctoPrint-OctoPrint-f027aaa  Ins

hgoldstein95, 2023-11-12:
Same issue for me, unfortunately. Galaxy S4 running Android 4.4.2.

ahsantahir01, 2023-11-16:
Any fix for this error

SparkyTheBoi, 2023-12-16:
I have the same issue on my Huawei union y538 running Android 5.1.1

ezxzeng, 2023-12-18:
same for samsung note 4 running android 6.0.1

Jbbs, 2024-01-22:
I'm running octo2a on a samsung S4 with lineage OS .

https://wiki.lineageos.org/devices/jfltexx/

dareno, 2024-02-26:
This looks like a duplicate of #435 : both seem to fail as a result of a deprecation warning: `The license_file parameter is deprecated, use license_files instead.`