feelfreelinux/octo4a #453: Error. Please help!

EBMX220, 2023-11-20:
I am running a Motorolla 5g uw ace, i am getting this 

<html>
<body>
<div class="nH" style="color: rgb(32, 33, 36); font-family: &quot;Google Sans&quot;, Roboto, RobotoDraft, Helvetica, Arial, sans-serif; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="aHU hx" style="background-color: transparent; color: rgb(34, 34, 34); min-width: 502px; position: relative; padding: 0px;"><div jslog="20686; u014N:xr6bB; 1:WyIjdGhyZWFkLWE6ci01MjE1MzA4MDU4OTA0Njk1NDAiXQ.." role="list"><div class="h7 ie" role="listitem" aria-expanded="true" tabindex="-1" style="padding-bottom: 0px; max-width: 100000px; clear: both; outline: none;"><div class="Bk" style="position: relative; margin-bottom: 0px; border-width: 0px; border-top-style: solid; border-right-style: initial; border-bottom-style: initial; border-left-style: initial; border-top-color: rgb(239, 239, 239); border-right-color: initial; border-left-color: initial; border-image: initial; border-bottom-color: initial; border-radius: 0px; float: left; width: 1576px;"><div class="G3 G2" style="border-top: none; padding-top: 0px; background-color: transparent; border-right: 0px; border-bottom: 0px rgba(100, 121, 143, 0.12); border-left: 0px; border-image: initial; border-radius: 0px; margin-bottom: 0px; margin-left: 0px; margin-right: 0px;"><div><div id=":ms"><div class="adn ads" data-message-id="#msg-a:r-4579388231180995356" data-legacy-message-id="18bed9fa1bc6a540" style="border-left: none; padding: 0px; display: flex;"><div class="gs" style="margin: 0px; padding: 0px 0px 20px; width: 1504px;"><div class=""><div id=":mv" class="ii gt" jslog="20277; u014N:xr6bB; 1:WyIjdGhyZWFkLWE6ci01MjE1MzA4MDU4OTA0Njk1NDAiXQ..; 4:WyIjbXNnLWE6ci00NTc5Mzg4MjMxMTgwOTk1MzU2Il0." style="direction: ltr; margin: 8px 0px 0px; padding: 0px; position: relative; font-size: 0.875rem;"><div id=":mu" class="a3s aiL " style="font: small / 1.5 Arial, Helvetica, sans-serif; overflow: hidden;"><div dir="auto">No bootstrap detected, proceeding with installation<div dir="auto">[image]<span> </span>Arch: aarch64</div><div dir="auto">[image]<span> </span>Downloading bootstrap 0.3.0 from<span> </span><a href="https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-aarch64.zip" target="_blank" data-saferedirecturl="https://www.google.com/url?q=https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-aarch64.zip&amp;source=gmail&amp;ust=1700585155314000&amp;usg=AOvVaw2zLEd9j6qI9BDH6uMaTqum" style="color: rgb(17, 85, 204);">https://github.com/<wbr>feelfreelinux/android-linux-<wbr>bootstrap/releases/download/0.<wbr>3.0/bootstrap-aarch64.zip</a></div><div dir="auto">[image]<span> </span>Request to<span> </span><a href="https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/ea14dee1-fef5-465a-9f22-4ff9e7631821?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231120%2Fus-east-1%2Fs3%2Faws4_request&amp;X-Amz-Date=20231120T164430Z&amp;X-Amz-Expires=300&amp;X-Amz-Signature=aa666e4618be190c104c1df143c7be4cfe8e4a3dbba5d68fe7a3b76232b2694d&amp;X-Amz-SignedHeaders=host&amp;actor_id=0&amp;key_id=0&amp;repo_id=352965812&amp;response-content-disposition=attachment%3B%20filename%3Dbootstrap-aarch64.zip&amp;response-content-type=application%2Foctet-stream" target="_blank" data-saferedirecturl="https://www.google.com/url?q=https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/ea14dee1-fef5-465a-9f22-4ff9e7631821?X-Amz-Algorithm%3DAWS4-HMAC-SHA256%26X-Amz-Credential%3DAKIAIWNJYAX4CSVEH53A%252F20231120%252Fus-east-1%252Fs3%252Faws4_request%26X-Amz-Date%3D20231120T164430Z%26X-Amz-Expires%3D300%26X-Amz-Signature%3Daa666e4618be190c104c1df143c7be4cfe8e4a3dbba5d68fe7a3b76232b2694d%26X-Amz-SignedHeaders%3Dhost%26actor_id%3D0%26key_id%3D0%26repo_id%3D352965812%26response-content-disposition%3Dattachment%253B%2520filename%253Dbootstrap-aarch64.zip%26response-content-type%3Dapplication%252Foctet-stream&amp;source=gmail&amp;ust=1700585155314000&amp;usg=AOvVaw3gkhjDj636Sv6UeR8qPgSb" style="color: rgb(17, 85, 204);">https://objects.<wbr>githubusercontent.com/github-<wbr>production-release-asset-<wbr>2e65be/352965812/ea14dee1-<wbr>fef5-465a-9f22-4ff9e7631821?X-<wbr>Amz-Algorithm=AWS4-HMAC-<wbr>SHA256&amp;X-Amz-Credential=<wbr>AKIAIWNJYAX4CSVEH53A%<wbr>2F20231120%2Fus-east-1%2Fs3%<wbr>2Faws4_request&amp;X-Amz-Date=<wbr>20231120T164430Z&amp;X-Amz-<wbr>Expires=300&amp;X-Amz-Signature=<wbr>aa666e4618be190c104c1df143c7be<wbr>4cfe8e4a3dbba5d68fe7a3b76232b2<wbr>694d&amp;X-Amz-SignedHeaders=host&amp;<wbr>actor_id=0&amp;key_id=0&amp;repo_id=<wbr>352965812&amp;response-content-<wbr>disposition=attachment%3B%<wbr>20filename%3Dbootstrap-<wbr>aarch64.zip&amp;response-content-<wbr>type=application%2Foctet-<wbr>stream</a><span> </span>returned status code 200</div><div dir="auto">[image]<span> </span>Bootstrap extracted, setting it up...</div><div dir="auto">[image]<span> </span>&gt;ls</div><div dir="auto">[image]<span> </span>add-user.sh</div><div dir="auto">[image]<span> </span>fake_proc_stat</div><div dir="auto">[image]<span> </span>install-bootstrap.sh</div><div dir="auto">[image]<span> </span>ioctlHook.so</div><div dir="auto">[image]<span> </span>root</div><div dir="auto">[image]<span> </span>rootfs.tar.xz</div><div dir="auto">[image]<span> </span>run-bootstrap.sh</div><div dir="auto">[image]<span> </span>&gt;chmod -R 700 .</div><div dir="auto">[image]<span> </span>&gt;sh install-bootstrap.sh</div><div dir="auto">[image]<span> </span>install-bootstrap.sh[8]: ../root/bin/minitar: can't execute: Permission denied</div><div dir="auto">[image]<span> </span>install-bootstrap.sh[13]: can't create bootstrap/etc/resolv.conf: No such file or directory</div><div dir="auto">[image]<span> </span>bootstrap ready, run with run-bootstrap.sh</div><div dir="auto">[image]<span> </span>&gt;sh add-user.sh octoprint</div><div dir="auto">[image]<span> </span>add-user.sh[5]: can't create bootstrap/etc/passwd: No such file or directory</div><div dir="auto">[image]<span> </span>An exception has occurred at:</div><div dir="auto">java.lang.RuntimeException: Process exited with error code 1. [add-user.sh[5]: can't create bootstrap/etc/passwd</div><div dir="auto"> at com.octo4a.utils.<wbr>ProcessUtilsKt.<wbr>waitAndPrintOutput(<wbr>ProcessUtils.kt:43)</div><div dir="auto"> at com.octo4a.utils.<wbr>ProcessUtilsKt.<wbr>waitAndPrintOutput$default(<wbr>ProcessUtils.kt:16)</div><div dir="auto"> at com.octo4a.repository.<wbr>BootstrapRepositoryImpl$<wbr>setupBootstrap$2.<wbr>invokeSuspend(<wbr>BootstrapRepository.kt:152)</div><div dir="auto"> at kotlin.coroutines.jvm.<wbr>internal.BaseContinuationImpl.<wbr>resumeWith(ContinuationImpl.<wbr>kt:33)</div><div dir="auto"> at io.ktor.util.pipeline.<wbr>SuspendFunctionGun.<wbr>resumeRootWith(<wbr>SuspendFunctionGun.kt:191)</div><div dir="auto"> at io.ktor.util.pipeline.<wbr>SuspendFunctionGun.loop(<wbr>SuspendFunctionGun.kt:147)</div><div dir="auto"> at io.ktor.util.pipeline.<wbr>SuspendFunctionGun.access$<wbr>loop(SuspendFunctionGun.kt:15)</div><div dir="auto"> at io.ktor.util.pipeline.<wbr>SuspendFunctionGun$<wbr>continuation$1.resumeWith(<wbr>SuspendFunctionGun.kt:93)</div><div dir="auto"> at kotlin.coroutines.jvm.<wbr>internal.BaseContinuationImpl.<wbr>resumeWith(ContinuationImpl.<wbr>kt:46)</div><div dir="auto"> at io.ktor.util.pipeline.<wbr>SuspendFunctionGun.<wbr>resumeRootWith(<wbr>SuspendFunctionGun.kt:191)</div><div dir="auto"> at io.ktor.util.pipeline.<wbr>SuspendFunctionGun.loop(<wbr>SuspendFunctionGun.kt:147)</div><div dir="auto"> at io.ktor.util.pipeline.<wbr>SuspendFunctionGun.access$<wbr>loop(SuspendFunctionGun.kt:15)</div><div dir="auto"> at io.ktor.util.pipeline.<wbr>SuspendFunctionGun$<wbr>continuation$1.resumeWith(<wbr>SuspendFunctionGun.kt:93)</div><div dir="auto"> at kotlin.coroutines.jvm.<wbr>internal.BaseContinuationImpl.<wbr>resumeWith(ContinuationImpl.<wbr>kt:46)</div><div dir="auto"> at kotlinx.coroutines.<wbr>DispatchedTask.run(<wbr>DispatchedTask.kt:106)</div><div dir="auto"> at kotlinx.coroutines.scheduling.<wbr>CoroutineScheduler.runSafely [... cut ...]
</div></div></div></div></div></div></div></div></div></div><div></div></div></div><div class="nH" style="color: rgb(32, 33, 36); font-family: &quot;Google Sans&quot;, Roboto, RobotoDraft, Helvetica, Arial, sans-serif; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><br class="Apple-interchange-newline">
</body>
</html>

EBMX220, 2023-11-20:
I got it.

SonOfSardaar, 2023-12-10:
At what percent does your installation fail ? you could try installing manually from the shell if it has gone past the installation of shell. Additionally it is worth trying a different ROM/Build for your phone. I tried different ones on HTC M8 and all ended up with different issues. But one of them took me farthest.

dangret, 2024-04-16:
I have the same problem in a moto G41, @SonOfSardaar bootstrap installation is the very beginning, shell has not been installed yet, I tried your suggestion without success

SonOfSardaar, 2024-04-16:
> I have the same problem in a moto G41, @SonOfSardaar bootstrap installation is the very beginning, shell has not been installed yet, I tried your suggestion without success

Have you tried different ROM ? I had to try 3-4 of them to find one that worked. It is likely to do with drivers available on these roms

feelfreelinux, 2024-05-17:
Please recheck on 2.0.0 release, possibly the new bootstrap changes resolved this issue