feelfreelinux/octo4a #461: Plugin Extras installed on latest version.  Still can't install pycryptodome or OctoApp plugin

sv0o, 2023-12-24:
this issue was closed, but I see the same exact output every time I try to install OctoApp plugin

https://github.com/feelfreelinux/octo4a/issues/321#issue-1477083145

My entries at the end of issue 321 describe the current state

sv0o, 2023-12-28:
saw an update regarding Octolapse not compiling correctly today and the error messages look the same to me.  I tried to add the path to cc1 and pip install pycryptodome but then I ran into the same 'include' error.  

Could there be a simpler way to update or fix the gcc build environment within octo4a?  

see https://github.com/FormerLurker/Octolapse/issues/900#issuecomment-1871410522

sv0o, 2024-02-08:
based on today's testing, I recommend avoiding  Octo4A on newer versions of Android or at least on the BLUphone running Android 11.  I have almost no issues at all installing plugins on the Jide Remix mini as I describe in https://github.com/feelfreelinux/octo4a/discussions/369

sv0o, 2024-05-20:
Good news:  with octo4A 2.0.0, I can install pycryptodome.  

Bad news: Can't install Pillow, upon which both OctoApp and Slicer Thumbnails depends

Building wheel for pycryptodome (pyproject.toml): started
Building wheel for pycryptodome (pyproject.toml): still running...
Building wheel for pycryptodome (pyproject.toml): still running...
Building wheel for pycryptodome (pyproject.toml): still running...
Building wheel for pycryptodome (pyproject.toml): finished with status 'done'
Created wheel for pycryptodome: filename=pycryptodome-3.20.0-cp35-abi3-linux_armv7l.whl size=2151777 sha256=95182a12b6bfd50d205136562d0e74d1428bacb19a439bf62f3b1b51982da639
Stored in directory: /tmp/pip-ephem-wheel-cache-9tcsz5e7/wheels/e3/98/5d/7d44287cd37ea6913aaa636057baac896346bb8716329ea2e8
Building wheel for pillow (pyproject.toml): started
error: subprocess-exited-with-error

× Building wheel for pillow (pyproject.toml) did not run successfully.
│ exit code: 1
╰─> [216 lines of output]
running bdist_wheel
running build
running build_py
creating build
creating build/lib.linux-armv7l-cpython-311
creating build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/WalImageFile.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/PcfFontFile.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageQt.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/PalmImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageGrab.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/Image.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageStat.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageMorph.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/PixarImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageDraw2.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/FtexImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/EpsImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/FliImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageChops.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageFont.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/JpegPresets.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/__init__.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/FontFile.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/IcoImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/FitsImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/SunImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageCms.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/PyAccess.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/MspImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageSequence.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/GbrImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/WebPImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/BmpImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/PcdImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ExifTags.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/DcxImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/_typing.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/TiffImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageTransform.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageShow.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/JpegImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/_version.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/ImageColor.py -> build/lib.linux-armv7l-cpython-311/PIL
copying src/PIL/BlpImagePlugin.py -> build/lib.linux-armv7l-cpython-311/PIL
[... 158 lines cut ...]
Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html

[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
ERROR: Failed building wheel for pillow
ERROR: Could not build wheels for pillow, which is required to install pyproject.toml-based projects
Building wheel for pillow (pyproject.toml): finished with status 'error'
Successfully built OctoApp pycryptodome
Failed to build pillow
Looks like the plugin was already installed. Forcing a reinstall.
/home/octoprint/octoprint-venv/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpzgkgktso/OctoApp-Plugin-release.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps
Processing /tmp/tmpzgkgktso/OctoApp-Plugin-release.zip
Preparing metadata (setup.py): started
[... cut ...]
Building wheels for collected packages: OctoApp
Building wheel for OctoApp (setup.py): started
Building wheel for OctoApp (setup.py): finished with status 'done'
Created wheel for OctoApp: filename=OctoApp-2.0.11-py3-none-any.whl size=98116 sha256=44e63f899bcd88cc12a29294cbec154e0ec0ee12c57f4b19268823cc61ced6d5
Stored in directory: /tmp/pip-ephem-wheel-cache-f8gkz_ky/wheels/8a/63/65/c4cfdebef43c662b6885c308b8f51e00bd13c24557dff90c0c
Successfully built OctoApp
Installing collected packages: OctoApp
Successfully installed OctoApp-2.0.11
Done!

sv0o, 2024-05-24:
Today, I backed up octoprint 1.10.0 and followed the Octo4A prompts in android to do its update to 2.0.0

I can report that I sucessfully updated octoprint to 1.10.1 and got back all of my plugins including octolapse.  

I let the octo4a installer do all the work and chose the 'ac1' recommended octoprint version.   I wasn't able to recover the full back up at first, but that's because the 'install plugin extras' had been reset.  Once that was installed, I could install octolapse 0.4.5 through the octoprint plugin installer.  

Thanks for the great app and the recent updates!

Portaill, 2024-05-25:
> Today, I backed up octoprint 1.10.0 and followed the Octo4A prompts in android to do its update to 2.0.0
>
> I can report that I sucessfully updated octoprint to 1.10.1 and got back all of my plugins [...]

Hello there how did u roll back, i got the same thing for octoeverywhere

thanks you

sv0o, 2024-05-26:
> Hello there how did u roll back, i got the same thing for octoeverywhere
>
> thanks you

to answer your question, the octo4a install I had was 1.2.6 on a Jide Remix Mini (android 7, I think) and when I connected a monitor to look at its home screen, I actually saw that octo4a had popped up a message saying that there was a new version to download.  I clicked a 'upgrade' link and it took me to the download page on github.  I downloaded the new octo4a and followed the prompts.  

I should note that I have also been trying to upgrade a BLU phone (android 11)  to octoprint 1.10.1, but when I tried to -after my last post- , I did run into the problem installing octoeverywhere and octolapse.  I think the pillow wheel building process is not able to complete, but I haven't spent much time digging further.

sv0o, 2024-10-27:
Thank you!  

I've been trying to get an LG V10 to update with the pycryptodome problem stopping me but this worked great.

apk add zlib-dev
apk add libjpeg-turbo-dev

Acetabulo, 2025-01-16:
> Thank you!  
> 
> I've been trying to get an LG V10 to update with the pycryptodome problem stopping me but this worked great.
> 
> apk add zlib-dev
> apk add libjpeg-turbo-dev
> 
> further update: 
> the BLU phone (android 11) that would not update before now works fine with all my plugins. 
> 
> I updated the Octo4A apk, then apk-added  zlib and libjepeg as above, then had no issues with updating Octoprint in the settings/octoprint/software updater.  From there, I had no issues adding my missing plugins in the plugin manager.

Acetabulo, 2025-01-16:
> Thank you!
>
> I've been trying to get an LG V10 to update with the pycryptodome problem stopping me but this worked great.
>
> apk add zlib-dev
> apk add libjpeg-turbo-dev
>
> further update:
> the BLU phone [...]

Hi, is it possible to expand this explanation a bit? I'm new and I have a similar problem but I don't know how to add this to the octo4a alpine container. Sorry but English is not my native language.

sv0o, 2025-01-16:
> Hi, is it possible to expand this explanation a bit? I'm new and I have a similar problem but I don't know how to add this to the octo4a alpine container. Sorry but English is not my native language.

If you can ssh into the octo4a alpine as root,  run the two apk add statements:

apk add zlib-dev
apk add libjpeg-turbo-dev

In the post by [akei-ai](https://github.com/akei-ai), the possible reason for the failure is described.