feelfreelinux/octo4a #473: I'm getting error when installing.

gadonner, 2024-04-24:
This is not my first time but I'm stuck right now.

gustaf@gustaf-laptop:~$ ssh root@192.186.0.24 -p 8022
ssh root@192.168.0.24 -p 8022
ssh: connect to host 192.186.0.24 port 8022: Connection timed out
gustaf@gustaf-laptop:~$ ssh root@192.168.0.24 -p 8022
root@192.168.0.24's password: 
Welcome to Alpine!

The Alpine Wiki contains a large amount of how-to guides and general
information about administrating Alpine systems.
See <https://wiki.alpinelinux.org/>.

You can setup the system with the command: setup-alpine

You may change this message by editing /etc/motd.

localhost:~# cd octoprint
-ash: cd: can't cd to octoprint: No such file or directory
localhost:~# dir
-ash: dir: not found
localhost:~# pip3 install --upgrade setuptools wheel tornado --no-cache-dir

Requirement already satisfied: setuptools in /usr/lib/python3.10/site-packages (69.5.1)
Requirement already satisfied: wheel in /usr/lib/python3.10/site-packages (0.43.0)
Requirement already satisfied: tornado in /usr/lib/python3.10/site-packages (6.4)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
localhost:~# 
localhost:~# curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/mast
er/scripts/setup-octo4a.sh | bash -s
OK: 390 MiB in 148 packages
Collecting packaging
  Using cached packaging-24.0-py3-none-any.whl (53 kB)
Installing collected packages: packaging
Successfully installed packaging-24.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Collecting https://github.com/feelfreelinux/octo4a-argon2-mock/archive/main.zip
  Using cached https://github.com/feelfreelinux/octo4a-argon2-mock/archive/main.zip
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
localhost:~# unzip octoprint.zip
-ash: unzip: Permission denied
localhost:~# cd octo*
-ash: cd: can't cd to octoprint.zip: Not a directory
localhost:~# cd Octo*
localhost:~/OctoPrint-OctoPrint-fed2dd7# pip3 install Octoprint
Collecting Octoprint
  Using cached OctoPrint-1.10.0-py2.py3-none-any.whl (3.4 MB)
Collecting Flask-Login<0.7,>=0.6.3
  Using cached Flask_Login-0.6.3-py3-none-any.whl (17 kB)
Collecting psutil<6,>=5.9.8
  Using cached psutil-5.9.8.tar.gz (503 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting PyYAML<7,>=6.0.1
  Using cached PyYAML-6.0.1.tar.gz (125 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting websocket-client==1.6.1
  Using cached websocket_client-1.6.1-py3-none-any.whl (56 kB)
Collecting pylru<2,>=1.2.1
  Using cached pylru-1.2.1-py3-none-any.whl (16 kB)
Collecting sentry-sdk<2,>=1.40.0
  Using cached sentry_sdk-1.45.0-py2.py3-none-any.whl (267 kB)
Collecting requests<3,>=2.31.0
  Using cached requests-2.31.0-py3-none-any.whl (62 kB)
[... 449 lines cut ...]
            building '_cffi_backend' extension
            creating build/temp.linux-armv7l-cpython-310
            creating build/temp.linux-armv7l-cpython-310/src
            creating build/temp.linux-armv7l-cpython-310/src/c
            gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -DFFI_BUILDING=1 -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python3.10 -c src/c/_cffi_backend.c -o build/temp.linux-armv7l-cpython-310/src/c/_cffi_backend.o
            gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
            compilation terminated.
            error: command 'gcc' failed with exit code 1
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for cffi
      Failed to build cffi
      ERROR: Could not build wheels for cffi, which is required to install pyproject.toml-based projects
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
localhost:~/OctoPrint-OctoPrint-fed2dd7#

hounslg, 2024-04-25:
The issue is related to the new release of Octoprint (1.10).  A fix or work around is required.

gadonner, 2024-04-25:
Should I try an older version?

hounslg, 2024-04-26:
You'll need to wait for an update to the app unfortunately.  It downloads the latest version of Octoprint.

gadonner, 2024-04-28:
Installed new release. Corrupted install but seems to work.

gadonner, 2024-04-28:
Sorry. When I restarted phone it was still install corrupted

feelfreelinux, 2024-05-17:
Released https://github.com/feelfreelinux/octo4a/releases/tag/2.0.0 which should resolve this issue :)

closed 2024-05-17 by feelfreelinux