feelfreelinux/octo4a #474: OctoPrint 1.9.x --> 1.10 Not Updating

atkeeler, 2024-04-24:
I've been unable to update OctoPrint on Octo4a from 1.9 to 1.10, continues to give me an error message. When looking into it, I could that it has something to do with Python. I tried to update after clearing the app data to see if that would help, but it gave me a "FATAL ERROR" message when it was installing the dependencies onto my Samsung Galaxy S20 FE. Error log is attached as an text file.
[Octo4a Log.txt](https://github.com/feelfreelinux/octo4a/files/15099780/Octo4a.Log.txt)

darkzeroat, 2024-04-25:
same issue here with fire hd 8 plus
any help is appreciated <3

hounslg, 2024-04-25:
Same here, updating a Pixel which was running 1.9.

Same error as the log file already attached.

alufers, 2024-04-26:
Will work on this tomorrow

sm3g, 2024-04-26:
Also failing for me, updating a Samsung Galaxy S8.  I'm getting a different error regarding ethtool.h being missing. I updated pip just prior to attempting to update OctoPrint, since the release notes mentioned that.

"psutil/_psutil_linux.c:13:10: fatal error: linux/ethtool.h: No such file or directory"
[plugin_softwareupdate_console.log](https://github.com/feelfreelinux/octo4a/files/15135950/plugin_softwareupdate_console.log)

Mario1968, 2024-04-27:
> Will work on this tomorrow

Hi alufers, is possible for now (as a workaround) to force download only Octoprint 1.9.x instead of the last Octoprint?
TYIA
Mario

feelfreelinux, 2024-04-27:
Just released octo4a 1.2.6 which should resolve this issue :)

FrogoYT, 2024-04-27:
The 1.2.6 didn't work for me, even in safe mode

Mario1968, 2024-04-28:
> Just released octo4a 1.2.6 which should resolve this issue :)

Thank you! it worked for me (Sasung S8 unmodified)

hounslg, 2024-04-28:
Just updated my pixel 3 and all working fine.

Thank you for the speed update.

joejtrsmoe, 2024-04-28:
This didn't help with my s7.  Similar issues to when I first set it up. Thanks though. Love octo4a.

remolyy, 2024-05-01:
Didn't help on Samsung s10+

remolyy, 2024-05-01:
[Error log.txt](https://github.com/feelfreelinux/octo4a/files/15176024/Error.log.txt)
psutil not instaled

sACLeFR, 2024-05-06:
Hello, updated Octo4a to 1.2.6 but still unable to update Octoprint due to psutil also

farfromuman, 2024-05-06:
I was not able to upgrade due to errors during the upgrade with 1.2.6.  I made a backup in OctoPrint 1.9, uninstalled everything and restored the backup once 1.10 was freshly installed it even downloaded all my plugins including psutil.

sACLeFR, 2024-05-06:
> I was not able to upgrade due to errors during the upgrade with 1.2.6. I made a backup in OctoPrint 1.9, uninstalled everything and restored the backup once 1.10 was freshly installed it even downloaded all my plugins including psutil.

It worked ! Thank you !

zaventh, 2024-05-07:
See my comment here to solve this manually https://github.com/OctoPrint/OctoPrint/issues/4998#issuecomment-2098983838

sv0o, 2024-05-09:
> See my comment here to solve this manually [OctoPrint/OctoPrint#4998 (comment)](https://github.com/OctoPrint/OctoPrint/issues/4998#issuecomment-2098983838)

I was excited to try this, but I think I'm now chasing my own tail:

apk del is saying 'no such package' for ps3-psutil or for psutil

the updater in octoprint and a manual pip3 install keeps saying

 _Attempting uninstall: psutil
    Found existing installation: psutil 5.9.4
ERROR: Cannot uninstall 'psutil'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall._

so I tried:
pip3 install --ignore-installed psutil

_Collecting psutil
  Using cached psutil-5.9.8-cp310-abi3-linux_aarch64.whl
Installing collected packages: psutil
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
octoprint 1.9.3 requires zeroconf==0.39.4, but you have zeroconf 0.132.2 which is incompatible.
Successfully installed psutil-5.9.4_

Like you, I didn't want to wipe out everything and start fresh, but that's looking like the only option.

zaventh, 2024-05-09:
> > See my comment here to solve this manually [OctoPrint/OctoPrint#4998 (comment)](https://github.com/OctoPrint/OctoPrint/issues/4998#issuecomment-2098983838)
> 
> I was excited to try this, but I think I'm now chasing my own tail:
> 
> apk del is saying 'no such package' for ps3-psutil or for psutil
> 
> the updater in octoprint and a manual pip3 install keeps saying
> 
> _Attempting uninstall: psutil Found existing installation: psutil 5.9.4 ERROR: Cannot uninstall 'psutil'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall._
> 
> so I tried: pip3 install --ignore-installed psutil
> 
> _Collecting psutil Using cached psutil-5.9.8-cp310-abi3-linux_aarch64.whl Installing collected packages: psutil ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts. octoprint 1.9.3 requires zeroconf==0.39.4, but you have zeroconf 0.132.2 which is incompatible. Successfully installed psutil-5.9.4_
> 
> Like you, I didn't want to wipe out everything and start fresh, but that's looking like the only option. Or is there a way to uninstall psutil using the same method that originally installed it? It was a couple of octo4a versions ago I think.

Sorry, typo. It's `py3-psutil` not `ps3-psutil` so `apk del py3-psutil`

sv0o, 2024-05-09:
I finally made it through the update process, thanks.  I'll try to do some test prints tonight or tomorrow

Mario1968, 2024-05-17:
> See my comment here to solve this manually [OctoPrint/OctoPrint#4998 (comment)](https://github.com/OctoPrint/OctoPrint/issues/4998#issuecomment-2098983838)

Hi, Thank you for this fix, I was able to update to Octoprint 1.10 with the new APK 1.2.6 but now with the new release 1.10.1 the psutil install/uninstall problem is still here, with your suggestion I finally was able to update to 1.10.1. Thank You again.

feelfreelinux, 2024-05-17:
Released https://github.com/feelfreelinux/octo4a/releases/tag/2.0.0 which should resolve this issue :)

closed 2024-05-17 by feelfreelinux