feelfreelinux/octo4a #475: error installing 1.2.6 on fire hd 8 (android 9)

darkzeroat, 2024-04-27:
not sure wheres the problem
the updated installer works on my newer tablet but not on my old fire hd 8

[octo4a log.txt](https://github.com/feelfreelinux/octo4a/files/15139932/octo4a.log.txt)

hope this helps you resolve the issue if possible

matt012345, 2024-05-06:
I believe I am having the same problem when attempting to download octo4a on my kindle fire HD8. I have tried version 1.2.6 three times as well as various older versions.
[octo4a.log](https://github.com/feelfreelinux/octo4a/files/15226906/octo4a.log)

hhocker, 2024-05-14:
@darkzeroat and @matt012345 the new 1.10.1 release rolls back netifaces back from the version requiring Rust. you might see if it will allow it to upgrade on your machine now. I am struggling to get it to install for me but will report if I figure out the issue.

hhocker, 2024-05-15:
Ok. I have figured out that if you have an existing working installation it should work. The rust error should be able to be bypassed if you dont get an error about cc1 not being found. 

However I had issues compiling CFFI for the argon2 framework with GCC errors so here are the steps I followed from a fresh install:
1. First download Octo4A 1.2.6 and install it. 
2. Attempt to install the latest OctoPrint, this step will likely fail
3. Setup SSH and remote into the alpine environment
4. I had to upgrade the base alpine system to a newer version from 3.17 to 3.18, had problems with 3.19. (steps abbreviated  from here https://wiki.alpinelinux.org/wiki/Upgrading_Alpine)
```
1. sed -i 's/3\.17/3\.18/g' /etc/apk/repositories
2. apk update
3. apk add --upgrade apk-tools
4. apk upgrade --available
6. sync
```
5. Run ``` apk add py3-argon2-cffi ``` to install the argon2-cffi package requirements to prevent needing to build with gcc. GCC has been having issues finding cc1 for me.
6. You should now be able to go into the downloaded octoprint in the root folder and install 1.10.1 
```
cd /root/Octo*
pip install . 
```
7. Once you are done with the build and install step reboot your phone / tablet. Once that reboot finished try launching the app and seeing if OctoPrint server comes online. It should successfully come online and be ready to go.

kekkodance, 2024-05-16:
do you have a solution for psutil doing the same cc1 error when upgrading from the octoprint ui?

hhocker, 2024-05-16:
No I sure dont, I am not sure what is causing errors with the CC1 compiler. I never could recreate a fix. 

So I would enable web ssh in the app or SSH in via terminal then try the following:

```
<ins>/_!_\</ins> Untested <ins>/_!_\</ins>
apk add py3-psutil --repository="http://dl-cdn.alpinelinux.org/alpine/v3.19/community"
```
To pull the necessary 5.9.6 that 1.10 is calling out.

kekkodance, 2024-05-17:
@hhocker i can't manage to install octolapse like this, cc1plus is erroring out and i can't figure out what package is doing it.

hhocker, 2024-05-17:
Interesting. I hadn't considered the issue of building plugins. I presumed they were some sort of precompiled scripts. I am hoping that Filip is successful in launching the new version soon. It seems odd that there have been a rash of GCC errors crop up recently.

feelfreelinux, 2024-05-17:
Well, the new release should be here later today. Just writing some release notes, and testing out the changes a bit more :)

feelfreelinux, 2024-05-17:
Released https://github.com/feelfreelinux/octo4a/releases/tag/2.0.0 which should resolve this issue :)

closed 2024-05-17 by feelfreelinux