feelfreelinux/octo4a #476: Can't install new apk, old version worked fine

koruptdata, 2024-04-27:
Hello, I'm trying to install the new version so I can update octoprint. But the install fails every time on the same phone where it used to work perfectly. 

Here is my log I had to send over email to myself so sorry if it's got weird formating.

[octo4a-log.txt](https://github.com/feelfreelinux/octo4a/files/15140278/octo4a-log.txt)

hhocker, 2024-05-16:
@koruptdata, This is likely due to the 1.10._0_ version requiring netifaces2 which is built with rust. The buildroot / bootstrap I dont believe comes with it. The new 1.10._1_ has rolled back that dependency and this should be fixed.

feelfreelinux, 2024-05-17:
Released https://github.com/feelfreelinux/octo4a/releases/tag/2.0.0 which should resolve this issue :)

closed 2024-05-17 by feelfreelinux