feelfreelinux/octo4a #480: Please help, can't get octoprint to install

JNightwind, 2024-05-02:
When I open the app and click install octoprint it starts to download and then this it gives a "Fatal Error" message with the following log.

💻 No bootstrap detected, proceeding with installation
💻 Arch: aarch64
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-aarch64.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/ea14dee1-fef5-465a-9f22-4ff9e7631821?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240502%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240502T044512Z&X-Amz-Expires=300&X-Amz-Signature=a1c3b4c90a6a00071b01fa7552cba7ea97eab71049b87e5a165823adc8a5595f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-aarch64.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 An exception has occurred at:
java.lang.RuntimeException: Unable to create directory: /data/data/com.octo4a/files/bootstrap-staging/root
	at com.octo4a.repository.BootstrapRepositoryImpl.ensureDirectoryExists(BootstrapRepository.kt:199)
	at com.octo4a.repository.BootstrapRepositoryImpl.access$ensureDirectoryExists(BootstrapRepository.kt:38)
	at com.octo4a.repository.BootstrapRepositoryImpl$setupBootstrap$2.invokeSuspend(BootstrapRepository.kt:120)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Unable to create directory: /data/data/com.octo4a/files/bootstrap-staging/root

I'm not sure what to don't fix this. The app has permission to use storage so I don't know what the issue is.

hhocker, 2024-05-16:
@JNightwind, do you still have enough storage on your internal card? Octo4A expands out ~200mb depending on if you add a bunch of plugins or do advanced things via the SSH system. 

Otherwise check to see if the application may need some addtional permissions as it is sideloaded from the internet and not the Play Store

feelfreelinux, 2024-05-17:
Seems to be an odd permission issue - please recheck on version 2.0, as the proot bootstrap logic was changed quite a bit in that release.