feelfreelinux/octo4a #481: Build Error with cffi

SparkyTheBoi, 2024-05-03:
I get build errors when trying to install Octo4a. Seems to be related to cffi? 

```
🐸 Collecting argon2-cffi-bindings
🐸   Downloading argon2-cffi-bindings-21.2.0.tar.gz (1.8 MB)
🐸      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.8/1.8 MB 3.4 MB/s eta 0:00:00
🐸   Installing build dependencies: started
🐸   Installing build dependencies: still running...
🐸   Installing build dependencies: still running...
🐸   Installing build dependencies: finished with status 'error'
🐸   error: subprocess-exited-with-error
🐸  
🐸   × pip subprocess to install build dependencies did not run successfully.
🐸   │ exit code: 1
🐸   ╰─> [89 lines of output]
🐸       Collecting setuptools>=45
🐸         Using cached setuptools-69.5.1-py3-none-any.whl (894 kB)
🐸       Collecting setuptools_scm>=6.2
🐸         Downloading setuptools_scm-8.0.4-py3-none-any.whl (42 kB)
🐸            ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 42.1/42.1 kB 155.4 kB/s eta 0:00:00
🐸       Collecting wheel
🐸         Using cached wheel-0.43.0-py3-none-any.whl (65 kB)
🐸       Collecting cffi>=1.0.1
🐸         Downloading cffi-1.16.0.tar.gz (512 kB)
🐸            ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 512.9/512.9 kB 926.6 kB/s eta 0:00:00
🐸         Installing build dependencies: started
🐸         Installing build dependencies: finished with status 'done'
🐸         Getting requirements to build wheel: started
🐸         Getting requirements to build wheel: finished with status 'done'
🐸         Installing backend dependencies: started
🐸         Installing backend dependencies: finished with status 'done'
🐸         Preparing metadata (pyproject.toml): started
🐸         Preparing metadata (pyproject.toml): finished with status 'done'
🐸       Collecting tomli>=1
🐸         Downloading tomli-2.0.1-py3-none-any.whl (12 kB)
🐸       Collecting packaging>=20
🐸         Using cached packaging-24.0-py3-none-any.whl (53 kB)
🐸       Collecting typing-extensions
🐸         Downloading typing_extensions-4.11.0-py3-none-any.whl (34 kB)
🐸       Collecting pycparser
🐸         Downloading pycparser-2.22-py3-none-any.whl (117 kB)
🐸            ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 117.6/117.6 kB 845.5 kB/s eta 0:00:00
🐸       Building wheels for collected packages: cffi
🐸         Building wheel for cffi (pyproject.toml): started
🐸         Building wheel for cffi (pyproject.toml): finished with status 'error'
🐸         error: subprocess-exited-with-error
🐸      
🐸         × Building wheel for cffi (pyproject.toml) did not run successfully.
🐸         │ exit code: 1
🐸         ╰─> [48 lines of output]
🐸      
🐸                 No working compiler found, or bogus compiler options passed to
🐸                 the compiler from Python's standard "distutils" module.  See
🐸                 the error messages above.  Likely, the problem is not related
🐸                 to CFFI but generic to the setup.py of any Python package that
🐸                 tries to compile C code.  (Hints: on OS/X 10.8, for errors about
🐸                 -mno-fused-madd see http://stackoverflow.com/questions/22313407/
🐸                 Otherwise, see https://wiki.python.org/moin/CompLangPython or
🐸                 the IRC channel #python on irc.libera.chat.)
🐸      
🐸                 Trying to continue anyway.  If you are trying to install CFFI from
🐸                 a build done in a different context, you can ignore this warning.
🐸      
🐸             running bdist_wheel
🐸             running build
🐸             running build_py
🐸             creating build
🐸             creating build/lib.linux-armv7l-cpython-310
🐸             creating build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/__init__.py -> build/lib.linux-armv7l-cpython-310/cffi
[... 49 lines cut ...]
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [WARNING: linker: ./root/bin/proot: unused DT entr
```
Here's the whole file if needed! <3
[Octo4a.txt](https://github.com/feelfreelinux/octo4a/files/15196125/Octo4a.txt)

revcrams, 2024-05-05:
Ditto
💻 No bootstrap detected, proceeding with installation
💻 Arch: arm
💻 Downloading bootstrap 0.3.0 from https://github.com/feelfreelinux/android-linux-bootstrap/releases/download/0.3.0/bootstrap-armv7a.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/352965812/2cd65898-00b6-4b5a-bd0f-7a036eff0c4f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240505%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240505T200741Z&X-Amz-Expires=300&X-Amz-Signature=741c7abdac0536a1ec14977e742ca01f880c8352ce41e00b6c65963104a1da50&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=352965812&response-content-disposition=attachment%3B%20filename%3Dbootstrap-armv7a.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap extracted, setting it up...
💻 >ls
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 root-pre5
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
💻 >chmod -R 700 .
💻 >sh install-bootstrap.sh
🐸 bootstrap ready, run with run-bootstrap.sh
💻 >sh add-user.sh octoprint
💻 >cat /etc/motd
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Welcome to Alpine!
🐸 
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <https://wiki.alpinelinux.org/>.
🐸 
🐸 You can setup the system with the command: setup-alpine
🐸 
🐸 You may change this message by editing /etc/motd.
🐸 
💻 >env
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 EXTRA_BIND=-b /storage/emulated/0/Android/data/com.octo4a/files:/root -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a -b /data/data/com.octo4a/files/bootstrap/ioctlHook.so:/home/octoprint/ioctlHook.so
🐸 USER=root
🐸 EXTERNAL_STORAGE=/sdcard
🐸 PROOT_L2S_DIR=/data/data/com.octo4a/files/bootstrap/bootstrap/.proot.meta
🐸 SHLVL=1
🐸 HOME=/root
🐸 ANDROID_ASSETS=/system/app
🐸 BOOTCLASSPATH=/system/framework/core-oj.jar:/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/bouncycastle.jar:/system/framework/apache-xml.jar:/system/framework/legacy-test.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/android.hidl.base-V1.0-java.jar:/system/framework/android.hidl.manager-V1.0-java.jar:/system/framework/com.lge.mdm.jar:/system/framework/com.lge.fmradio.nablabs.jar:/system/framework/qcom.fmradio.jar:/system/framework/QPerformance.jar:/system/framework/WfdCommon.jar:/system/framework/oem-services.jar:/system/framework/tcmiface.jar:/system/framework/telephony-ext.jar
🐸 _=./root/bin/proot
🐸 DOWNLOAD_CACHE=/data/cache
🐸 ANDROID_DATA=/data
🐸 PATH=/sbin:/usr/sbin:/bin:/usr/bin
🐸 ANDROID_ROOT=/system
🐸 LANG='en_US.UTF-8'
🐸 ANDROID_SOCKET_zygote=10
🐸 ASEC_MOUNTPOINT=/mnt/asec
🐸 PWD=/root
🐸 SYSTEMSERVERCLASSPATH=/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar:/system/framework/com.android.location.provider.jar
🐸 ANDROID_STORAGE=/storage
🐸 PROOT_TMP_DIR=/data/data/com.octo4a/files/bootstrap/tmp
🐸 ANDROID_BOOTLOGO=1
💻 >ls /
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 bin
🐸 dev
🐸 etc
🐸 home
🐸 lib
🐸 media
🐸 mnt
🐸 opt
🐸 proc
🐸 root
🐸 run
🐸 sbin
🐸 srv
🐸 storage
[... 524 lines cut ...]
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:43)
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:16)
	at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:168)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.UndispatchedCoroutine.afterResume(Builders.common.kt:265)
	at kotlinx.coroutines.AbstractCoroutine.resumeWith(AbstractCoroutine.kt:113)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [proot warning: can't sanitize binding "/data/data

revcrams, 2024-05-05:
Ditto on the error - installing octo4a 1.2.6 on an android 8.1 LG Q6

Downloading pycparser-2.22-py3-none-any.whl (117 kB)
🐸            ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 117.6/117.6 kB 225.8 kB/s eta 0:00:00
🐸       Building wheels for collected packages: cffi
🐸         Building wheel for cffi (pyproject.toml): started
🐸         Building wheel for cffi (pyproject.toml): finished with status 'error'
🐸         error: subprocess-exited-with-error
🐸      
🐸         × Building wheel for cffi (pyproject.toml) did not run successfully.
🐸         │ exit code: 1
🐸         ╰─> [48 lines of output]
🐸      
🐸                 No working compiler found, or bogus compiler options passed to
🐸                 the compiler from Python's standard "distutils" module.  See
🐸                 the error messages above.  Likely, the problem is not related
🐸                 to CFFI but generic to the setup.py of any Python package that
🐸                 tries to compile C code.  (Hints: on OS/X 10.8, for errors about
🐸                 -mno-fused-madd see http://stackoverflow.com/questions/22313407/
🐸                 Otherwise, see https://wiki.python.org/moin/CompLangPython or
🐸                 the IRC channel #python on irc.libera.chat.)
🐸      
🐸                 Trying to continue anyway.  If you are trying to install CFFI from
🐸                 a build done in a different context, you can ignore this warning.
🐸      
🐸             running bdist_wheel
🐸             running build
🐸             running build_py
🐸             creating build
🐸             creating build/lib.linux-armv7l-cpython-310
🐸             creating build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/vengine_gen.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/pkgconfig.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/api.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/recompiler.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/model.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/verifier.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/_shimmed_dist_utils.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/cffi_opcode.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/vengine_cpy.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/__init__.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/error.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/commontypes.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/ffiplatform.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/backend_ctypes.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/cparser.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/setuptools_ext.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/lock.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/_imp_emulation.py -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/_cffi_include.h -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/parse_c_type.h -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/_embedding.h -> build/lib.linux-armv7l-cpython-310/cffi
🐸             copying src/cffi/_cffi_errors.h -> build/lib.linux-armv7l-cpython-310/cffi
🐸             running build_ext
🐸             building '_cffi_backend' extension
🐸             creating build/temp.linux-armv7l-cpython-310
🐸             creating build/temp.linux-armv7l-cpython-310/src
🐸             creating build/temp.linux-armv7l-cpython-310/src/c
🐸             gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -DFFI_BUILDING=1 -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python3.10 -c src/c/_cffi_backend.c -o build/temp.linux-armv7l-cpython-310/src/c/_cffi_backend.o
🐸             error: command 'gcc' failed: No such file or directory
🐸             [end of output]
🐸      
🐸         note: This error originates from a subprocess, and is likely not a problem with pip.
🐸         ERROR: Failed building wheel for cffi
🐸       Failed to build cffi
🐸       ERROR: Could not build wheels for cffi, which is required to install pyproject.toml-based projects
🐸       [end of output]
🐸  
🐸   note: This error originates from a subprocess, and is likely not a problem with pip.
🐸 error: subprocess-exited-with-error
🐸
🐸 × pip subprocess to install build dependencies did not run successfully.
🐸 │ exit code: 1
🐸 ╰─> See above for output.

sv0o, 2024-05-07:
Mine got stuck around the same place (cffi) trying to upgrade from octoprint 1.9.3.  This is with octo4a 1.2.6 on a Jide Remix Mini 

2024-04-30 21:19:06,300 > Building wheel for psutil (pyproject.toml): started
2024-04-30 21:19:14,463 ! error: subprocess-exited-with-error
2024-04-30 21:19:14,469 ! 
2024-04-30 21:19:14,472 ! × Building wheel for psutil (pyproject.toml) did not run successfully.
2024-04-30 21:19:14,479 ! │ exit code: 1
2024-04-30 21:19:14,482 ! ╰─> [55 lines of output]
2024-04-30 21:19:14,486 ! /tmp/tmpdlscfc63.c:1:10: fatal error: linux/ethtool.h: No such file or directory
2024-04-30 21:19:14,488 ! 1 | #include <linux/ethtool.h>
2024-04-30 21:19:14,491 ! |          ^~~~~~~~~~~~~~~~~
2024-04-30 21:19:14,494 ! compilation terminated.
2024-04-30 21:19:14,498 ! running bdist_wheel
2024-04-30 21:19:14,502 ! running build
2024-04-30 21:19:14,508 ! running build_py
2024-04-30 21:19:14,512 ! creating build
2024-04-30 21:19:14,516 ! creating build/lib.linux-aarch64-cpython-310
2024-04-30 21:19:14,519 ! creating build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,522 ! copying psutil/__init__.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,526 ! copying psutil/_common.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,529 ! copying psutil/_compat.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,532 ! copying psutil/_psaix.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,535 ! copying psutil/_psbsd.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,538 ! copying psutil/_pslinux.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,541 ! copying psutil/_psosx.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,544 ! copying psutil/_psposix.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,548 ! copying psutil/_pssunos.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,552 ! copying psutil/_pswindows.py -> build/lib.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,555 ! creating build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,558 ! copying psutil/tests/__init__.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,560 ! copying psutil/tests/__main__.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,564 ! copying psutil/tests/runner.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,567 ! copying psutil/tests/test_aix.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,570 ! copying psutil/tests/test_bsd.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,573 ! copying psutil/tests/test_connections.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,576 ! copying psutil/tests/test_contracts.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,580 ! copying psutil/tests/test_linux.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,584 ! copying psutil/tests/test_memleaks.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,587 ! copying psutil/tests/test_misc.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,591 ! copying psutil/tests/test_osx.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,594 ! copying psutil/tests/test_posix.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,598 ! copying psutil/tests/test_process.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,603 ! copying psutil/tests/test_process_all.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,606 ! copying psutil/tests/test_sunos.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,609 ! copying psutil/tests/test_system.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,612 ! copying psutil/tests/test_testutils.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,616 ! copying psutil/tests/test_unicode.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,620 ! copying psutil/tests/test_windows.py -> build/lib.linux-aarch64-cpython-310/psutil/tests
2024-04-30 21:19:14,622 ! running build_ext
2024-04-30 21:19:14,626 ! building 'psutil._psutil_linux' extension
2024-04-30 21:19:14,628 ! creating build/temp.linux-aarch64-cpython-310
2024-04-30 21:19:14,631 ! creating build/temp.linux-aarch64-cpython-310/psutil
2024-04-30 21:19:14,636 ! creating build/temp.linux-aarch64-cpython-310/psutil/arch
2024-04-30 21:19:14,639 ! creating build/temp.linux-aarch64-cpython-310/psutil/arch/linux
2024-04-30 21:19:14,642 ! gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -g -O2 -Os -g -O2 -Os -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -DPSUTIL_POSIX=1 -DPSUTIL_SIZEOF_PID_T=4 -DPSUTIL_VERSION=598 -DPy_LIMITED_API=0x03060000 -DPSUTIL_ETHTOOL_MISSING_TYPES=1 -DPSUTIL_LINUX=1 -I/usr/include/python3.10 -c psutil/_psutil_common.c -o build/temp.linux-aarch64-cpython-310/psutil/_psutil_common.o
2024-04-30 21:19:14,645 ! gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -g -O2 -Os -g -O2 -Os -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -DPSUTIL_POSIX=1 -DPSUTIL_SIZEOF_PID_T=4 -DPSUTIL_VERSION=598 -DPy_LIMITED_API=0x03060000 -DPSUTIL_ETHTOOL_MISSING_TYPES=1 -DPSUTIL_LINUX=1 -I/usr/include/python3.10 -c psuti [... cut ...]
2024-04-30 21:55:44,097 ! note: This error originates from a subprocess, and is likely not a problem with pip.
2024-04-30 21:55:44,103 ! ERROR: Failed building wheel for cffi
2024-04-30 21:55:44,108 ! ERROR: Could not build wheels for psutil, cffi, which is required to install pyproject.toml-based projects
2024-04-30 21:55:46,877 > Building wheel for cffi (pyproject.toml): finished with status 'error'
2024-04-30 21:55:46,886 > Successfully built OctoPrint zeroconf
2024-04-30 21:55:46,907 > Failed to build psutil cffi

revcrams, 2024-05-10:
[https://github.com/feelfreelinux/octo4a/issues/483](url)

feelfreelinux, 2024-05-17:
Released https://github.com/feelfreelinux/octo4a/releases/tag/2.0.0 which should resolve this issue :)

closed 2024-05-17 by feelfreelinux