feelfreelinux/octo4a #482: Java runtime exception when installing octoprint

AndrewBreyen, 2024-05-04:
Hello! I am getting an error when trying to install Octo4a. Full log is below, looks like issues building cffi?

`java.lang.RuntimeException: Process exited with error code 1. [Processing /root/OctoPrint- OctoPrint-fed2dd7 Ins`

```
🐸 Collecting argon2-cffi-bindings
🐸   Downloading argon2-cffi-bindings-21.2.0.tar.gz (1.8 MB)
🐸      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.8/1.8 MB 1.5 MB/s eta 0:00:00
🐸   Installing build dependencies: started
🐸   Installing build dependencies: still running...
🐸   Installing build dependencies: still running...
🐸   Installing build dependencies: finished with status 'error'
🐸   error: subprocess-exited-with-error
🐸  
🐸   × pip subprocess to install build dependencies did not run successfully.
🐸   │ exit code: 1
🐸   ╰─> [89 lines of output]
🐸       Collecting setuptools>=45
🐸         Using cached setuptools-69.5.1-py3-none-any.whl (894 kB)
🐸       Collecting setuptools_scm>=6.2
🐸         Downloading setuptools_scm-8.0.4-py3-none-any.whl (42 kB)
🐸            ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 42.1/42.1 kB 77.9 kB/s eta 0:00:00
🐸       Collecting wheel
🐸         Using cached wheel-0.43.0-py3-none-any.whl (65 kB)
🐸       Collecting cffi>=1.0.1
🐸         Downloading cffi-1.16.0.tar.gz (512 kB)
🐸            ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 512.9/512.9 kB 605.9 kB/s eta 0:00:00
🐸         Installing build dependencies: started
🐸         Installing build dependencies: finished with status 'done'
🐸         Getting requirements to build wheel: started
🐸         Getting requirements to build wheel: finished with status 'done'
🐸         Installing backend dependencies: started
🐸         Installing backend dependencies: finished with status 'done'
🐸         Preparing metadata (pyproject.toml): started
🐸         Preparing metadata (pyproject.toml): finished with status 'done'
🐸       Collecting tomli>=1
🐸         Downloading tomli-2.0.1-py3-none-any.whl (12 kB)
🐸       Collecting packaging>=20
🐸         Using cached packaging-24.0-py3-none-any.whl (53 kB)
🐸       Collecting typing-extensions
🐸         Downloading typing_extensions-4.11.0-py3-none-any.whl (34 kB)
🐸       Collecting pycparser
🐸         Downloading pycparser-2.22-py3-none-any.whl (117 kB)
🐸            ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 117.6/117.6 kB 434.6 kB/s eta 0:00:00
🐸       Building wheels for collected packages: cffi
🐸         Building wheel for cffi (pyproject.toml): started
🐸         Building wheel for cffi (pyproject.toml): finished with status 'error'
🐸         error: subprocess-exited-with-error
🐸      
🐸         × Building wheel for cffi (pyproject.toml) did not run successfully.
🐸         │ exit code: 1
🐸         ╰─> [48 lines of output]
🐸      
🐸                 No working compiler found, or bogus compiler options passed to
🐸                 the compiler from Python's standard "distutils" module.  See
🐸                 the error messages above.  Likely, the problem is not related
🐸                 to CFFI but generic to the setup.py of any Python package that
🐸                 tries to compile C code.  (Hints: on OS/X 10.8, for errors about
🐸                 -mno-fused-madd see http://stackoverflow.com/questions/22313407/
🐸                 Otherwise, see https://wiki.python.org/moin/CompLangPython or
🐸                 the IRC channel #python on irc.libera.chat.)
🐸      
🐸                 Trying to continue anyway.  If you are trying to install CFFI from
🐸                 a build done in a different context, you can ignore this warning.
🐸      
🐸             running bdist_wheel
🐸             running build
🐸             running build_py
🐸             creating build
🐸             creating build/lib.linux-armv7l-cpython-310
[... 54 lines cut ...]
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [Processing /root/OctoPrint-OctoPrint-fed2dd7  Ins

```

[image]

[octolog.log](https://github.com/feelfreelinux/octo4a/files/15207099/octolog.log)

hhocker, 2024-05-16:
@AndrewBreyen Not sure if you are keeping up or still having issues with this but take a look at this post:
https://github.com/feelfreelinux/octo4a/issues/483#issue-2286399198
for a possible fix. If you continue to  have issues (gcc errors out on cc1) then take a look at the workaround I found. 
https://github.com/feelfreelinux/octo4a/issues/475#issuecomment-2111350510
Also it appears that the maintainer is tweaking and working on a new release that hopefully will resolve your issue.

feelfreelinux, 2024-05-17:
Released https://github.com/feelfreelinux/octo4a/releases/tag/2.0.0 which should resolve this issue :)

closed 2024-05-17 by feelfreelinux