feelfreelinux/octo4a #483: Workaround on CFFI build issue

revcrams, 2024-05-08:
Not an Issue but a workaround - sorry I don't know how to work github as I'm a code dinosaur from day before yon internet.....

I've managed to work around the issue that seems to be related to building the package argon2-cffi-bindings when installing octo4a and octoprint 1.10 on legacy devices. It requires a bit of fiddling but if you're techie this may work for you also. I'm running Octo4a and octoprint on an LG Q6 with android 8.1 running

The basic problem on my build seems to be that the packages **gcc**, **musl-dev**, **python3-dev** & **libffi-dev** are not present within the alpine build that has installed - I've been out of touch with linux for years, but I presume that this may be some sort of architecture issue?, I've managed now to work around it by installing them manually and then kickstarting the octoprint build again from within alpine

This was my process:

> Install Octo4a 1.2.6
> Start the app as usual and tell it to install
> Let it run until you hit the error screen
> Exit the app then start it up again.

On mine I then get the Octoprint for Android screen we all know and love, but the server details look wrong and clear that nothing is running properly

> Click the cogwheel icons for settings
> Enable OpenSSH server
> set a password for root and note the server port **8022**
> Make a note of your phone's **ip address** (android settings -> about phone -> network)

Now on a PC or another device install an SSH client such as putty. If you're using windows ssh may just be built in
Connect to your phone's IP address, using port 8022, username: root and password <whatever you set it to in the above step>
(for me, I started a command window on windows and typed "`ssh <ip address> -p 8022 -l root`" )

You should now be logged into the alpine linux virtual machine running inside your phone. 

You then need to install some packages which seem to be required under this architecture:

> `apk add gcc musl-dev python3-dev libffi-dev`
You could install the packages one at a time: apk add gcc then musl-dev etc

Next run the following
> `cd /root`
> `cd Octo* && sed -i '/psutil/d' ./setup.py && pip3 install .`

I've just cribbed the main line from the octo4a build log. the build took a long time, which I guess is to be expected as compiling c on an lg Q6 is not a  quick job. Don't let windows go to sleep or you may get kicked from the ssh session and who knows what happens then.......

When the build had completed I then rebooted my phone, started octo4a and voila octoprint is up and happy

If you still get further failures, try and id the package that failed to build. for most of us it seemed to be argon2-cffi-bindings. you could try building it or whatever package is failing individually and working out the prerequisites:
`pip install <name of package>` e.g.  `pip install argon2-cffi-bindings` 
all i did was do some googling to look for the appropriate package to install for alpine - e.g. install libcffi for alpine

Octo4a is an amazing app, and I'm so grateful to Filip for creating so we can run octoprint on our phones - coping with so many different architectures must be a nightmare

kekkodance, 2024-05-10:
i can't get it to work, i even installed build-base again

kekkodance, 2024-05-11:
@feelfreelinux is there an ETA for the fix?

Paladin173, 2024-05-14:
This fixed it for me, thanks!!

hhocker, 2024-05-15:
> i can't get it to work, i even installed build-base again

@kekkodance Are you just getting a bunch of errors with GCC and cc1?

koruptdata, 2024-05-15:
> > i can't get it to work, i even installed build-base again
> 
> @kekkodance Are you just getting a bunch of errors with GCC and cc1?

Yes that's my issue, cc1 and GCC

hhocker, 2024-05-15:
I have tried a bunch of things to get the GCC error working. I finally gave up and updated the base alpine to 3.18 then could install py3-argon2-cffi package and bypass the need to build from source. 3.18 is necessary to get the right package version called out in 1.10.x

See my instructions here: https://github.com/feelfreelinux/octo4a/issues/475#issuecomment-2111350510

I think Filip is getting close on 1.3.0 as I see he has a pull request in that will radically change the root environment hopefully closing out a bunch of recent issues.

koruptdata, 2024-05-15:
> I have tried a bunch of things to get the GCC error working. I finally gave up and updated the base alpine to 3.18 then could install py3-argon2-cffi package and bypass the need to build from source. 3.18 is necessary to get the right package version called out in 1.10.x
> 
> See my instructions here: [#475 (comment)](https://github.com/feelfreelinux/octo4a/issues/475#issuecomment-2111350510)
> 
> I think Filip is getting close on 1.3.0 as I see he has a pull request in that will radically change the root environment hopefully closing out a bunch of recent issues.

THANK YOU! this worked perfectly! And thank you to feelfreelinux for making this possible in the first place.

feelfreelinux, 2024-05-17:
Shipped the new release, which should resolve this issue. I'll leave it open, until it is reconfirmed as working by a few users :)

https://github.com/feelfreelinux/octo4a/releases/tag/2.0.0

Tobiaqs, 2024-05-18:
Seems to properly work again on my device with this new release.

closed 2024-05-21 by feelfreelinux