feelfreelinux/octo4a #484: REQUEST: Allow selection of OctoPrint version after initial install failure

hhocker, 2024-05-14:
Filip, I know this whole 1.10 thing has been hard with things failing pretty bad across the board. I had been exploring errors with Rust compilers on my Fire HD device. I think it is an older arm architecture and found that 1.9.3 was working for me but alas I have broken my backup. I am attempting to reinstall but the system wont install due to pulling the latest. I have been attempting to understand the Kotlin code but I am more of a traditional Java man not Kotlin so It has been slow going.

What I am proposing is a version override that would allow folks after the initial "installation failed, clear and restart" dialog appears to choose which version they wish to run. I am trying to figure out how to install 1.9.3 manually over SSH but it would be much easier for folks to get devices working if we could run older versions instead of latest if it is known working. 

Here is the flow I am trying to put together. User gets the "there is no boot environment installed, install now?" like on the initial screen. They then attempt to automatically install the latest version. IF that should fail and they get the "installation was unsuccessful, clear and reattempt screen" there would be an option for "version selection". Choosing that option would cause the system to pop up a dialog that would have some scary wording that would say "Using older version is not recommended and may introduce security vulnerabilities. It is recommended you use the latest version. Really use other version?" And the selections would be "use latest" or "specify version". Clicking on Use latest would close the window and offer them the retry option as before. If they click on "Specify Version" then a popup appears that says "Non supported configuration, support for older versions will be unlikely to be unavailable by community members. Project contributors will not take action to fix errors found with older versions." Then offer a place to put in the Octoprint tag number to be downloaded and built. 

If I can figure out Kotlin I will attempt this on my older Fire HD but at this time I am still working to copy things manually. Let me know how hard it would be to support older versions. Thanks.

kekkodance, 2024-05-14:
He's already planning for it in the next release.

hhocker, 2024-05-14:
AH. My apologies. I didn't know where to look for a roadmap. Thanks for the info.

closed 2024-05-14 by hhocker

feelfreelinux, 2024-05-17:
Implemented in version 2.0.0 :)