feelfreelinux/octo4a #488: OctoPrint-Octo4a_battery

joejtrsmoe, 2024-06-05:
https://github.com/tobiasgraf/OctoPrint-Octo4a_battery 
https://github.com/tobiasgraf/OctoPrint-Octo4a_battery/archive/master.zip
This plugin no longer works in version 2.0.0.

Also "start on boot" doesnt work on my new old phone u325ac cricket icon 2.
Think both are permission issues?

Downgraded my s7 back to 1.2.6 during octoprint 1.0.0 with workaround and everything works fine. 
Cant install 1.2.6 on u325ac octoprint 1.1.0. same old crash on install problems. I couldnt find a work around.

hhocker, 2024-06-05:
Im going to guess that this is caused by a change in the base OS of the proot system. I would have to see if I can find the information on where to get battery info in the new Alpine 3.19 core.

hhocker, 2024-06-05:
Also you MIGHT be able to use my old workaround to upgrade the base proot environment to support the necessary dependencies of 1.10.1
https://github.com/feelfreelinux/octo4a/issues/475?notification_referrer_id=NT_kwDOAKyvN7QxMDQzODUxODI2MDoxMTMxNzA0Nw#issuecomment-2111350510

hhocker, 2024-06-05:
@feelfreelinux Filip, wondering if the new boot strap entry script forgets to bind /sys.

Looking at https://github.com/feelfreelinux/octo4a-bootstrap-builder/blob/main/scripts/run-bootstrap-android.sh should there be an "-b /sys " added in there to get the sysfs to pull in? Just a thought for this plugin. Not sure how to test this myself.

joejtrsmoe, 2024-06-06:
> Also you MIGHT be able to use my old workaround to upgrade the base proot environment to support the necessary dependencies of 1.10.1 #475

I was able to get cricket to install 1.2.6 using workaround. All plugins are broke however. Unable to install any. Also didnt fix start on boot so that must be phone issue not Octo4a 2.0 issue.

My S7 is fine on 1.2.6 but was trying to setup second printer on cricket icon 2.
Cricket Icon 2 works fine on 2.0 but no auto boot or one small plugin not working. Auto boot would be more important.

Still cant complain. Octo4a is awesome software. Thanks so much.

joejtrsmoe, 2024-06-10:
Fixed auto boot with adb command
adb shell pm grant com.octo4a android.permission.SYSTEM_ALERT_WINDOW
just down to battery plugin not working which is fine

jzaor, 2025-02-16:
Please test this app to resolve battery status exposition problem:
[https://github.com/jzaor/Battery-capacity-file/](https://github.com/feelfreelinux/octo4a/issues/)