feelfreelinux/octo4a #493: octoklipper error: externally-managed-environment

zaid-hm, 2024-07-06:
I'm trying to run klipper on a galaxy tab A (2019)
But when running the command in the instruction I get the following error:
```

bash -c "$(curl -fsSL https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scrip
ts/setup-klipper.sh)"
Setting up klipper
Do you have "Plugin extras" installed? (y/n): y
Installing dependencies...

OK: 404 MiB in 183 packages
error: externally-managed-environment

× This environment is externally managed
╰─>
    The system-wide python installation should be maintained using the system
    package manager (apk) only.

    If the package in question is not packaged already (and hence installable via
    "apk add py3-somepackage"), please consider installing it inside a virtual
    environment, e.g.:

    python3 -m venv /path/to/venv
    . /path/to/venv/bin/activate
    pip install mypackage

    To exit the virtual environment, run:

    deactivate

    The virtual environment is not deleted, and can be re-entered by re-sourcing
    the activate file.

    To automatically manage virtual environments, consider using pipx (from the
    pipx package).

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
localhost:~# bash -c "$(curl -fsSL https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scrip
ts/setup-klipper.sh)"
```

Bartouzz, 2024-07-08:
There is a new script for the 2.0.0 version of octo4a

https://github.com/feelfreelinux/octo4a/blob/master/scripts/setup-klipper-moonraker.sh

By the way i'm still figuring out, how to get it to work i'm running network error.
Maybe you should install 1.2.6 version of octo4a and use the command line you did.

zaid-hm, 2024-07-08:
@Bartouzz did you manage to get it to work, I'm running into a network issue as well

Bartouzz, 2024-07-08:
@zaid-hm Still not ... Starting to lose hope, see my current klippy log:

mcu 'mcu': Starting serial connect
mcu 'mcu': Unable to open serial port: [Errno 13] Could not exclusively lock port /dev/ttyOcto4a: [Errno 13] Permission denied
webhooks client 489798200720: New connection
webhooks client 489798200720: Client info {'program': 'Moonraker', 'version': 'v0.8.0-384-g8f3b30a'}
mcu 'mcu': Unable to open serial port: [Errno 13] Could not exclusively lock port /dev/ttyOcto4a: [Errno 13] Permission denied

seem's like permission problem but i can't find the right answer. i'll post if get something

schmttc, 2024-07-18:
Try using this install script, it ignores the 'externally managed' error:
https://raw.githubusercontent.com/schmttc/octo4a/fixklipperinstall/scripts/setup-klipper.sh

schmttc, 2024-07-26:
I'm working on this issue, am updating the klipper install script to work with octo4a 2.0
Once I've done some testing will provide the details for you to confirm.

schmttc, 2024-07-27:
I've fixed the environment and permission denied error, but have run into the next error (non rooted device) https://github.com/schmttc/octo4a/blob/schmttc-patch-1/scripts/setup-klipper.sh
Have confirmed that Klipper on a Raspberry Pi 3B is working with the printer.

```
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/klipper/klippy/serialhdl.py", line 262, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/klipper/klippy/serialhdl.py", line 319, in get_response
    self.serial.raw_send_wait_ack(cmds[-1], minclock, reqclock,
  File "/klipper/klippy/serialhdl.py", line 254, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/klipper/klippy/serialhdl.py", line 61, in _error
    raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed
```

schmttc, 2024-08-01:
I've used setup-klipper-moonraker.sh and reduced it to only install Klipper, which resolves all issues.

Please try https://github.com/schmttc/octo4a/blob/klipper-fix/scripts/setup-klipper.sh

schmttc, 2024-08-01:
This may need more troubleshooting, while everything kicked off OK either Octoprint or Klipper crashed part way though a test print

apparle, 2024-09-10:
I'm trying to use [setup-klipper-moonraker.sh](https://github.com/feelfreelinux/octo4a/blob/master/scripts/setup-klipper-moonraker.sh) itself and faced similar `Permission denied` errors.
Based on your pull request, I added `LD_PRELOAD=/home/octoprint/ioctl-hook.so` while launching klippy and now I'm getting errors similar to you. @schmttc how did you solve those errors?
```
mcu 'mcu': Wait for identify_response
```
 `Wait for identify_response` as you noted above.

What is connection information in your printer.cfg? Mine is:
```
[mcu]
serial: /dev/ttyOcto4a
baud: 250000
restart_method: command
```
Also, I checked that my printer is setup correctly because other devices running Klippy are able to control it. 
And FWIW, I'm running Octo4a on a non-rooted Android 11 device -- and that device could control a Marlin-firmware-printer correctly without the extension.

schmttc, 2024-09-11:
> I'm trying to use [setup-klipper-moonraker.sh](https://github.com/feelfreelinux/octo4a/blob/master/scripts/setup-klipper-moonraker.sh) itself and faced similar `Permission denied` errors. Based on your pull request, I added `LD_PRELOAD=/home/octoprint/ioctl-hook.so` while launching klippy.py and now I'm getting errors similar to you. @schmttc how did you solve those errors?
> 
> ```
> mcu 'mcu': Wait for identify_response
> ```
> 
> What is connection information in your printer.cfg? Mine is:
> 
> ```
> [mcu]
> serial: /dev/ttyOcto4a
> baud: 250000
> restart_method: command
> ```
> 
> Also, I checked that my printer is setup correctly because other devices running Klippy are able to control it. And FWIW, I'm running Octo4a on a non-rooted Android 11 device -- and that device could control a Marlin-firmware-printer correctly without the extension.

Sadly I never got to the bottom of this, it worked intermittently a couple of times and then not again. 
Same issue on unrooted Android 11 and 13.
Connection works when connecting to a Raspberry Pi 3
MCU config
```
[mcu]
serial: /dev/ttyOcto4a
restart_method: command
```
Error
```
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/root/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/klipper/klippy/serialhdl.py", line 262, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/klipper/klippy/serialhdl.py", line 319, in get_response
    self.serial.raw_send_wait_ack(cmds[-1], minclock, reqclock,
  File "/root/klipper/klippy/serialhdl.py", line 254, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/root/klipper/klippy/serialhdl.py", line 61, in _error
    raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed
```

schmttc, 2024-09-11:
@apparle I've done some testing, and if you use the older Octo4a app version 1.2.6, it installs the latest Octoprint, and fixes the Klipper issues

spenc302, 2024-10-16:
Find a fix or at least a cause? Tried to roll back to older version of octo4a and it's still giving me permission errors, would a work around be using a rooted device

apparle, 2024-10-28:
@akei-ai can you report which version of the https://github.com/feelfreelinux/octo4a-bootstrap-builder was used (you select that the first time set up octoprint)?
Based on my conversation with @feelfreelinux, he had fixed the ioctl issue in https://github.com/feelfreelinux/octo4a-bootstrap-builder/commit/364ee8b282b7f751bd8fa3b84a46bce6fbe6aaf4 such that klipper can open the device, and I was supposed to test it. But for other reasons, I had to rollback to Marlin firmware and never got around to testing it.

apparle, 2025-03-21:
I tried with a latest build of the builder that includes the fix. And I also did the changes to include LD_PRELOAD, but I still can't get it to work :-( 
Still plagued with `Serial connection closed` errors. And I tried with Android 9 (lineage OS) and Android 11 (e/OS/).

@akei-ai did you use a rooted device? Mine isn't rooted, and I wonder if that's the big difference.

that-ambuj, 2025-03-21:
@apparle Octo4a doesn't require root and nor does it ask for it even if your device is rooted.

Not using root is one of its drawbacks. If you want some advice from me, just root that phone, use linux deploy for klipper and point mainsail to webcam(localhost:<some port>) url for Octo4a. Worked pretty well for me, you'll find plenty of guides online as well, except for the webcam part.

apparle, 2025-03-21:
> [@apparle](https://github.com/apparle) Octo4a doesn't require root and nor does it ask for it even if your device is rooted.
>
> Not using root is one of its drawbacks. If you want some advice from me, [...]

I did try that route but it's harder than it looks.
1st Linux Deploy is broken Android 10 onwards, so I've to downgrade to Android 9. 
2nd, the linux deploy debian images use Debian10 which is super old and doesn't have some packages like newer version of Python, which are required by KIAUH scriptware, so that has to be compiled from scratch. 
3rd, my phone's kernel lacks the CH34x driver needed for USB-to-serial on my printer, so debian doesn't have to serial port. Instead I've to rely on the octo4a's userspace usb-to-serial driver by actually starting octo4a and then symlinking the userspace serialport file into my debian session and using that.
And for whatever weird reason, I see that if my phone stays locked, the whole thing slows down -- and none of the settings I could find were solving this throttling. 

It get's extremely janky, so I'm looking for a simpler fix (like octo4a's serial driver to just work).