feelfreelinux/octo4a #494: Installed FW 3.14.0 on mk3s+ and now just get "thermal anomaly" and "paused thermal error"

rs-garrick, 2024-07-10:
I don't know what happened.  I just updated the firmware on my mk3s+ to 3.14.0 and I can't print anything.  During warm up, it beeps with either "thermal anomaly" or "paused thermal error".  The display briefly shows the desired hot end and bed temperatures.  After the beep, the desired hot end temp is 0.

AnthonyZ71, 2024-07-17:
I am seeing this error on an mk3 as well.  I had been printing fine, upgraded my slicer, after pressing ok to continue at the "you need to upgrade your firmware" message, I went ahead and upgrade, and now I can no longer print.

ferrofem, 2024-07-25:
I'm having the same issue, I haven't tried flashing back to the previous version yet. It has been sitting the last 2 years and hadn't been updated since 2022.

rs-garrick, 2024-07-29:
I've moved away from octo4a. So I'm going to close my issue.

closed 2024-07-29 by rs-garrick