feelfreelinux/octo4a #497: 2.1.0 reported as harmful by Play Protect

sm3g, 2024-07-16:
Google Play Protect is reporting the new release 2.1.0 as 'harmful' but without any additional details about why. It just states "This app may be harmful."
I've been running versions for a couple years, including 1.2.5, 1.2.6 and 2.0.0 and haven't seen this with any earlier release. A warning message popped up while upgrading from 2.0.0 and is also shown in the Play Store app. 
I'm guessing this is a false alarm, but wanted to share. I took a quick look at the changes that were merged with 2.1.0 and see that most are related to Camera modules (as advertised) but going any deeper is above my head.

This is on a Samsung Galaxy S8 running Android 9 (relatively stock and up to date for the hardware)

biddster, 2024-08-12:
I've just had this issue. Dug an old galaxy s8 out of the drawer, factory reset and booted. 

Didn't sign in to google or samsung account. 
Updated all existing apps and google play services. 
Tried to install octo4a 2.1.0 via adb and directory on device and I get "app may be harmful"
I click install anyway and the install fails with:

```
adb: failed to install octo4a-2.1.0.apk: Failure [INSTALL_FAILED_VERIFICATION_FAILURE]
```

sebastianheyn, 2024-08-14:
Hey guys, I am also a user. This message could happen, if your OS version is not marked as supported.