feelfreelinux/octo4a #505: [Feature request] Expose battery API

Zydnar, 2024-08-22:
I would love to have battery level available in Octo4a like with plugin https://github.com/tobiasgraf/OctoPrint-Octo4a_battery but I know it's connected to Android SE Status and if it's set to "enforcing" battery status isn't available within Alpine in Octo4a so to workaround that the app would require extra permission to read battery level and expose it that way.

jzaor, 2025-02-16:
Please test this app to resolve battery status exposition problem:

https://github.com/jzaor/Battery-capacity-file/tree/prerelease

Zydnar, 2025-02-17:
> Please test this app to resolve battery status exposition problem: https://github.com/jzaor/Battery-capacity-file/

Can't test it on that phone from my issue yet maybe on next weekend but on a newer model of this series seems working fine.