feelfreelinux/octo4a #518: Installation fails - S24 Ultra - Android 14 - One UI 6.1

zeljkokalezic, 2024-11-30:
💻 No bootstrap detected, proceeding with installation
💻 Arch: aarch64
💻 Downloading bootstrap 073b9017 from https://github.com/feelfreelinux/octo4a-bootstrap-builder/releases/download/v073b9017-1.10.3/bootstrap-1.10.3-aarch64.zip
💻 Request to https://objects.githubusercontent.com/github-production-release-asset-2e65be/795680395/68514361-7d4a-4685-afa7-46886385e461?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20241130%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20241130T003242Z&X-Amz-Expires=300&X-Amz-Signature=735303649d2df39020989ede91e0dcb73b53e981ef8a84c8b8e5009c7a57a9b5&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dbootstrap-1.10.3-aarch64.zip&response-content-type=application%2Foctet-stream returned status code 200
💻 Bootstrap downloaded, extracting it...
💻 >ls
💻 /storage/emulated/0/Android/data/com.octo4a/files
🐸 bin
🐸 build-version.txt
🐸 entrypoint.sh
🐸 fake_proc_stat
🐸 libexec
🐸 rootfs.tar.xz
💻 >ls
💻 /storage/emulated/0/Android/data/com.octo4a/files
🐸 No bootstrap detected, extracting
🐸 proot error: execve("/bin/sh"): Function not implemented
🐸 proot info: possible causes:
🐸   * the program is a script but its interpreter (eg. /bin/sh) was not found;
🐸   * the program is an ELF but its interpreter (eg. ld-linux.so) was not found;
🐸   * the program is a foreign binary but qemu was not specified;
🐸   * qemu does not work correctly (if specified);
🐸   * the loader was not found or doesn't work.
🐸 fatal error: see `proot --help`.
🐸 proot warning: ptrace(PEEKDATA): I/O error
🐸 proot error: can't chmod '/data/data/com.octo4a/files/bootstrap/tmp/proot-9430-IgBgSX': Function not implemented
🐸 proot error: can't chdir to '': Function not implemented
💻 An exception has occurred at:
java.lang.RuntimeException: Process exited with error code 1. [No bootstrap detected, extractingproot error: exe
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput(ProcessUtils.kt:43)
	at com.octo4a.utils.ProcessUtilsKt.waitAndPrintOutput$default(ProcessUtils.kt:16)
	at com.octo4a.repository.BootstrapRepositoryImpl$extractBootstrap$2.invokeSuspend(BootstrapRepository.kt:188)
	at com.octo4a.repository.BootstrapRepositoryImpl$extractBootstrap$2.invoke(Unknown Source:10)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:91)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:165)
	at kotlinx.coroutines.BuildersKt.withContext(Unknown Source:1)
	at com.octo4a.repository.BootstrapRepositoryImpl.extractBootstrap(BootstrapRepository.kt:176)
	at com.octo4a.repository.OctoPrintHandlerRepositoryImpl$beginInstallation$2.invokeSuspend(OctoPrintHandlerRepository.kt:134)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
	at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
	at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
	at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

Exception:
java.lang.RuntimeException: Process exited with error code 1. [No bootstrap detected, extractingproot error: exe

BigSpice, 2024-12-27:
Project not active i think : same issue, javalang.

mjkapkan, 2025-01-20:
@BigSpice Do you mean that project is not maintained anymore? I also got this issue on pixel 6a. At first it worked well, but then there was an OS update. And after it installation got corrupted. I tired re-installing, clearign cashe and storage and re-installing, but none of that helped. Any clue what could be causing this?

squee666, 2025-03-28:
same issue , also says no storage permissions even when assigned